john30/ebusd-configuration #348: Wolf WPM-1

ThxAndBye, 2023-08-04:
Hello,

I'm in the process of integrating my Wolf WPM-1 heat pump into Home Assistant running on a Raspberry Pi 4 with an eBus v5 adapter connected via GPIO. The connection seems to work and `ebusctl scan full` does also detect the devices on the bus.  

**ebusctl info** reports:  
```
version: ebusd 23.2.p20230716
update check: revision 23.2 available, broadcast.csv: different version available, vaillant/05.vd6.csv: different version available, vaillant/23.solsy.cc.csv: different version available, vaillant/25.ehp.hwc.csv: different version available, vaillant/51.vr630.mc.3.csv: different version available
device: /dev/ttyAMA0, enhanced high speed, firmware 1.1[3612].1[3612]
signal: acquired
symbol rate: 51
max symbol rate: 79
min arbitration micros: 9
max arbitration micros: 64
min symbol latency: 4
max symbol latency: 5
reconnects: 0
masters: 7
messages: 314
conditional: 16
poll: 44
update: 4
address 03: master #11
address 07: master #16
address 08: slave #11, scanned "MF=Wolf;ID=  ;SW=4001;HW=0000"
address 0c: slave #16, scanned "MF=Kromschroeder;ID=  ;SW=0204;HW=-"
address 10: master #2
address 15: slave #2, scanned "MF=Kromschroeder;ID=  ;SW=0204;HW=-"
address 30: master #3
address 31: master #8, ebusd
address 35: slave #3, scanned "MF=Kromschroeder;ID=  ;SW=0204;HW=-"
address 36: slave #8, ebusd, scanning
address 51: slave, scanned "MF=Wolf;ID=  ;SW=4001;HW=0000"
address 70: master #4
address 75: slave #4, scanned "MF=Kromschroeder;ID=  ;SW=0204;HW=-"
address f1: master #10
address f6: slave #10, scanned "MF=Kromschroeder;ID=  ;SW=0204;HW=-"
```  

But other than `datetime` and `error` I'm not able to receive any data using the csv config from: ebusd-configuration/ebusd-2.x.x/de

ThxAndBye, 2023-12-26:
i have some commands on the eBus that are send every 10sec or so. How would I get ebusd to recognize them?  

These are the commands I've seen that I'm sure what they are:
```
T_WWasser
f151502203cc0e00 / 02fe01 = 10
UIN   fe01=510

Mischer1
0751502203900f00 / 02ec00 = 6
UIN   ec00=236

Brennerstunden
0708502203de2a02 / 02484a = 13
UIN   484a=19016

Zeitprogramm
0708502203f62702 / 020100 = 6
UIN   0100=1
```

And this is what I've added to the .csv:
```
u,feuerung,t_wwasser,temperatur_warmwasser,f1,51,5022,03cc0e,t_wwasser,,UIN,10,Grad Celcius,t_wwasser,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
u,feuerung,mischer_eins,mischer_1,07,51,5022,03900f,mischer_eins,,UIN,10,Grad Celcius,mischer_eins,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
u,feuerung,bennerstunden,Brennerstunden,07,08,5022,03de2a,brennerstunden,,UIN,,,brennerstunden,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
u,feuerung,zeitprogranmm,Zeitprogramm,07,08,5022,03f627,zeitprogramm,,UIN,,,zeitprogramm,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
```

But they aren't added to the MQTT topic. Any help would be appreciated.

ProgWolf18, 2024-05-30:
ThxAndBye, "03" in your example is the number of bytes of the following data and not part of the message definition (https://github.com/john30/ebusd/wiki/eBUS-background). So your csv file has to look like 
...,5022,cc0e00,... 
...,5022,900f00,... and so on
HTH

OSmdt, 2024-12-02:
Hello @ThxAndBye  and @ProgWolf18 ,
I am trying as well to get a Wolf heat pump analysed and read out with ebusd (WPM-1, BWL-1, ...).
I managed to see the output on console and e.g. the info with ebusctl. 
I also manged to use other CSV files from local drive. 
However as you also mentioned above, the "ebusd-2.x.x" don't conatin many info and also don't really match the setup with a heat pump I guess.

So I wonder if you managed to get more messages and values decoded, and if you can share some cvs files.  

That would really be helpfull. I am also willing to analyze on my side, but having a good starting point will make this much easier.

Thanks for your help in advance.

domasist, 2025-06-02:
@OSmdt did you managed to get Wolf BWL-1 working over ebusd with Home Assistant?