john30/ebusd-configuration #350: Vailant Recovair 260 - no values

cvastag, 2023-08-08:
I'm getting no data from Vailant Recovair 260. Also csv is loaded, there are no values stored. What's wrong?

config:
EBUSD_OPTS="--scanconfig=full --configpath=/etc/default/ebusd-configuration/ebusd-2.1.x/de --enablehex -d ens:10.0.20.245:9999 --mqtttopic=ebusd/%circuit/%name --mqtthost=10.0.20.14 --mqttport=1883 --mqttuser=mqtt --mqttpass=123456 --mqttlog --accesslevel=* --loglevel=debug --address=ff --httpport=8889 --htmlpath=/var/ebusd/html"

ebusctl I:
version: ebusd 23.2.23.2
update check: OK
device: 10.0.20.245:9999, enhanced, firmware 1.1[3612].1[3612]
access: *
signal: acquired
symbol rate: 23
max symbol rate: 37
scan: finished
reconnects: 1
masters: 2
messages: 79
conditional: 0
poll: 0
update: 8
address 03: master #11
address 04: slave #25, ebusd, scanning
address 08: slave #11, scanned "MF=Vaillant;ID=recoV;SW=0227;HW=0203", loaded "vaillant/08.recov.csv"
address ff: master #25, ebusd

ebusctl find: (shorten)
....
recov VolumeFlow = no data stored
recov VolumeFlowAct = no data stored
recov WTTyp = no data stored
recov YearKwh = no data stored
recov YesterdayKwh = no data stored
scan id = no data stored
scan.08  = Vaillant;recoV;0227;0203
scan.08 id = 21;18;20;0010016040;0001;005522;N3

Data: (shorten)
   },
   "YesterdayKwh": {
    "name": "YesterdayKwh",
    "passive": false,
    "write": false,
    "lastup": 0
   }
  }
 },
 "scan.08": {
  "messages": {   "": {
    "name": "",
    "passive": false,
    "write": false,
    "lastup": 1691480446,
    "zz": 8,
    "fields": {
     "MF": {"value": "Vaillant"},
     "ID": {"value": "recoV"},
     "SW": {"value": "0227"},
     "HW": {"value": "0203"}
    }
   },
   "id": {
    "name": "id",
    "passive": false,
    "write": false,
    "lastup": 1691480452,
    "zz": 8,
    "fields": {
     "prefix": {"value": "21"},
     "year": {"value": "18"},
     "week": {"value": "20"},
     "product": {"value": "0010016040"},
     "supplier": {"value": "0001"},
     "counter": {"value": "005522"},
     "suffix": {"value": "N3"}

cvastag, 2023-08-08:
ahh found something out:

if I send out hex commands, I receive an answer
How can I do send hex commands automatically?

2023-08-08 15:01:23.850 [main notice] hex cmd: 3108b509030d1a0e
2023-08-08 15:01:24.114 [update notice] sent read recov InstallationHight QQ=31: 150

Silverstar, 2023-08-30:
By default, only values which are configured to be polled will be automatically fetched. Some will be send by devices on the bus. You can change them in the config file to be polled: https://github.com/john30/ebusd/wiki/4.1.-Message-definition

john30, 2023-09-23:
> By default, only values which are configured to be polled will be automatically fetched.

exactly