john30/ebusd-configuration #352: How to listen to unknown messages

stalniy, 2023-08-11:
Hi guys,

Thank you very much for this project! The work done here is just enormous!

I’m able to control most of my Vaillant eloblock parameters but there is one which doesn’t work. It’s part load HWC. I can change using diagnostic codes on my boiler but then I don’t see any new message on Enid logs. 

Can somebody tell me how to catch this change so then I can add proper message id in configuration? Do I need just to run ebusd for a long time to see it or I need vaillant thermostat and somehow listen to its changes?

I’d like to update documentation then to make this thing clear for others.

Silverstar, 2023-08-30:
As the part load is set on the burner unit itself, and not on the controller, which does not necessarily need that information, the burner does not send that change. So you would have to read all addresses on the burner (which you don't already know), change the part load setting, read all again, and check for another value in a addresses. You may find multiple addresses, decode it and hope that you find your part load value...

Silverstar, 2023-08-30:
https://github.com/john30/ebusd/tree/master/contrib/scripts/readallvaillantregisters.sh

stalniy, 2023-09-02:
Awesome! thank you very much for pointing me to this script!

closed 2023-09-02 by stalniy