john30/ebusd-configuration #353: Handling of writable items might fail starting with HA 2024.2.0

JongsmaSimon, 2023-08-19:
I have 3 writable items in my ebusd configuration.
Today (after HA upgrade) there appeared the following warning:

“This stops working in version 2024.2.0. Please address before upgrading.
 Some MQTT entities have an entity name that starts with the device name. This is not expected. To avoid a duplicate name the device name prefix is stripped off the entity name as a work-a-round. Please inform the maintainer of the software application that supplies the affected entities to fix this issue.”
List of affected entities:

number.ebusd_hmu_doeltempboiler
number.ebusd_hmu_doeltempwoning
number.ebusd_hmu_stooklijn

edudele, 2023-08-21:
here -> https://github.com/john30/ebusd/issues/975

john30, 2023-09-03:
duplicate of https://github.com/john30/ebusd/issues/975

closed 2023-09-03 by john30