john30/ebusd-configuration #357: How SetOverride mode in Protherm

itProfi, 2023-09-22:
> It's the method of controlling without any thermostat, if you have thermostat you must set values in it. Also it depends on mode (heating, hot water, both or nether). Can you change values by keys, did they stay or reset?

I will try later today. To check on all modes and with keys.
My plan is to put thermo valves on collector with relays connected to it, to control every room. I will get temperature of BLE or DS18B20 temperature meters and want to make automation. 

Easy way would be to leave flow temp  like 27. Slowly close water valves in collector which reached temperature, after reaching wanted temperature in all rooms shut off HeatingSwitch (or just add simple relay to Protherm and use this one to start/stop)

If you have any better advice, please let me know :) Would be nice to chat over telegra/signal/whatsapp

> @Trilis29 are you using radio module or directly connected one? Radio is resetting values from one in remote unit, and i did not found reliable way to override it.

using ebus-adapter 3 wemos now. Do not have spare RPI.

EDIT:

`root@5fcef0cf7e74:/# ebusctl write -c bai FlowSetPotmeter 27.00
done

root@5fcef0cf7e74:/# ebusctl find -w
bai FlowSetPotmeter = 27.00
bai HeatingSwitch = no data stored
bai HwcSetPotmeter = no data stored
bai HwcSwitch = no data stored
broadcast id = no data stored
broadcast queryexistence = no data stored

root@5fcef0cf7e74:/# ebusctl find -w
bai FlowSetPotmeter = 27.00
bai HeatingSwitch = no data stored
bai HwcSetPotmeter = no data stored
bai HwcSwitch = no data stored
broadcast id = no data stored
broadcast queryexistence = no data stored

root@5fcef0cf7e74:/# ebusctl read -f -c bai FlowSetPotmeter
28.00

root@5fcef0cf7e74:/# ebusctl find -w
bai FlowSetPotmeter = no data stored
bai HeatingSwitch = no data stored
bai HwcSetPotmeter = no data stored
bai HwcSwitch = no data stored
broadcast id = no data stored
broadcast queryexistence = no data stored`

Seems data is stored but not sent to Protherm. Using read again restores to same

EDIT2:

[https://github.com/john30/ebusd/issues/179](url)

with SetModeOverride it is possible to change FlowSetPotmeter. Checking if possible over mqtt

_Originally posted by @Trilis29 in https://github.com/john30/ebusd-configuration/issues/205#issuecomment-818843422_

itProfi, 2023-09-22:
> with SetModeOverride it is possible to change FlowSetPotmeter. Checking if possible over mqtt

Can you describe how you set this on Proterm?

Trilis29, 2023-09-23:
One question, why are you not using HomeAssistant? Would be easier to control everything

closed 2024-07-04 by itProfi