john30/ebusd-configuration #358: unable to load scan config 15

Benni1123, 2023-10-08:
Hi @john30 
I habe this issue.
`2023-10-08 18:47:49.170 [update notice] sent scan-read scan.15 id QQ=31: 21;22;32;0010045485;0953;011543;N1
2023-10-08 18:47:49.170 [bus notice] scan 15: ;21;22;32;0010045485;0953;011543;N1
2023-10-08 18:47:49.295 [main error] unable to load scan config 15: no file from vaillant with prefix 15 matches ID "ctlv3", SW0612, HW3004
2023-10-08 18:47:49.295 [main error] scan config 15: ERR: element not found`

ebusctl find output:
`scan.08  = Vaillant;BAI00;0807;7603
scan.08 id = 21;22;44;0010022010;0001;017235;N0
scan.15  = Vaillant;CTLV3;0612;3004
scan.15 id = 21;22;32;0010045485;0953;011543;N1`

Can you please help me, so that I can use my heating?

john30, 2023-10-14:
then there is no such file yet in the repo (15.ctlv3.csv) and you'd have to create that yourself. you could look for another ctlv* and see if that works

john30, 2023-10-14:
[#487 ](https://github.com/john30/ebusd/issues/487) might be related

Benni1123, 2023-10-15:
> [#487 ](https://github.com/john30/ebusd/issues/487) might be related
[image]

it looks not bad.

[image]
but unfortunately most of it doesn't work.

cjoe-51, 2024-11-20:
Linking 15.ctlv2.csv to 15.ctlv3.csv made the ebusd loading the config file finally and 279 of new variables appeared however **all** with result " **no data stored**" :(
What did I made wrong?

scan.15  = Vaillant;CTLV3;0709;3704
scan.15 id = 21;23;34;0020260919;0953;007103;N6

john30, 2024-12-28:
done with dec8be9e12b8a85d04cd2db9b41e5e0633d84969

closed 2024-12-28 by john30