john30/ebusd-configuration #362: Configuration for TEM ES 5941 FW

sporkbyspork, 2023-11-03:
I'd like to create a configuration for the TEM ES 5941 FW controller.

It presents itself as TEM;25467;0041;0010.
I receive only very limited data (e.g. time and date) by listening on the bus.
In issue #209  it is stated that I might have to query the registers manually.

How do I do that?

For example, the language selection is saved in ID 04:056.

There is a bit more information about the available data in the datasheet of a Modbus interface (https://www.taconova.com/fileadmin/downloads/BA/Modbus_Interface.pdf) if that helps.

mhbacks, 2024-03-27:
Hi,
I have the same fresh water station at home. Have you found out anything? What is your status?
best regards,
Max

mhbacks, 2024-03-28:
How are you able to access the time? which hex command do you send to get it?
in meanwhile i asked TEM directly for support. doesnt think that i get a good response, but when i do not ask, the "no" is fix..

sporkbyspork, 2024-03-28:
How did you contact TEM? I might do the same and maybe it helps.

This is my latest config .csv:
`#type,circuit,name,comment,QQ,ZZ,PBSB,ID,field1,part,type/templates,divider/values,unit,comment,field2,part,type/templates,divider/values,unit,comment
r1,abcdef,test1,,31,10,5010,50,test1,,STR,,,,,,,,,
r1,abcdef,test2,,31,10,5010,51,test2,,STR,,,,,,,,,
r1,abcdef,test3,,10,fe,fe01,0b3539,test3,,STR,,,,,,,,,
`
Reading the issues, it seems that other devices are much more "communicative" than the TEM ones.
But I have no idea how to manually read registers or how to trigger the TEM device to send more data.

mhbacks, 2024-03-28:
i try the email and phonenumber on tem.ch for contact. also some other companies i discovered with a relation to tem. i will update the case in the next two weeks (some are on holiday)

mhbacks, 2024-05-14:
Hi,
in meanwhile i had contact with a technican at tem. He said that the documents exist, but he is not authorized for publish them. 

@john30 is it possible get assistance from your side? can you give us a hint, how we are able to convert the given ebus ids in the documentation to a valid ebus command, so we are able to continue our discovery? 

the TEM ZIF 180 ebus to modbus converter is online not orderable. I only found a Ochsner branded device for holy shit 800€. This is a little bit too high risk to order it only for two weeks and ask for reversal..

best regards,
Max

sporkbyspork, 2026-04-06:
With the help of Claude, I've created a working read-only CSV configuration starting with my findings above.
The TEM_P data type handles register addressing as documented, but the ID suffix must be 0040, not 0042/004e as used in the existing heat pump config. This matches the suffix used for TEM 11111 gas boilers [https://github.com/john30/ebusd-configuration/issues/27](https://github.com/john30/ebusd-configuration/issues/27) and [https://github.com/dlbogdan/ebusd-TEM](https://github.com/dlbogdan/ebusd-TEM).
The register map is based on the above mentioned [ZIF 180 eBUS-Modbus Interface datasheet](https://www.taconova.com/fileadmin/downloads/BA/Modbus_Interface.pdf).

Most parameters can be read, for example:
`ebusctl read DhwTemp`
results in 
`00-067;Temperatur;_C;300.0;-50.0;32.3`.

I had no success with writing values but maybe this can become a starting point for supporting this controller.

I've uploaded the CSV draft here for the time being: [https://gist.github.com/sporkbyspork/5c0a3ed027e1ff309c054665ad8e6d4c](https://gist.github.com/sporkbyspork/5c0a3ed027e1ff309c054665ad8e6d4c)

sporkbyspork, 2026-11-10:
Writes work, too, and the password only gates against manual input via the buttons, not via ebus.

Gist [updated](https://gist.github.com/sporkbyspork/5c0a3ed027e1ff309c054665ad8e6d4c).