john30/ebusd-configuration #363: Problem with vr_71 module

Desmoyo, 2023-11-05:
I have a problem with this module, I cannot automatically integrate it into home assistant, I can see its readings of SetActorState, Mc1FlowTempDesired, Mc2FlowTempDesired, Mc3FlowTempDesired, MixerStatus and MixerMovement, in an ebusd terminal and I see its traffic in mqttExplorer, but they are not added to the device in home assistant, does anyone know what could happen or if there is poor management on this device?

Thanks

Desmoyo, 2023-11-05:
Solved by brute force

miho9, 2023-11-11:
what do you mean by brute force? I am also using the VR71 in my setup and am interested in these values.

Dinth, 2023-11-12:
@Desmoyo yes, please share your solution. Im VR70 myself, but maybe i can use your solution too