john30/ebusd-configuration #366: vaillant/ec.solsc.sc.csv does not fit my heating system SOL00, but all other .csvs fit well

dynahenry, 2023-11-17:
Liebe Gemeinschaft,
Ich betreibe hier eine Gas-Hybrid-Heizung von Vaillant mit
ecoTEC plus VC636/5, Regler multiMATIC 700, Speicher allSTORE exclusive, FrischwasserStation aguaFLOW exclusive VPM 40, 
SolarladeStation auroFLOW exclusive VPM 60/2, Solarmodul VR_71 und Internet Modul VR921.
Daran betreibe ich seit kurzem den **ebus-Adapter v5** mit einem Raspberry PI 3 mit dem ebusd Deamon.

Es werden mit **--scanconfig**  6 .csv Dateien erkannt und den Geräten zugeordnet:

address 00: master #1
address 01: master #6
address 03: master #11
address 05: slave #1, scanned "MF=Vaillant;ID=VR921;SW=2801;HW=5703"
address 06: slave #6, scanned "MF=Vaillant;ID=PMS02;SW=0210;HW=8402", loaded "vaillant/06.pms.csv"
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0404;HW=9602", loaded "vaillant/bai.0010015600.inc" ([HW=9602]), "vaillant/08.bai.csv"
address 0a: slave, scanned "MF=Vaillant;ID=PMW01;SW=0206;HW=8302", loaded "vaillant/0a.pmw.hwc.csv"
address 10: master #2
address 12: slave, scanned "MF=Vaillant;ID=PMW01;SW=0206;HW=8302"
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0614;HW=6903", loaded "vaillant/15.700.csv"
address 26: slave, scanned "MF=Vaillant;ID=VR_71;SW=0100;HW=5904", loaded "vaillant/26.vr_71.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
**address ec: slave, scanned "MF=Vaillant;ID=SOL00;SW=0614;HW=6903", loaded "vaillant/ec.sol.sc.csv"**
address ed: slave, scanned "MF=Vaillant;ID=PMS02;SW=0210;HW=8402"
address f7: master #20
address fc: slave #20, scanned "MF=Vaillant;ID=PMW01;SW=0206;HW=8302"

Das scheint alles ok zu sein, viele Werte werden im Betrieb vom Bus gelesen, alle erkannten Variablen werden mit 
ebusctl -f aufgelistet und viele davon kann ich mit **ebusctl -r xyz** vom Bus lesen.

Nur mit der Adresse **ec SOL00 vaillant/ec.sol.sc.csv** geht das nicht.
Keinen der Werte, die mit **ebusctl f -c sc -v** gelistet werden, kann ich auslesen.
z.B. **ebusctl r -c sc -f Col** ergibt  -> **ERR: invalid position in decode**
Bei allen andern Variablen von **sc** erhalte ich die gleiche Fehlermeldung.

Mit dem **ebusctl grap** Kommando erhalte ich viele "unknown MS oder BC cmds, unter Anderem diese:

01ecb5040110 / 0a0080e801008000800080 = 83
01ecb5040134 / 03000001 = 28
01ecb505084d01020000000000 / 00 = 1
01ecb505084d04020000000000 / 00 = 1
01ecb505084d06020000000000 / 00 = 1
01ecb505084d08020000000000 / 00 = 1
01ecb505084d09020000000000 / 00 = 1
01ecb505084d0c020000000000 / 00 = 1
01ecb505084d0e020000000000 / 00 = 1

Hier wird, so weit ich das verstehe, vom master auf adr 01 auf den slave adr ec zugegriffen. Das sollte SOL00 sein.
Es handelt sich also um ein Vaillant Gerät b5, jedoch wird hier auf **PBSB b504** und **b505** zugegriffen, welche beide nicht
in der configuration  ec.sol.sc.csv vorkommen. Folglich können diese Daten nicht dekodiert werden.
Somit ist die .csv für das vorliegende Gerät "SOL00;SW=0614;HW=6903" nicht geeignet. Es gibt dort nur **PBSB b509**.

An dieser Stelle komme ich nicht weiter. Vielleicht ist jemand schon auf das gleiche Problem gestoßen und kann helfen.
Mit dem Dekodieren tue ich mich schwer.

Ansonsten läuft das System ohne Probleme und ich kann viele Temperaturen und Zustände in HA abrufen und 
grafisch darstellen. Das wird bei der Optimierung der Anlage helfen.
Viele der Daten aus dem Solarmodul kann ich auch aus anderen Gerräten abrufen, jedoch nicht die Kollektor Temperetur
und die Erträge. Die Erträge bekomme ich aus der Vaillant App, dort muß ich sie jedoch abschreiben.😢 Sie sind aber 
offensichtlich irgendwo auf dem Bus vorhanden, sonst könnte Vaillant sie nicht mit dem VR921 täglich und stündlich abgreifen. 

An dieser Stelle möchte ich hier allen Beteiligten, vor Allem John30, meinen Dank aussprechen. Es ist eine sehr gute Arbeit.

Liebe Grüße dynahenry

chrizzzp, 2023-11-17:
Ich hatte das gleiche Problem, habe aber einen anderen Regler (VRC720). Ich lese die Werte des Solarkreises alle aus dem Regler aus. Könnte auch beim VRC700 (15.700.csv) funktionieren. Kollektorfühler ist bei mir auf S7 (Ertragsfühler S8) vom VR71 geklemmt. Das VR71 schickt die Sensorwerte eigentlich alle par Sekunden über den ebus.

```
*r,,,,,,B524,02000000,,,IGN:4,,,
*w,,,,,,B524,02010000,,,,,,
r;w,,SolarFlowRateQuantity,,,,,1900,,,EXP,,l/min,preset flow rate solar circuit for solar yield calculation

### Solar Circuit
*r,,,,,,B524,02000400,,,IGN:4,,,
*w,,,,,,B524,02010400,,,,,,
r,,SolarUnknown01,,,,,0100,,,HEX:*,,,
r;w,,SolarPumpKick,,,,,0200,,,onoff,,,
r,,SolarCollectorTemp,,,,,0300,,,tempv,,,
r;w,,SolarMinCollectorTemp,,,,,0400,,,tempv,,,
r;w,,SolarMaxCollectorTemp,,,,,0500,,,tempv,,,
r,,SolarUnknown06,,,,,0600,,,HEX:*,,,
r,,SolarYieldTemp,,,,,0700,,,tempv,,,
r,,SolarPump,,,,,0800,,,onoff,,,
r,,SolarPumpHours,,,,,0B00,,,hoursum,,,
### Solar Cylinder 1
*r,,,,,,B524,02000500,,,IGN:4,,,
*w,,,,,,B524,02010500,,,,,,
r;w,,SolarCyl1MaxTemp,,,,,0100,,,tempv,,,maximum cylinder charging temperature (BottomTemp sensor)
r;w,,SolarCyl1SwitchOnDifferential,,,,,0200,,,EXP,,K,charge if the temperature difference between the cylinder temperature sensor at the bottom and the collector temperature sensor is greater
r;w,,SolarCyl1SwitchOffDifferential,,,,,0300,,,EXP,,K,turn charge off if temperature difference between the bottom cylinder temperature sensor and the collector temperature sensor is smaller
r,,SolarCyl1BottomTemp,,,,,0400,,,tempv,,,bottom cylinder temperature sensor
```

Viel Erfolg!

dynahenry, 2023-11-17:
@chrizzzp 
Vielen Dank für die Antwort.
Bei der vorliegenden Anlage wurde die Konfiguration 1 und das Systemschema 6 angewendet. Damit ist der Kollektor-Fühler nicht am VR71 angeschlossen.
Dieser scheint nur am Solar Modul angeschlossen zu sein.
VR71 erfasst hier nur die Temperaturen des allSTORE und des Heizkreises.
Den Solar Ertrag kann ich nur am VRC700 abfragen, dort taucht er aber erst nach 
Mitternacht auf. Den Tages-Ertrag kann ich vom PMS abfragen.
Leider gibt es noch kein .csv für den VR921, womit Vaillant alle Werte abfragt.
Ich versuche weiterhin die unknown MS zu entschlüsseln.

dynahenry

chrizzzp, 2023-11-17:
Welche Werte willst Du denn abfragen?

Solarertrag lese ich auch aus dem Regler aus. Probier mal, ob es bei Dir nicht auch geht durch folgende Definition in der CSV vom Regler (15.700.csv):

*r,,,,,,B524,02000000,,,IGN:4,,,
*w,,,,,,B524,02010000,,,,,,
r,,SolarYieldTotal,total solar yield,,,,3D00,,,energy4,,,solar yield total

dynahenry, 2023-11-25:
@chrizzzp
Welche Werte ich sehen möchte?
Den Solar-Ertrag z.B. stündlich, z.Z. wird er nur nach Mitternacht aktualisiert. In der Vaillant-App ist er stündlich vorhanden.
Die Kollektor-Temperatur. Dabei bin ich mir nicht sicher, ob es oben einen Sensor gibt.
Die Solar Temperaturen kann ich jedoch auch in der SolarStation (pms) am Wärmetauscher messen. Das mache ich alle 15 Minuten mit dem cron (crontab) damit ich den Betrieb nicht zu sehr störe.
Mit der aktuellen Lösung bin ich zufrieden.
Den SOL00 habe ich entfernt, da keiner der Werte lesbar ist.

PeterBaumbach, 2024-01-18:
Ich habe das selbe Problem, bis auf die Solar Werte klappt alles perfekt. Ich bekomme ständig die Meldung decode failed, wrong Position. Nun würde ich das trotzdem sehr gerne auslesen .. gibt es jemand der mir sagen könnte wie ich es dekodiere oder wo ich suchen muss!!? Danke

gmx999, 2024-01-18:
Ich habe das selbe Problem, bis auf die Solar Werte klappt alles perfekt. Ich bekomme ständig die Meldung decode failed, wrong Position. Nun würde ich das trotzdem sehr gerne auslesen .. gibt es jemand der mir sagen könnte wie ich es dekodiere oder wo ich suchen muss!!? Danke

realdognose, 2024-03-12:
Same for me... 

Die Momentan-Produktion der Solar-Anlage bekomme ich über folgenden Wert. (Klingt nach Debug-Parameter, wurde wohl vergessen): 

MonitorMatlabLast5minSolAverage

Das ist so wie ich das Sehe die Durschschnitliche Solarleistung der letzten 5 Minuten. 

Allerdings hätte ich gerne noch die Kollekor-Temperatur, aber das funktioniert mit der ec.sol.sc.csv nicht ("Invalid Position in decode")

dynahenry, 2024-03-12:
@realdognose 
Danke, das werde ich mal ausprobieren.
Bisher lese ich die 4 Temperatursensoren am pms zwischen 10 und 18 Uhr alle 15 Minuten ein. Damit bekomme ich zwar nicht die Kollektor-Temperatur, aber ein wenig später die Vor-und Rücklauftemperatur zu den Kollektoren und Vor und Rücklauftemperatur vom Solar-Wärmetauscher zum Speicher.
Die Kollektor-Temperatur geht bei mir aus dem gleichen Grund nicht. (Invalid Position…)

Revilo91, 2024-04-16:
Hier hat jemand das selbe Problem gehabt, und hat sich in die Daten reingelesen:
#55 

Leider bekomme ich das nicht so richtig zum Laufen. Hab da meine Schwierigkeiten, dies umzusetzten. Vielleicht kann ja jemand hier helfen?

UnderPalms, 2026-10-28:
[image]

Moin zusammen,
zusammen mit Claude.ai bin ich auf eine Lösung gekommen. 
Und zwar: die Temperatur des Kollektorfühlers meiner Solarstation (und wahrscheinlich auch für andere Typen) wird über ein broadcast auf dem Bus zur Verfügung gestellt.  Bei mir über die ID ecb505084d.

Die KI ist folgendermaßen vorgegangen: Ich habe zu einem Zeitpunkt x die Kollektortemperatur am Display abgelesen. Claude hat mir folgendes Such-Skript geschrieben: 

cat > ~/suche.py <<'EOF'
import subprocess, sys
t = float(sys.argv[1])
tol = float(sys.argv[2]) if len(sys.argv) > 2 else 1.0
lo, hi = round((t - tol) * 16), round((t + tol) * 16)
out = subprocess.run(["ebusctl", "grab", "result", "all"],
                     capture_output=True, text=True).stdout
for line in out.splitlines():
    try:
        b = bytes.fromhex(line.split(" = ")[0].replace(" / ", ""))
    except ValueError:
        continue
    for i in range(len(b) - 1):
        v = int.from_bytes(b[i:i+2], "little", signed=True)
        if lo <= v <= hi:
            print(f"Byte {i}: {v/16:.2f} °C  <- {line}")
EOF

Dieses lässt sich dann aufrufen mit der Temperatur und einem Toleranzbereich, z.b. wie folgt:
python3 ~/suche.py [abgelesener Temperaturwert am Display] 1.5

Das gibt alle Telegramme aus, für die die Temperatur zwischen [abgelesener Temperaturwert am Display] + und - 1.5 Grad liegt.

Um diese Variable dauerhaft zu tracken und zu validieren, hat Claude eine neue Variable definiert:

1. Option prüfen

ebusd --help | grep -i -A1 define

Dort sollte die Option zum define-Befehl stehen (meines Wissens --enabledefine). Steht dort ein anderer Name, nimm diesen in Schritt 2.

2. Option in die Startoptionen eintragen

sudo sed -i '/^ *EBUSD_OPTS="/ s/"$/ --enabledefine"/' /etc/default/ebusd
grep EBUSD_OPTS /etc/default/ebusd

Die aktive Zeile (ohne #) sollte jetzt am Ende --enabledefine" haben.

3. Neu starten und prüfen

sudo systemctl restart ebusd
ps aux | grep [e]busd

In der Prozessliste muss --enabledefine auftauchen.

4. Definition setzen

ebusctl define 'u,solar,CollectorTemp,Kollektor-Kandidat,01,ec,b505,4d,temp,m,D2C,,°C,,ignore,m,IGN:5,,,'

5. Nach etwa zwei Minuten abrufen

ebusctl find -c solar -v
ebusctl read -c solar CollectorTemp

Gleich danach vergleichst du mit dem Journal:

sudo journalctl -u ebusd --since "3 min ago" | grep 01ecb505084d | tail -2

Den Hexwert (die vier Zeichen nach 4d) rechnest du wie vorher um: Bytes tauschen, in Dezimal wandeln, durch 16 teilen. Der Wert aus ebusctl sollte dazu passen.

6. Dauerhaft machen (erst wenn Schritt 5 passt): Das Skript und den Eintrag im Override aus meiner letzten Anleitung anlegen.

Schick mir die Ausgabe von Schritt 4 und 5, bevor du Schritt 6 machst. Falls define eine Fehlermeldung bringt, ist es meist die Option oder ein Tippfehler in der Zeile.

Im Anhang findet ihr einen Plot, in dem ich überprüft habe, dass es sich bei dem Telegramm wirklich um die Kollektortemperatur handelt. Der Graph sieht im Zusammenhang mit Solarvor- und Rücklauf absolut plausibel aus.

Gruß
Peter