john30/ebusd-configuration #378: Config DONE valiant + sensoconfort 380

1achy, 2023-12-23:
Hi all,

i have valliant ecoteck pro with vrt380.

caldaia

Equipment:
0010021995 - Caldaia Vaillant EcoTEC Pro a condensazione VMW 236/5-3
0020260943 - Termostato Caldaia Vaillant sensoHOME 380
 

 
dont recognize csv automatic way

Copy all file:

https://github.com/john30/ebusd-configuration/tree/master/ebusd-2.1.x/en/vaillant/

broadcast.csv
_templates.csv
scan.csv
08.bai.csv
... etc....

IN TO

home/pi/config

Rename https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/en/vaillant/15.700.csv
in to 15.ctlv0.csv

I dont know i need change some info in file
disable auto config scan and set --configpath=home/pi/config/valliant
Restart service

 
BUT SOME DATA NO AVAIABLE after I mod file for modified some id data:
 
 
FILE MOD:

[BACKUOP CARTELLA IN WWW ebusd_mod_ctlv2.zip](https://github.com/john30/ebusd/files/13759288/BACKUOP.CARTELLA.IN.WWW.ebusd_mod_ctlv2.zip)

NOW IT IS WORKING:

Uploading Schermata 2023-12-23 alle 15.08.19.png…

pattyland, 2025-11-24:
Will that solve #286? Would you be so kind as to provide your configuration as a PR so that it can be added to the repo?

BertVano, 2026-10-28:
Any update on this?