john30/ebusd-configuration #379: Demirdöküm Nitromix 24 - Same as Saunier Duval Semiatec Condense 24

bayc4d, 2023-12-28:
hi i success with setmodeoverride but i want to normal control way. i want to modulation. with  setmodeoverride, modulation not work so i want to make csv file for my boiler. what can i do?

2023-11-29 21:00:11.996 [update notice] sent scan-read scan.08 QQ=31: Vaillant;BAI00;0505;3302
2023-11-29 21:00:11.996 [bus notice] scan 08: ;Vaillant;BAI00;0505;3302
2023-11-29 21:00:12.007 [main notice] read common config file vaillant/general.csv
2023-11-29 21:00:12.008 [main notice] read common config file vaillant/broadcast.csv
2023-11-29 21:00:12.009 [main notice] read common config file vaillant/scan.csv
2023-11-29 21:00:12.011 [main notice] read scan config file vaillant/08.bai.csv for ID "bai00", SW0505, HW3302

pacman789, 2024-01-10:
Have you found/created config file for Saunier Duval?

bayc4d, 2024-01-13:
I'm doing research and trying to get information from here, but I can't fully understand it.