john30/ebusd-configuration #382: No config file for address 15 ID "bass2", SW0507, HW1704

romP44, 2024-01-07:
Hello,

I have a Saunier Duval boiler with wireless thermostat (ref 0020260979, model SRC 720f/2).
There is a "Base station" gateway for the wireless thermostat connected to the boiler (over ebus), ref 0020262149.
No config file is found for address 15.
Log details:
```
2024-01-07 14:53:14.092 [bus notice] scan 15: ;Vaillant;BASS2;0507;1704
2024-01-07 14:53:14.093 [update notice] store 15 ident: done
2024-01-07 14:53:14.093 [update notice] sent scan-read scan.15  QQ=31: Vaillant;BASS2;0507;1704
2024-01-07 14:53:14.093 [bus notice] scan 15: ;Vaillant;BASS2;0507;1704
2024-01-07 14:53:14.342 [update notice] sent unknown MS cmd: 3115b5090124 / 09003231323233363030
2024-01-07 14:53:14.592 [update notice] sent scan-read scan.15 id QQ=31: 
2024-01-07 14:53:14.852 [update notice] sent scan-read scan.15 id QQ=31: 
2024-01-07 14:53:15.102 [update notice] sent scan-read scan.15 id QQ=31: 21;22;36;0020262149;0082;011095;N5
2024-01-07 14:53:15.102 [bus notice] scan 15: ;21;22;36;0020262149;0082;011095;N5
2024-01-07 14:53:15.260 [main error] unable to load scan config 15: no file from vaillant with prefix 15 matches ID "bass2", SW0507, HW1704
2024-01-07 14:53:15.260 [main error] scan config 15: ERR: element not found
```

Please, can you implement it?

romP44, 2024-01-07:
I created a pull request for it: #383

john30, 2024-12-28:
solved with the merged PR

closed 2024-12-28 by john30