john30/ebusd-configuration #385: Need to change via ebusctl the "d.00 Heating partiak load" boiler menu setting

telece2, 2024-01-15:
Hello,

I would like to change the _d.00 Heating part load_ BAI setting via ebusctl. This setting is manageable from the boiler config menu, it is the first option. Possible values are: auto, 6, 7, ... ,30 (where 30 is the maximum heating power in kW, in my case).
Although this setting can be changed via the boiler menu, it is marked as read-only in the ebusd BAI config file.

[image]

My boiler specs:
- Vaillant VU INT II 306/5-5 ecoTec Plus 630 (30kW)
- Product number 0010011740
(NB: I needed to force ebusd to use the _bai.0010021961.inc_ file which is far more suitable for this boiler, instead of the fallback _bai.308523.inc_)

Apparently, I can write a new value, but it actually doesn't change when re-reading it:
```
ebusctl r -f -c bai PartloadHcKW
30

ebusctl w -c bai PartloadHcKW 6
done

 ebusctl r -f -c bai PartloadHcKW
30
```

```
admin@NAS2:~ # ebusctl scan result
08;Vaillant;BAI00;0609;5502;21;16;31;0010011740;0001;005691;N2
15;Vaillant;UI   ;0509;6201;21;15;27;0020080468;0907;006125;N9
18;Vaillant;V32;0117;9802
23;Vaillant;VR630;0500;6301;21;15;27;0020092440;0907;005308;N2
25;Vaillant;VR630;0500;6301;21;15;27;0020092440;0907;005308;N2
26;Vaillant;VR630;0500;6301;21;15;27;0020092440;0907;005308;N2
50;Vaillant;VR630;0500;6301;21;15;27;0020092440;0907;005308;N2
51;Vaillant;VR630;0500;6301;21;15;27;0020092440;0907;005308;N2
```

```
admin@NAS2:~ # ebusctl grab result all|grep PartloadHcKW
3108b509030d6c00 / 011e = 10: bai PartloadHcKW
3108b509040e6c0006 / 00 = 3: bai PartloadHcKW
```

Is there any way to change this setting via ebusctl?
I need to control this remotely, to limit the boiler consumption to 6kW during periods when the house is empty and I must maintain the temperature above 15-16 Celsius.
I only need potential full power (_Heating partial load_ set to auto) occasionally.
The boiler is operated via a Calormatic 630 controller.

thanks,
Cristian

HiQual, 2024-01-17:
May I suggest that you use setmodeoverride instead of trying to adjust partloadhckw.  This sets the boiler flow temp, and in conjunction with D00 set to auto on your boiler, you will find that after a learning period the boiler will use the lower modulation automatically.
Also you said you use a different boiler.inc file - did you achieve this just by using local configuration files and if so did you have to rename the local inc file?

telece2, 2024-01-18:
Thanks for the tip.

My hardware configuration is:
- 2 x cascaded boilers vaillant vu int II 306/5-5 630, 30kW each
- 1 x VR32 (to attach the second boiler on the same ebus line)
- 1 x Calormatic 630
- 1 x ebus adapter ethernet v3
- 1 x lowloss header with 3 floor pumps and 1 hwc 800L storage tank (all connected to the lowloss header).

(NB: as I currently don't know how to address devices behind a VR32, I am currently "playing around" only with the 1st boiler).

Regarding the boiler .inc file: I had to force ebusd to use local configuration files (--configpath=/ebusd/ebusd-configuration-master/ebusd-2.1.x/en):

```
  ebusd:
    restart: always
    image: "john30/ebusd:latest"
    container_name: "ebusd"
    network_mode: host
    volumes:
      - "/etc/localtime:/etc/localtime:ro"
      - "/share/3tb_1/docker_data/ebusd/ebusd-configuration-master:/ebusd/ebusd-configuration-master"
      - "${TZ_FILE}:${TZ_FILE}:ro"
    command: --scanconfig **--configpath=/ebusd/ebusd-configuration-master/ebusd-2.1.x/en** --accesslevel=* --device=enh:10.10.0.7:9999 --port=8888 --loglevel=error -f -l ebusd.log --pidfile=/var/run/ebusd1.pid --mqttretain --mqttjson --mqtttopic=ebusd1 --mqtthost=127.0.0.1 --mqttport=1883 --mqttuser=${MQTT_USER} --mqttpass=${MQTT_PASS} --mqttclientid=ebus_1_eth --latency=10
```

Then I edited the 08.bai.csv as follows and added this new line: `[HW=5502]!load,bai.0010021961.inc,,,`:
```
admin@NAS2:/share/3tb_1/docker_data/ebusd/ebusd-configuration-master/ebusd-2.1.x/en/vaillant # diff 08.bai.csv 08.bai.csv_bck_20240115
--- 08.bai.csv
+++ 08.bai.csv_bck_20240115
@@ -19,7 +19,6 @@
 [PROD='0010021961';'0010021962';'0010021963';'0010021964';'0010021965';'0010021966';'0010021967']!load,bai.0010021961.inc,,,
 [PROD='0010008045';'0010008863';'0010023648']!load,bai.0010008045.inc,,,
 #fallbacks:,,,,
-[HW=5502]!load,bai.0010021961.inc,,,
 [HW=6701]!load,bai.0010003886.inc,,,
 [HW=0902]!load,bai.0010004121.inc,,,
 [HW=8801]!load,bai.0010006101.inc,,,
```

The current configuration is:
```
admin@NAS2:~ # ebusctl scan result
done

admin@NAS2:~ # ebusctl scan full
08;Vaillant;BAI00;0609;5502;21;16;31;0010011740;0001;005691;N2
15;Vaillant;UI   ;0509;6201;21;15;27;0020080468;0907;006125;N9
18;Vaillant;V32;0117;9802
23;Vaillant;VR630;0500;6301
25;Vaillant;VR630;0500;6301;21;15;27;0020092440;0907;005308;N2
26;Vaillant;VR630;0500;6301;21;15;27;0020092440;0907;005308;N2
50;Vaillant;VR630;0500;6301;21;15;27;0020092440;0907;005308;N2
51;Vaillant;VR630;0500;6301;21;15;27;0020092440;0907;005308;N2
```

```
admin@NAS2:/share/3tb_1/docker_data/ebusd/ebusd-configuration-master/ebusd-2.1.x/en/vaillant # ebusctl info
version: ebusd 23.3.23.3
update check: OK, different configuration files available
device: 10.10.0.7:9999, TCP, enhanced
access: *
signal: acquired
symbol rate: 53
max symbol rate: 169
min arbitration micros: 2
max arbitration micros: 140
min symbol latency: 0
max symbol latency: 50
scan: finished, some messages pending
reconnects: 0
masters: 6
messages: 744
conditional: 15
poll: 1
update: 12
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0609;HW=5502", loaded "vaillant/bai.0010021961.inc" ([HW=5502]), "vaillant/08.bai.csv"
address 10: master #2
address 13: master #12
address 15: slave #2, scanned "MF=Vaillant;ID=UI   ;SW=0509;HW=6201", loaded "vaillant/15.ui.csv"
address 18: slave #12, scanned "MF=Vaillant;ID=V32;SW=0117;HW=9802"
address 23: slave, scanned "MF=Vaillant;ID=VR630;SW=0500;HW=6301", loaded "vaillant/23.vr630.cc.csv"
address 25: slave, scanned "MF=Vaillant;ID=VR630;SW=0500;HW=6301", loaded "vaillant/25.vr630.hwc.csv"
address 26: slave, scanned "MF=Vaillant;ID=VR630;SW=0500;HW=6301", loaded "vaillant/26.vr630.hc.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd, scanning
address 3f: master #23
address 50: slave, scanned "MF=Vaillant;ID=VR630;SW=0500;HW=6301", loaded "vaillant/50.vr630.mc.csv"
address 51: slave, scanned "MF=Vaillant;ID=VR630;SW=0500;HW=6301", loaded "vaillant/51.vr630.mc.3.csv"
address 7f: master #24
```

Returning to our topic here: I could not find setmodeoverride in the list of ebusd commands (based on the current config files):

`admin@NAS2:~ # ebusctl find -f|grep -i setmodeoverride
admin@NAS2:~ #`

Also, the parameter is not mentioned in any of the ebusd configuration files:
```
admin@NAS2:/share/3tb_1/docker_data/ebusd/ebusd-configuration-master # find . -type f|xargs grep -i setmodeoverride
admin@NAS2:/share/3tb_1/docker_data/ebusd/ebusd-configuration-master #
```

Can you kindly advise how you are using setmodeoverride?

To (partially) achieve my goal, I am currently limiting the heating pump power to 51% (minimal manual setting):
`r;wi,,PumpPowerDesired,d.14 Desired heating pump power,,,,A100,,,UCH,0=auto;1=53;2=60;3=70;4=85;5=100,%,PWM-Desired central heating pump power,,,`
but I would really like to force the boiler to stay permanently at the minimal heating power (6kW) or respectively at its minimal possible modulation.
(if you have a better approach, please share, thanks)

Thanks a lot,
Cristian

HiQual, 2024-01-18:
1. Please see my "Issue" #361 for the additional (new) line to be added to your boiler ".inc" file.
2. Please see "Issue" #242 for more details on "setmodeoverride"
3. Please note that for Vaillant boilers it is usually required to write the setmodeoverride command at least once every 10 minutes to keep ebusd in control of the boiler flowtempdesired.  (if you only send the setmodeoverride command once, it lasts for about 10 minutes before the boiler reverts back to its inbuilt settings/control.
4. Good luck.

telece2, 2024-01-18:
Thanks a lot!
I managed to add the SetModeOverride, but in a slightly different format - the one mentioned here: https://github.com/john30/ebusd-configuration/issues/170#issuecomment-588916628

When I run the command `ebusctl w -c bai SetModeOverride '0;34;-;-;0;0;0;0;0;0'`:
- Flame went from on to off (and again to on, in less than one minute)
- ModulationTempDesired went from 37 to 17 (passing through 0 when Flame was off)
- PumpPower remained at 15 (NB: during this test, PumpPowerDesired was set to 53 - I will set it to auto in the next set of tests).

I am now waiting 10 minutes :)

HiQual, 2024-01-18:
1. For what it is worth, I recall that previous posters had problems with the 10 field setmodeoverride containing "IGN#".  Quote ... "Key to resolve it was replacing ,,IGN:1,,,, with setmode1,,UCH,,,, and setmode2,,UCH,,,, correspondingly."
2. They instead use the one I pointed to with 12 fields in the command.  Sorry I cannot remember what the 12 fields are, just that field 2 is the flowtempdesired,  None of them are Modulation.
3. Your heating system is much more involved than mine so I cannot comment on your other items except to say that the VR630 and/or any other Vaillant ebus controllers/modules are likely to continuously override your ebusd commands - sorry.
4. One more thing you might like to take a look at is here .... https://forum.buildhub.org.uk/topic/31261-zombie-boiler/   Note it talks about a Glowworm boiler but they are almost the same as Vaillant.
Good luck.

JonesGW-MGD, 2024-02-22:
@telece2 for what it is worth, have you tried accessing the boiler behind the VR32?  I'm just curious!  I recently helped someone who had a Recovair that was behind a VR32 and they were able to finally access.  I know there are problems with latency etc. but it ended up being as 'simple' as identifying the VR32 slave address ( after setting --latency=100000 and restarting ) and then having your boiler config file present as 38.vr32.csv (or whatever address the VR32.  Maybe that would work for you - I have no idea!

telece2, 2025-08-14:
@JonesGW-MGD thanks a lot for your incredible discovery! I finally managed to fully control the 2nd BAI which is behind a VR32 coupler by simply creating a symlink with the v32 name (ln -s 08.bai.csv 18.V32.csv) in the local config subdir ebusd-configuration-master/ebusd-2.1.x/en/vaillant. Great work, really appreciate it!