john30/ebusd-configuration #387: Cannot write to energy counters: PrEnergyCountHwc1 PrEnergyCountHc1 PrEnergySumHwc1 PrEnergySumHc1

KrzysiekZZZ, 2024-01-21:
Hello,
Thank you for the great HW and software.
After 10y of using Vaillant boiler the energy registers reach its limit. However when I do try to reset them I've got:

# ebusctl read bai PrEnergyCountHwc1
ERR: element not found

The read is ok:
# ebusctl read  PrEnergyCountHwc1
14961994

Looking at the files loaded I have:
# ebusctl info
version: ebusd 23.3.23.3
update check: OK
device: 192.168.88.127:9999, TCP, enhanced, firmware 1.1[4106].1[4106]
signal: acquired
symbol rate: 42
max symbol rate: 84
min arbitration micros: 1
max arbitration micros: 54
min symbol latency: 7
max symbol latency: 69
scan: finished
reconnects: 0
masters: 3
messages: 476
conditional: 3
poll: 1
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0600;HW=5502", loaded "vaillant/**bai.308523.inc**", "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=47000;SW=0130;HW=6002", loaded "vaillant/15.470.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd

I've checked out the repo with definitions and in the bai.308523.inc file I've have:
r;wi,,PrEnergySumHc1,PrEnergySumCH1_DK,,,,"F500",,,ULG,,,Predictive Maintenance data
I assume that the 'wi' indicates that the register has also write property.

Do you have any idea why this happens and how to update above values? 

Issue is the same on docker instance and HA addone (which I assume is the same thing).

Cheers,
Chris

Pelikanoooo, 2024-01-22:
hello, I'm joining with a question, unfortunately I can't even read these values ​​since the last update :(

KrzysiekZZZ, 2024-01-22:
@Pelikanoooo  I'm sure the issue you have is different one :)
For the reference fix to my issue is that the type: "wi" means writable by installation which  I assume is done for security reasons.
To write to wi registers you have to run ebusd with enhanced access level:
     --accesslevel=install
     
Afterwards standard ebusclt write works