john30/ebusd-configuration #392: 15.ctlv2.csv missing?

hhw0088, 2024-02-05:
Hi.

"ebusctl info" reveals the following:

address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0901;HW=5103", loaded "vaillant/08.hmu.csv"
address 10: master #2
**address 15: slave #2, scanned "MF=Vaillant;ID=CTLV2;SW=0514;HW=1104"**
address 26: slave, scanned "MF=Vaillant;ID=VR_71;SW=0203;HW=5904", loaded "vaillant/26.vr_71.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0901;HW=5103"
address f1: master #10

It appears that for the CTLV2 at address 15, no csv-file is loaded. (See the line in bold letters.) I don't see a file named 15.ctlv2.csv in the repository, so it is not very much surprising. Is this file missing for a particular reason? (I've understood from some other posts that it used to be there.)

Does anybody have an idea how I can make an appropriate csv-file be loaded? 

I am really desparate. Any help would be very much appreciated.

KR
Hendrik

Koky05, 2024-02-05:
These file is located in pull request and wait for master marge.

hhw0088, 2024-02-05:
Can this file then be used only locally?

dialerp, 2024-02-08:
Is planning also the addition of a file for
"MF=Vaillant;ID=VR_92;SW=0514;HW=1204"?
Additionally, how can I enforce the use of a local CSV file?

mklooss, 2024-02-08:
https://github.com/john30/ebusd-configuration/pull/266

dialerp, 2024-02-09:
Thank you for the information. In that case, we eagerly await the implementation of this file.

DrFrankBV, 2024-02-11:
me to. please do the merge

closed 2024-02-11 by john30