john30/ebusd-configuration #395: Config and auto-detect for Vaillant EcoTec Pure VUW 236/7-2

voltlog, 2024-02-11:
Hi,
I'm using ebusd integration in HomeAssistant (ver 23.2.3) with a wifi ebus adapter connected over ENH mode to my Vaillant EcoTec Pure VUW 236/7-2. Logs below show a successful connection is established, a boiler is found on auto-scan and a config file is selected automatically.  I can see some data over MQTT but some parameters show as unavailable. For example I'd like to know when the heating circuit pump is running, but HcPumpMode shows as unavailable.

[image]

Based on my research so far it is my understanding that my particular boiler Vaillant EcoTec Pure VUW 236/7-2, discovered as SW0706, HW1303, product = 0010019976 is not actually contained in the existing config file at this path `ebusd-configuration/ebusd-2.1.x/en/vaillant` and so I suspect this could be the reason why some parameters are not showed correctly?

**Additionally I see some error messages of this form, are these also because of unsupported boiler? What can I do to fix this?:**
> 2024-02-11 11:58:25.212 [update error] unable to parse poll-read bai externalFlowTempDesired from 3108b509030d2500 / 00: ERR: invalid position
> 2024-02-11 11:58:26.902 [mqtt error] decode bai externalFlowTempDesired: ERR: invalid position

**Full log below:**

> 2024-02-11 11:56:54.642 [main info] configPath URL is valid
> 2024-02-11 11:56:54.643 [main notice] ebusd 23.2.p20230716 started with auto scan on device: 192.168.0.216:3335, enhanced
> 2024-02-11 11:56:54.643 [main info] loading configuration files from https://cfg.ebusd.eu/
> 2024-02-11 11:56:54.912 [main info] reading templates /
> 2024-02-11 11:56:55.112 [main info] read templates in /
> 2024-02-11 11:56:55.113 [main info] reading file broadcast.csv
> 2024-02-11 11:56:55.340 [main info] successfully read file broadcast.csv
> 2024-02-11 11:56:55.340 [main info] reading file memory.csv
> 2024-02-11 11:56:55.541 [main info] successfully read file memory.csv
> 2024-02-11 11:56:55.541 [main info] read config files, got 11 messages
> 2024-02-11 11:56:55.650 [bus notice] device status: resetting
> 2024-02-11 11:56:55.651 [bus notice] bus started with own address 31/36
> 2024-02-11 11:56:55.651 [main info] registering data handlers
> 2024-02-11 11:56:55.651 [mqtt info] mosquitto version 2.0.18 (compiled with 2.0.15)
> 2024-02-11 11:56:55.654 [bus notice] device status: reset
> 2024-02-11 11:56:55.658 [main info] registered data handlers
> 2024-02-11 11:56:55.687 [bus notice] signal acquired
> 2024-02-11 11:56:56.174 [mqtt notice] connection established
> 2024-02-11 11:57:05.660 [main debug] performing regular tasks
> 2024-02-11 11:57:24.277 [bus notice] new master 03, master count 2
> 2024-02-11 11:57:25.661 [main debug] performing regular tasks
> 2024-02-11 11:57:25.661 [bus info] scan 08 cmd: 3108070400
> 2024-02-11 11:57:25.725 [bus info] arbitration delay 17 - 17 micros
> 2024-02-11 11:57:25.738 [bus info] send/receive symbol latency 13 - 13 ms
> 2024-02-11 11:57:25.750 [bus info] send/receive symbol latency 11 - 13 ms
> 2024-02-11 11:57:25.759 [bus info] send/receive symbol latency 9 - 13 ms
> 2024-02-11 11:57:25.857 [update info] sent MS cmd: 3108070400 / 0ab5424149303007061303
> 2024-02-11 11:57:25.857 [bus notice] scan 08: ;Vaillant;BAI00;0706;1303
> 2024-02-11 11:57:25.857 [update notice] store 08 ident: done
> 2024-02-11 11:57:25.857 [update notice] sent scan-read scan.08  QQ=31: Vaillant;BAI00;0706;1303
> 2024-02-11 11:57:25.857 [bus notice] scan 08: ;Vaillant;BAI00;0706;1303
> 2024-02-11 11:57:26.071 [main debug] found 4 matching scan config files from vaillant with prefix 08: done
> 2024-02-11 11:57:26.072 [main info] reading templates vaillant
> 2024-02-11 11:57:26.282 [main info] read templates in vaillant
> 2024-02-11 11:57:26.805 [main notice] read common config file vaillant/scan.csv
> 2024-02-11 11:57:27.040 [main notice] read common config file vaillant/broadcast.csv
> 2024-02-11 11:57:27.295 [main notice] read common config file vaillant/general.csv
> 2024-02-11 11:57:27.491 [main notice] read scan config file vaillant/08.bai.csv for ID "bai00", SW0706, HW1303
> 2024-02-11 11:57:27.492 [bus info] send message: 3108b5090124
> 2024-02-11 11:57:27.556 [bus info] arbitration delay 17 - 22 micros
> 2024-02-11 11:57:27.576 [bus info] send/receive symbol latency 9 - 20 ms
> 2024-02-11 11:57:27.647 [bus info] send/receive symbol latency 9 - 26 ms
> 2024-02-11 11:57:27.705 [update info] sent MS cmd: 3108b5090124 / 09003231323033373030
> 2024-02-11 11:57:27.705 [update notice] sent scan-read scan.08 id QQ=31: 
> 2024-02-11 11:57:27.705 [bus info] send message: 3108b5090125
> 2024-02-11 11:57:27.824 [bus info] send/receive symbol latency 8 - 26 ms
> 2024-02-11 11:57:27.907 [update info] sent MS cmd: 3108b5090125 / 09313030313939373633
> 2024-02-11 11:57:27.908 [update notice] sent scan-read scan.08 id QQ=31: 
> 2024-02-11 11:57:27.908 [bus info] send message: 3108b5090126
> 2024-02-11 11:57:28.111 [update info] sent MS cmd: 3108b5090126 / 09313030303134313433
> 2024-02-11 11:57:28.111 [update notice] sent scan-read scan.08 id QQ=31: 
> 2024-02-11 11:57:28.111 [bus info] send message: 3108b5090127
> 2024-02-11 11:57:28.170 [bus info] arbitration delay 14 - 22 micros
> 2024-02-11 11:57:28.321 [update info] sent MS cmd: 3108b5090127 / 094e353c3c3c3c3c3c3c
> 2024-02-11 11:57:28.321 [update notice] sent scan-read scan.08 id QQ=31: 21;20;37;0010019976;3100;014143;N5
> 2024-02-11 11:57:28.940 [main notice] found messages: 224 (3 conditional on 28 conditions, 0 poll, 10 update)
> 2024-02-11 11:57:28.941 [bus info] scan 08 cmd: 3108b5090124
[... 30 lines cut ...]
> 2024-02-11 11:58:01.188 [update info] sent MS cmd: 3108b509030d8200 / 0107
> 2024-02-11 11:58:01.189 [update notice] sent poll-read bai CounterStartAttempts4 QQ=31: 7
> 2024-02-11 11:58:01.713 [main debug] performing regular tasks
> 2024-02-11 11:58:07.010 [bus info] poll cmd: 3108b509030d2000
> 2024-02-11 11:58:07.189 [update info] sent MS cmd: 3108b509030d2000 / 0100
> 2024-02-11 11:58:07.189 [update notice] sent poll-read bai DeactivationsTemplimiter QQ=31: 0
> 2024-02-11 11:58:11.714 [main debug] performing regular tasks
> 2024-02-11 11:58:13.009 [bus info] poll cmd: 3108b509030dda00
> 2024-02-11 11:58:13.096 [bus info] send/receive symbol latency 7 - 27 ms
> 2024-02-11 11:58:13.221 [update info] sent MS cmd: 3108b509030dda00 / 0102
> 2024-02-11 11:58:13.221 [update notice] sent poll-read bai DisplayMode QQ=31: 2
> 2024-02-11 11:58:19.039 [bus info] poll cmd: 3108b509030d6b00
> 2024-02-11 11:58:19.216 [update info] sent MS cmd: 3108b509030d6b00 / 03e3ffaa
> 2024-02-11 11:58:19.217 [update notice] sent poll-read bai expertlevel_ReturnTemp QQ=31: -1.81;cutoff
> 2024-02-11 11:58:21.714 [main debug] performing regular tasks
> 2024-02-11 11:58:25.038 [bus info] poll cmd: 3108b509030d2500
> 2024-02-11 11:58:25.211 [update info] sent MS cmd: 3108b509030d2500 / 00
> 2024-02-11 11:58:25.212 [update error] unable to parse poll-read bai externalFlowTempDesired from 3108b509030d2500 / 00: ERR: invalid position
> 2024-02-11 11:58:26.902 [mqtt error] decode bai externalFlowTempDesired: ERR: invalid position
> 2024-02-11 11:58:31.027 [bus info] poll cmd: 3108b509030d6e04
> 2024-02-11 11:58:31.213 [update info] sent MS cmd: 3108b509030d6e04 / 00
> 2024-02-11 11:58:31.213 [update error] unable to parse poll-read bai ExtFlowTempDesiredMin from 3108b509030d6e04 / 00: ERR: invalid position
> 2024-02-11 11:58:31.715 [main debug] performing regular tasks
> 2024-02-11 11:58:36.914 [mqtt error] decode bai ExtFlowTempDesiredMin: ERR: invalid position
> 2024-02-11 11:58:37.002 [bus info] poll cmd: 3108b509030d1b00

voltlog, 2024-02-15:
Thank you for pointing me in the right direction. I confirm that running those commands shows the default 'bai.308523.inc' being loaded. I can also read the pump state and flame status as indicated. I will start reading the discussion section to figure out if there is more I can do on my own.
[image]

JonesGW-MGD, 2024-02-15:
Great!  Good luck, and glad to be of help!  You should be able to use Flame and Pump State as sensors in HA

voltlog, 2024-02-16:
I have already started looking into [your write-up here](https://github.com/john30/ebusd/discussions/1177) which seems like a great resource!

alexit84, 2026-02-23:
Hello @JonesGW-MGD , 
I have the same Vaillant EcoTec Pure, could you suggest me how to get the Flame sensor into HA ? It shows correctly through ebusctl command.
Moreover, do you have any custom config file made for your boiler to share? As @voltlog, I have a lot of errors and missing sensors using the default bai.308523.inc

Thank you!