john30/ebusd-configuration #396: 08HMU CurrentCompressorUtil datatype

pulquero, 2024-02-12:
I think CurrentCompressorUtil datatype should be UIN,16. At least this works well for HMU00;0607;5103. The current value of D1B gives me some negative values. And UIN,10 that is suggested in other PRs gives me readings above 100%.

pulquero, 2024-02-12:
D2C also works fine.

kecksk0enig, 2024-11-08:
jepp, i recently observed strange values on CurrentCompressorUtil, changing the type to D2C solved the issue. now values match the liveMonitor

morpheus99999, 2024-11-15:
Hi
How can i change this. My loxberry plugin takes the csv from the github-Server.
My CurrentCompressorUtil is also negative.

Or can i change this in Loxone with an calculator?

kecksk0enig, 2024-11-15:
check out the ebusd-configuration repository -> change the files local-> tell your ebusd daemon (where ever it runs) to use the local ebusd-configuration  

- https://github.com/john30/ebusd/wiki/4.-Configuration#files
- https://github.com/john30/ebusd/wiki/2.-Run#message-configuration-options (the -c flag)

morpheus99999, 2024-11-22:
Ok, 
And why my ebusd dont read the Config files...

[image]
[image]