john30/ebusd-configuration #400: scanconfig picks wrong CSV

jonathanschneider, 2024-02-22:
Hello,
I have a Vaillant heater with the article number 0010009351. Looking at [08.bai.HW7401.csv](ebusd-2.x.x/de/vaillant/08.bai.HW7401.csv), I would assume that bai.0010003886.inc gets picked because of [line 8](https://github.com/john30/ebusd-configuration/blob/cf37c58578857a723a14871a50feb6c5bdd53cd8/ebusd-2.1.x/de/vaillant/08.bai.csv#L8).

Instead bai.0010006101.inc gets picked because of [line 12](https://github.com/john30/ebusd-configuration/blob/cf37c58578857a723a14871a50feb6c5bdd53cd8/ebusd-2.1.x/de/vaillant/08.bai.csv#L12).

1. How does ebusd determine the product and is there a way to check what product ebusd actually uses to pick the CSV in this file?
2. If I am missing definitions in the picked CSV, would it make more sense to add those definitions to that CSV or try to alter which CSV gets picked? bai.0010003886.inc includes the definitions I require.

Thanks!

JonesGW-MGD, 2024-02-23:
1. You can check the product ID by either checking you logs on start up, you should see something like:

```
2024-02-23 17:30:12.073 [update notice] sent scan-read scan.08 id QQ=31: 21;22;46;0010017064;1300;806453;N4
```

or you can run 

```
ebusctl scan 08
```

How you do that depends a bit on how you are running ebusd

2. I think if you have the configurations cloned locally it doesn't really matter which method you chose to help ebusd pick the 'richest' config file - but 'm a relative novice and others may disagree.  I guess you need to think about future proofing and how you might assimilate changes that might get added to the config files that might fill in more and more gaps for you over time?

jonathanschneider, 2024-02-23:
Thanks! That was a great hint because neither the logs nor the `ebusctl scan 08` output the product ID. All I get is `08;Vaillant;BAI00;0518;7401;;;;;;;` and that would explain why 08.bai.csv (I named the v2.x.x file but the link was correct) picks another inc.
Since I know my product ID I think I will add the definitions to that inc and manipulate 08.bai.csv to pick that.
Thanks again!

jonathanschneider, 2024-02-23:
OK, here's my attempt to fix the issue: #401

JonesGW-MGD, 2024-02-23:
Glad to help in a small way - all the best!

john30, 2024-12-24:
is it possible that your main circuit board was exchanged and that it never got a product id set in the first place @jonathanschneider ? that would explain it.
could you please post the output of "ebusctl scan result"?

jonathanschneider, 2024-12-24:
My main circuit board did get exchanged! Good catch. Unfortunately I don’t live in that house anymore. I guess this issue is closed then.

closed 2024-12-24 by jonathanschneider