john30/ebusd-configuration #405: After new release 08.hmu.csv, strange behavior heat pump Vaillant

mthomalla, 2024-03-01:
Hy,
after the new release of 08.hmu.csv, i noticed that my Vaillant VWL 75/5AS heat pump suddenly had very short cycles and also very short operation. I then stopped the HA addon Ebusd. Has anyone else had this experience, or does anyone know what's happening here? Before the release of new 08.hmu.csv, Ebusd ran without this behavior.

[ebusd-2.1.x/de/vaillant/08.hmu.csv](url)

I use Ebusd with the adapter [https://adapter.ebusd.eu/v5/](url) and the HA addon ebusd.
I only read the ebus, never write to it. Logging all sensors works great, the issu is the behavior of the heatpump as you can see in the scrennshot, between 19.00 and 20.30.

[image]

The configuration and the log when starting the ebusd addon is:

`s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
ebusd --foreground --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=addons --mqttpass=pi2gahqu6uquasee6oop8tha1re1eikeezah5jee8Sooxoo5phe6JioshiuShai6 --scanconfig --mqttjson --latency=10 --accesslevel="*" --pollinterval=5 --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqtttopic=Vaillant --device=ens:192.168.2.155:9999 --log=all:notice --log=main:notice --log=bus:notice
2024-03-01 07:51:04.592 [main notice] ebusd 23.2.p20230716 started with auto scan on device: 192.168.2.155:9999, enhanced
2024-03-01 07:51:05.172 [bus notice] device status: resetting
2024-03-01 07:51:05.173 [bus notice] bus started with own address 31/36
2024-03-01 07:51:05.194 [bus notice] signal acquired
2024-03-01 07:51:05.249 [mqtt notice] connection established`

JonesGW-MGD, 2024-03-01:
Hi there, you may wish to join the discussion at https://github.com/john30/ebusd/issues/1205 and https://github.com/john30/ebusd/discussions/1203 if you determine they are the same symptoms (which they appear to be), you may wish to join in with your particular configuration, what has changed recently etc.  What steps you've taken

mthomalla, 2024-03-01:
I updated the first post. Thanks fpr the mention. Now i know, that there are more user with almost same issue.

john30, 2024-03-01:
see https://github.com/john30/ebusd/issues/1205#issuecomment-1973714243

mthomalla, 2024-03-01:
Hello John, 
after reading your comment, I set the Ebusd integration to autostart and completely restarted Homeassistant. The reverted 08.hmu.csv was loaded automatically, as well as other csv for my system. The corresponding data is received correctly. 

The heat pump has been working absolutely normally since then!

If the developers require further data from and for my Vaillant Arotherm Split system, I am ready to support them.
Thank you very much for this great project.
Best regards Marko

jkunczik, 2024-03-04:
An update from john30/ebusd/issues/1205: We found the culprit. All Home Assistant integrations (MQTT and ebusdpy) actively poll read values. Either the HMU or the controller doesn't take kindly if one of the introduced read values in the updated 08.hmu.csv (we still don't know which) is actively polled by another master, which then leads to a full stop of the heat pump. An issue was opened at ebuspy CrazYoshi/ebusdpy/#10 (for the core integration) and a pull request was opened john30/ebusd#1213 to prevent MQTT from actively polling.

mthomalla, 2024-03-04:
Thanks for the information. Thank you that there are users who really delve deeply into a topic like this.

closed 2024-04-19 by mthomalla