john30/ebusd-configuration #409: Vaillant ecoVIT exclusiv VKK 0010007510

jabdoa2, 2024-03-20:
I got a Vaillant ecoVIT exclusiv VKK (which reports as `"MF=Vaillant;ID=BAI00;SW=0703;HW=7401", loaded "vaillant/bai.0010007508.inc" ([PROD='0010007510']), "vaillant/08.bai.csv"`). Most of the stuff works out of the box (including `SetMode`).  My model does not have warm water stuff connected but it should support that. I noticed a few things which are not working as expected:

```
# ebusctl find -r | grep ERR
bai actoSTOREEPreventiveCounter =  (ERR: invalid position for 3108b509030dec00 / 00)
bai DCFTimeDate = 09:10:14;17. (ERR: argument value out of valid range for 3108b509030de500 / 07141009170c0507)
bai ExtStorageModulCon =  (ERR: invalid position for 3108b509030dbf00 / 00)
bai FlowSetPotmeter =  (ERR: invalid position for 3108b509030d1900 / 00)
bai PrEnergySumHc1 =  (ERR: invalid position for 3108b509030df400 / 020000)
bai PumpHwcFlowNumber =  (ERR: invalid position for 3108b509030dc200 / 00)
bai PumpHwcFlowSum =  (ERR: invalid position for 3108b509030dc100 / 00)
bai SHEMaxDeltaHwcFlow =  (ERR: invalid position for 3108b509030dc400 / 00)
bai SHEMaxFlowTemp =  (ERR: invalid position for 3108b509030dc300 / 00)
bai SRTFINSDFeedback =  (ERR: invalid position for 3108b509030d8400 / 00)
bai Status16 =  (ERR: invalid position for 3108b5040116 / 00)
bai Status =  (ERR: invalid position for 3108b5110103 / 00)
```

`FlowSetPotmeter` is actually at `f003` similar to some other models but unlike 0010007508 (ecoVit pneumatic/4) which was loaded here. This works:
```
# ebusctl read -m 1 -def 'r,bai,FlowSetPotmeter,Vorlaufsollwert Poti,,08,b509,0df003,temp,s,D2C,,°C,Vorlaufsollwert Potentiometer'
35.19
```
`DCFTimeDate` is probably broken since my heater does not have a DCF antenna (or maybe even receiver) connected. Bus transfer looks like this:
```
[update error] unable to parse read bai DCFTimeDate from 3108b509030de500 / 07261c09170c0507: ERR: argument value out of valid range
[main error] read bai DCFTimeDate: decode ERR: argument value out of valid range
[bus notice] >3108b509030de500e0<0007261c09170c050730>00
```

For most of the other values the return is usually just 0s. For instance:
```
[update error] unable to parse read bai Status16 from 3108b5040116 / 00: ERR: invalid position
[main error] read bai Status16: decode ERR: invalid position
[bus notice] >3108b504011684<000000>00
```

Maybe responses look different if you do not operate the warm water part of the heater? Not sure.

I guess that it would be incorrect to change this in `vaillant/bai.0010007508.inc` as that would probably break those values for users of that device. What would be the best way here?

- Duplicate the file as `vaillant/bai.0010007510.inc`?
- Or is there a way to conditionally change some values for this particular model?
- Should I remove broken values or is there a way to find another address/interpretation?

jabdoa2, 2024-03-21:
I investigated some more and it turns out that all `PrEnergySumHc` value are at the wrong offset. This is the correct config:
```
r;wi,,PrEnergySumHc1,PrEnergySumCH1_DK,,,,"F500",,,ULG,,,Wartungsdaten
r;wi,,PrEnergyCountHc1,PrEnergyCountCH1_DK,,,,"F600",,,ULG,,,Wartungsdaten
r;wi,,PrEnergySumHc2,PrEnergySumCH2_DK,,,,"F700",,,ULG,,,Wartungsdaten
r;wi,,PrEnergyCountHc2,PrEnergyCountCH2_DK,,,,"F800",,,ULG,,,Wartungsdaten
r;wi,,PrEnergySumHc3,PrEnergySumCH3_DK,,,,"F900",,,ULG,,,Wartungsdaten
r;wi,,PrEnergyCountHc3,PrEnergyCountCH3_DK,,,,"FA00",,,ULG,,,Wartungsdaten
```
Those are also writable (unlike in `vaillant/bai.0010007508.inc`).

There seem to be other vaillant files which contain both changes (FlowSetPotmeter and PrEnergySumHcX). Should I map 0010007510 to one of those? Or create a new file?

john30, 2024-12-27:
solved with reworked+merged PR #416

closed 2024-12-27 by john30