john30/ebusd-configuration #414: Saunier Duval Semiatek Condens 24 still not supported?

mefisto22, 2024-03-30:
I only want to set FlowTempDesired, nothing else, but i cant do anything, only read some sensors. I tried as a HomeAssistant addon, and as a standalone docker container too.

root@017394b08039:/# ebusctl scan result
08;Vaillant;BAI00;0505;3302

root@017394b08039:/# ebusctl info
version: ebusd 23.3.23.3
device: 192.168.0.13:9999, TCP, enhanced, firmware 1.1[7f16].2[73c8]
signal: acquired
symbol rate: 24
max symbol rate: 54
min arbitration micros: 3
max arbitration micros: 53
min symbol latency: 7
max symbol latency: 48
scan: finished
reconnects: 0
masters: 2
messages: 1
conditional: 0
poll: 0
update: 0
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0505;HW=3302"
address 31: master #8, ebusd
address 36: slave #8, ebusd, scanning

root@017394b08039:/# ebusctl f -a -l "*" -w
ERR: element not found

jabdoa2, 2024-04-14:
If your device is not defined you have to create your own definition file (i.e. reverse engineer those values). It usually helps to look at similar devices and read the same addresses or sniff official components.

For FlowTempDesired there is no way to set that directly in bai. You have to use `SetMode`. This is documented on the ebus website (a bit hard to find). With set you can emulate an external controller. You have to repeat the message every 3-5 min or the heater will fall back to its internal value.

mefisto22, 2024-04-15:
> If your device is not defined you have to create your own definition file (i.e. reverse engineer those values). It usually helps to look at similar devices and read the same addresses or sniff [...]

please send me a link