john30/ebusd-configuration #415: No definitions for VWF flexoTHERM exclusiv 87/4 / VWL arcoCOLLECT 87/4?

Timon321, 2024-04-01:
I have this combination of Hardware/Software:

```
VWF flexoTHERM exclusiv 87/4
VWL aroCOLLECT 87/4
Gateway VR921
Adaptershield v5, Firmware 20240330
ebusd Version 23.3
```

I run it with this config:
`sudo ebusd --scanconfig=full --configpath=/etc/ebusd/ -d ens:192.168.4.190:9999 -f --latency 20000 --accesslevel=*`

In the ebusd are the latest csv-Files avalaible from [ebusd-2.1.x](https://github.com/john30/ebusd-configuration/tree/master/ebusd-2.1.x)
There are no readings for many values of my heatpump.

```
user@debian-ssh2:~$ ebusctl i
version: ebusd 23.3.23.3
update check: OK
device: 192.168.4.190:9999, TCP, enhanced, firmware 1.1[431e].1[431e]
access: *
signal: acquired
symbol rate: 19
max symbol rate: 160
min arbitration micros: 2
max arbitration micros: 51
min symbol latency: 5
max symbol latency: 39
scan: finished
reconnects: 0
masters: 5
messages: 660
conditional: 49
poll: 0
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0303;HW=0403", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0209;HW=4103", loaded "vaillant/15.700.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd, scanning
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZ00;SW=0303;HW=0403"
address e0: slave, scanned "MF=Vaillant;ID=OMU00;SW=0131;HW=5202", loaded "vaillant/e0.omu.csv"
address ec: slave, scanned "MF=Vaillant;ID=70000;SW=0209;HW=4103"
address f1: master #10
address f6: slave #10, scanned "MF=Vaillant;ID=NETX2;SW=4034;HW=5703"
```

```
user@debian-ssh2:~$ ebusctl scan result
08;Vaillant;HMU00;0303;0403
15;Vaillant;70000;0209;4103;21;16;01;0020171314;0082;005092;N6
76;Vaillant;VWZ00;0303;0403
e0;Vaillant;OMU00;0131;5202;??;??;??;??????????;????;??????;??
ec;Vaillant;70000;0209;4103;21;16;01;0020171314;0082;005092;N6
```

```
user@debian-ssh2:~$ ebusctl grab result
1008b5110100 / 0978010f000008000100 = 11
1076b5110101 / 09ff2f6008ff730000ff = 63
10feb505025c00 = 11
f108b5090124 / 09003231313631353030 = 1
f108b5090125 / 09313030313634323130 = 1
f108b5090126 / 09303036303035313631 = 1
f108b5090127 / 094e303c3c3c3c3c3c3c = 1
f108b5110100 / 0978010f000008000100 = 3
1008b50702092a / 025902 = 11
1008b508020900 / 00 = 1
1008b512020464 / 0101 = 11
1008b513020528 / 0101 = 1
f108b503020002 / 0affffffffffffffffffff = 10
f108b503020003 / 0a6500ffffffffffffffff = 3
f108b503020004 / 0affffffffffffffffffff = 11
f108b503020005 / 00 = 8
f108b509022802 / 0c02000303300100484d553030 = 1
f115b503020002 / 0affffffffffffffffffff = 11
f115b531020300 / 00 = 1
1008b5100305ff03 / 00 = 11
1076b512030f0201 / 079b030000800f03 = 62
1076b51303040d00 / 02ffff = 11
f108b5310301ffff / 00 = 1
f115b52403000000 / 040000803f = 1
f115b52403000100 / 040000803f = 1
f115b52403000200 / 0400000000 = 1
f115b52403000300 / 0400000000 = 1
f115b52403000400 / 0400004040 = 1
f115b52403000800 / 0400000000 = 1
f115b52403000900 / 0400000000 = 1
f115b52403000a00 / 0400000000 = 1
f115b52403000b00 / 0400000000 = 1
f115b52403000c00 / 04ffffff7f = 1
f115b52403000d00 / 04ffffff7f = 1
f115b52403000e00 / 04ffffff7f = 1
f115b52403000f00 / 04ffffff7f = 1
f115b52403001100 / 04ffffff7f = 1
f115b52403001200 / 04ffffff7f = 1
f176b5310301ffff / 00 = 1
1076b51009000000ffffff050000 / 0101 = 62
f108b50905540200180b / 060201180b2400 = 6
f115b52406020002020600 / 050202060000 = 3
f115b52406020003021c00 / 0500031c00ff = 2
f115b52406020002031e00 / 0600021e000000 = 18
f115b52406020003031c00 / 0500031c00ff = 3
f115b52406020000004800 / 06010048000100 = 20
f115b52406020001000f00 / 0601010f000000 = 20
f115b52406020002000600 / 050302060000 = 3
f115b52406020003001b00 / 0601031b000000 = 23
f115b52406020002010600 / 050202060000 = 3
f115b52406020003011c00 / 0500031c00ff = 2
f115b52406020002060200 / 00 = 1
f115b52406020003061c00 / 0500031c0000 = 3
f115b52406020002070200 / 00 = 1
f115b52406020003071c00 / 0500031c00ff = 3
f115b524050101ff0400 / 0f01040000000c4200008c420000803f = 1
f115b52406020002040200 / 00 = 1
f115b52406020003041c00 / 0500031c00ff = 3
f115b52406020002050200 / 00 = 1
f115b52406020003051c00 / 0500031c00ff = 3
f115b52406060009010700 / 0800090700ffffff7f = 1
f115b52406020002080200 / 00 = 1
f115b52406020003081c00 / 0500031c00ff = 3
03e0b5210900002100e4ffff4600 / 0b0002610072000700000800 = 26
```

Here is some output from the ebusd:
```
2024-04-01 22:07:30.939 [update notice] received read 700 z1DayTemp QQ=f1: 21
2024-04-01 22:07:31.035 [update notice] received read hmu Status01 QQ=10: 23.5;23.5;8.375;-;57.5;off
2024-04-01 22:07:31.181 [update notice] received read 700 currenterror QQ=f1: -;-;-;-;-
2024-04-01 22:07:31.361 [update notice] received unknown MS cmd: 1076b5110101 / 09ff2f6008ff730000ff
2024-04-01 22:07:31.507 [update notice] received unknown MS cmd: f115b503020002 / 0affffffffffffffffffff
2024-04-01 22:07:31.692 [update notice] received unknown MS cmd: 1076b512030f0201 / 079b030000800f03
2024-04-01 22:07:31.841 [update notice] received read hmu currenterror QQ=f1: -;-;-;-;-
2024-04-01 22:07:32.026 [update notice] received update-write hmu SetMode QQ=10: auto;21.0;-;-;0;1;1;0;0;0
2024-04-01 22:07:32.172 [update notice] received unknown MS cmd: f108b503020002 / 0affffffffffffffffffff
2024-04-01 22:07:32.357 [update notice] received unknown MS cmd: 1076b51009000000ffffff050000 / 0101
2024-04-01 22:07:32.502 [update notice] received unknown MS cmd: f108b503020004 / 0affffffffffffffffffff
2024-04-01 22:07:32.653 [update notice] received unknown MS cmd: f108b503020005 / 00
2024-04-01 22:07:32.849 [update notice] received read 700 z1ActualRoomTempDesired QQ=f1: 15
2024-04-01 22:07:33.036 [update notice] received read 700 z1OpMode QQ=f1: auto
2024-04-01 22:07:33.222 [update notice] received unknown MS cmd: f115b52406020003001b00 / 0601031b000000
2024-04-01 22:07:33.409 [update notice] received unknown MS cmd: f115b52406020001000f00 / 0601010f000000
```

TeXiCiTy, 2024-04-02:
08;Vaillant;HMU00;0303;0403 --> FlexoTherm main print
15;Vaillant;70000;0209;4103;21;16;01;0020171314;0082;005092;N6 --> Thermostat
76;Vaillant;VWZ00;0303;0403 --> FlexoTherm extra input/output print
e0;Vaillant;OMU00;0131;5202;??;??;??;??????????;????;??????;?? --> Outside unit aroCollect

There's some info on the outside unit [here](https://github.com/john30/ebusd/issues/1061).
[08.hmu.csv](https://github.com/john30/ebusd-configuration/files/14836445/08.hmu.csv)
This csv works with a lot of the geoTherm/flexoTherm sensors.

Timon321, 2024-04-02:
I have replaced my 08.hmu.csv from [ebusd-2.1.x](https://github.com/john30/ebusd-configuration/tree/master/ebusd-2.1.x) with your 08.hmu.csv. Now there are some new lines:

[image]

But the other devices (for example aroCollect) are still not recognized.

TeXiCiTy, 2024-04-03:
Try these for the e0.omu
```
# B521 service commands
*r,,,,,,"B521","0D",,,,,,
*w,,,,,,"B521","0E",,,,,,
[SW<131]w,,OMUControl,Control command from HMU to OMU,03,E0,"B521","00",NewStatus,m,appliancecode,,,Neuer Statuscode,unknown1,m,UCH,,,meistens 07 oder 08,reserved1,m,UCH,,,immer 00,reserved2,m,UCH,,,immer E7,Status,s,appliancecode,,,Aktueller Statuscode,LastStatus,s,appliancecode,,,Vorheriger Statuscode,TempBrineOutlet,s,D2C,16,°C,Temperatur Sole Rücklauf,TempAirInlet,s,D2C,16,°C,Temperatur Luft Einlass,unknown2,s,UCH,,,unbekannt,FanSpeed,s,UCH,-10,/min,Lüfterdrehzahl,FanPower,s,UCH,,%,Lüfteransteuerung
[SW>=131]w,,OMUControl,Control command from HMU to OMU,03,E0,"B521","00",NewStatus,m,appliancecode,,,Neuer Statuscode,reserved1,m,UCH,,,immer 22,reserved2,m,UCH,,,immer 00,reserved3,m,UCH,,,immer E6,reserved4,m,SCH,,,immer FF,reserved5,m,SCH,,,immer FF,reserved6,m,UCH,,,immer 46,reserved7,m,UCH,,,immer 00,Status,s,appliancecode,,,Aktueller Statuscode,LastStatus,s,appliancecode,,,Vorheriger Statuscode,TempBrineOutlet,s,D2C,16,°C,Temperatur Sole Rücklauf,TempAirInlet,s,D2C,16,°C,Temperatur Luft Einlass,unknown1,s,UCH,,,unbekannt,FanSpeed,s,UCH,-10,/min,Lüfterdrehzahl,FanPower,s,UCH,,%,Lüfteransteuerung,unknown2,s,UCH,,,unbekannt,reserved5,s,UCH,,,immer 00
```

Styx85, 2026-11-13:
Some findings for the **flexoTHERM exclusive VWF 87/4** (part no. 0010016421), in case they help with proper definitions.

**Setup:** ebusd 26.1.26.1, eBUS Adapter Shield C6 (fw 1.1, `ens:`), VRC 720 (CTLV2 SW 0514), VR 70.
Scan: `08;Vaillant;HMU00;0308;0403;21;21;36;0010016421;0001;…` → loads `vaillant/08.hmu.csv`.

**`RunStatsCompressorStarts` / `RunStatsCompressorHours` (B509 540200c3/c20b) are not counters on this unit.**

| message | request | response | decoded |
|---|---|---|---|
| RunStatsCompressorStarts | `08 b509 05 540200c30b` | `060201c30bae01` | 430 |
| RunStatsCompressorHours | `08 b509 05 540200c20b` | `060201c20bf801` | 504 h |
| CompressorStarts (r5) | `08 b51a 04 05ff3401` | `0aff08281d670000000000` | 26397 |
| CompressorHours (r5) | `08 b51a 04 05ff3400` | `0aff0827f9230000000000` | 9209 h |

Over several days (polled, recorded in Home Assistant) the RunStats values move up *and down*:
- `RunStatsCompressorStarts` e.g. 579 → 436 within one evening, decreasing by a few every 4–7 min, then rising again.
- `RunStatsCompressorHours` jumps in steps of 8 between ~464 and ~536.

Meanwhile `CompressorStarts` / `CompressorHours` (B51A) increase monotonically and plausibly (~1 start per hour currently), so those are the real counters on this unit. With `state_class: total_increasing` the RunStats entities produce statistics warnings in HA. I'm not sure what registers c2/c3 hold on the flexoTHERM (they look like slowly varying values rather than counters), so maybe these two should be excluded for HMU00 SW 0308, or documented as aroTHERM-only.

**`RunStatsCompressorHc` / `RunStatsCompressorHwc` (B511 1801/1802)**: the unit answers with an empty response (`08 b511 02 1801` → `00`), so ebusd reports `ERR: invalid position in decode`. Probably not supported on this unit.