john30/ebusd-configuration #417: Reading room humidity for VRC700 (register found, but in the "unusual" location)

stadid, 2024-05-08:
Dear All,
Analyzing traffic of gateway<->VRC700 while using MyVaillant app in the system with the singe heating circuit it was finally possible to find register which is responsible for the room humidity:

**You need to add the following to the 15.700.csv file right before #includes section i.e.**

add
```
# ##### Room humidity #####,,,,,,,,,,,,,
*r,,,,,,B524,06000901,,,IGN:4,,,
r,,RoomHumidity,room humidity shown by regulator,,,,0700,,,exp,,,room humidity shown by regulator
```

before the following lines 
```
# includes,,,,,,,,,,,,,
!include,errors.inc,,,,,,,,,,,,
```

**Please check and confirm if you can read humidity on your VRC700 using this config.**

### What is unusual

The strange thing is that this register location **06000901** not present in the current 15.700 nor 15.ctlv2 configs,
moreover in VRC720 room humidity register is located in the correct Zone section (see below).

**extract from 15.ctlv2.csv**

```
# ##### zone 1 #####,,,,,,,,,,,,,
*r,,,,,,B524,02000300,,,IGN:4,,,
*w,,,,,,B524,02010300,,,,,,

r,,z1RoomHumidity,,,,,2800,,,EXP,,%,room humidity zone 1 

```
which is correct since any of the VR92 could read and report humidity.

Similar behavior should be observed for VRC700 (as they are very close in config structure) but instead we have completely different address.

Anyway this behavior should be checked on the system with several heating zones and additional VR91 unit connected and mapped to the corresponding zone. 

**If somebody has system with VRC700+VR91 and using ebusd and MyVaillant app - could you check the following:**
1) could you read humidity in the app for the heating zone mapped with VR91 unit ?
2) could you check the traffic and confirm which register is used by gateway for reading humidity.

wurmthomas80, 2024-07-30:
Hi, 
I'm using this register exactly like you described for a whild now and it works perfectly!
Thanks for investigating and sharing with us!

Harted, 2024-08-06:
I can confirm this works for VRC700 

```sh
#info
...
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0614;HW=6903", loaded "vaillant/15.700.csv"
...
```
Connected via tcp...

Define 

```sh
define -r 'r,extra,z2Humidity,humidity zone 2,,15,b524,060009010700,,s,IGN:4,,,,,,exp,,%,current room humidity in zone 2'
```

Read

```sh
r z2Humidity
62
```

Works :)

### VR91

I have one

```sh
#info
...
address 35: slave #3, scanned "MF=Vaillant;ID=VR_91;SW=0415;HW=4803"
...
```

So, I thought...

```sh
define -r 'r,extra,z1Humidity,humidity zone 1,,35,b524,060009010700,,s,IGN:4,,,,,,exp,,%,current room humidity in zone 1'
```
But no...
```sh
r z1Humidity
ERR: invalid position in decode
```

> could you check the traffic and confirm which register is used by gateway for reading humidity.

How do I check the the traffic?

stadid, 2024-08-07:
> I can confirm this works for VRC700
> 
> ```shell
> #info
> ...
> address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0614;HW=6903", loaded "vaillant/15.700.csv"
> ...
> ```
> 
> Connected via tcp...
> 
> Define
> 
> ```shell
> define -r 'r,extra,z2Humidity,humidity zone 2,,15,b524,060009010700,,s,IGN:4,,,,,,exp,,%,current room humidity in zone 2'
> ```
> 
> Read
> 
> ```shell
> r z2Humidity
> 62
> ```
> 
> Works :)
> ### VR91
> 
> I have one
> 
> ```shell
> #info
> ...
> address 35: slave #3, scanned "MF=Vaillant;ID=VR_91;SW=0415;HW=4803"
> ...
> ```
> 
> So, I thought...
> 
> ```shell
> define -r 'r,extra,z1Humidity,humidity zone 1,,35,b524,060009010700,,s,IGN:4,,,,,,exp,,%,current room humidity in zone 1'
> ```
> 
> But no...
> 
> ```shell
> r z1Humidity
> ERR: invalid position in decode
> ```
> 
> > could you check the traffic and confirm which register is used by gateway for reading humidity.
> 
> How do I check the the traffic?

I believe your approach to link humidity to zone 2 is "logically incorrect". 
What I mean: VRC700 series regulator has humidity sensor which register in the ebusd configuration is not linked to any particular zone.
Native app (MyVaillant) checks to which zone VRC700 is mapped and automatically shows humidity for this specific zone.
In my current understanding, you could not read from VRC700 humidity of the zones mapped to VR91 units. (If you have MyVaillant app, you could confirm this by checking zone data - only zone mapped to VR700 will show humidity while zones mapped to VR91 shows only temperature). 

For VRC720 situation is different (also my assumption) - in 720 the main unit by himself reads humidity of the zones mapped to VR92 units and it could be read from VRC720 directly by using corresponding zone register.

So in short - If in the native app humidity is not shown for the zones mapped to VR91 - you'll be unable to read this parameter from VRC700. Maybe there is another approach to read humidity directly from VR91 unit, but I ado not have info how to perform this.