john30/ebusd-configuration #418: Missing configuration files

TomStarren, 2024-05-17:
Hello, 

I recently got a Vaillant heatpump and installed a eBus to Wifi adapter. With the config available only 2 of the 5 devices on the eBus are found.

address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0902;HW=5103", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=CTLV3;SW=0709;HW=3704"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 52: slave, scanned "MF=Vaillant;ID=VR_70;SW=0109;HW=2903", loaded "vaillant/52.vr_70.csv"
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0202;HW=0103"
address f1: master #10
address f6: slave #10, scanned "MF=Vaillant;ID=NETX2;SW=4034;HW=5703"

I got the CTLV3 working bij renaming the CTLV2 to CTLV3 but I don't know if everything is then available. I also see poll errors in the logs, is this because the heatpump isn't currently on and therefore the information is not available?

chillout2k, 2024-09-11:
> address 15: slave #2, scanned "MF=Vaillant;ID=CTLV3;SW=0709;HW=3704"

Copying from `15.ctlv2.csv` to `15.ctlv3.csv` worked for me out too :-)  Now, I´m able to read many for me relevant metrics (like hot water buffer).

But, does anybody know which (relevant/usable) metrics get published by the remaining two ebus masters?
* "MF=Vaillant;ID=VWZIO;SW=0902;HW=5103"
* "MF=Vaillant;ID=NETX3;SW=0121;HW=0404"

john30, 2024-12-29:
please check the [next](https://ebus.github.io/next/) generated csvs with ebusd 24.1 (or current source) as described there or by using a local clone of https://github.com/eBUS/ebus.github.io and using the "next" folder in there

topa-LE, 2025-01-03:
Hi John, thanks for the support. I have been using the clone from next for days.

Unfortunately, there is no support for the master:
`address f6: slave #10, scanned “MF=Vaillant;ID=NETX3;SW=0124;HW=0404”`

no CSV is loaded. I am missing some metrics from the installation.

Regards

fabian1512, 2025-05-17:
same for me

habo20, 2026-01-11:
is there e new csv for vaillant?
`ebusctl info
version: ebusd 25.1.25.1
update check: OK, device firmware v5-c6 1[6104] available, vaillant/0a.pmw.hwc.csv: different version available, vaillant/15.ctlv3.csv: different version available, vaillant/76.vwzio.csv: different version available
device: 192.168.2.44:9999, TCP, enhanced, firmware 1.1[5c0e].1[5c0e]
signal: acquired
symbol rate: 23
max symbol rate: 206
min arbitration micros: 0
max arbitration micros: 109
min symbol latency: 4
max symbol latency: 18
scan: finished
reconnects: 0
masters: 7
messages: 667
conditional: 15
poll: 255
update: 14
address 03: master #11
address 04: slave #25, scanned "MF=Vaillant;ID=NETX3;SW=0128;HW=0404"
address 08: slave #11, scanned "MF=Vaillant;ID=HMUX0;SW=0405;HW=0504"
address 0a: slave, scanned "MF=Vaillant;ID=PMW01;SW=0206;HW=8302", loaded "vaillant/0a.pmw.hwc.csv"
address 10: master #2
address 12: slave, scanned "MF=Vaillant;ID=PMW01;SW=0206;HW=8302"
address 15: slave #2, scanned "MF=Vaillant;ID=CTLV3;SW=0808;HW=8004", loaded "vaillant/15.ctlv3.csv"
address 26: slave, scanned "MF=Vaillant;ID=VR_71;SW=0201;HW=0503", loaded "vaillant/26.vr_71.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd, scanning
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0405;HW=0504", loaded "vaillant/76.vwzio.csv"
address ec: slave, scanned "MF=Vaillant;ID=SOL00;SW=0808;HW=8004", loaded "vaillant/ec.sol.sc.csv"
address f1: master #10
address f6: slave #10, scanned "MF=Vaillant;ID=NETX3;SW=0128;HW=0404"
address f7: master #20
address fc: slave #20, scanned "MF=Vaillant;ID=PMW01;SW=0206;HW=8302"
address ff: master #25
`
Thank you for a feedback