john30/ebusd-configuration #419: Question on MQTT and HA

wrongisthenewright, 2024-05-31:
Is there a way to set ebusd to send to HA every message decoded, even if it doesn't have a complete match for auto discovery? I have an Ariston system that send some messages that I'd like to have in HA (serial numbers and, if I got it right, program schedules). I'm struggling with mqtt-hassio.cfg.
I have removed (at least I think) all filters but I think the lack of type mappings blocks the flow.
It seems that mqtt messages are sent so I can set HA to subscribe to a topic but I 'd have to set many different sensors and I 'd like to avoid the effort..ty in advance

Koky05, 2024-05-31:
Comment out all filters.
[image]

wrongisthenewright, 2024-05-31:
Hi, 
thank you for reaching out but, as I wrote, I already removed almost all filters, I left only this filter 

`filter-non-circuit = scan|ignored 
`

to clean out only explicitly tagged messages, during the trials I had also this line commented.
I can see that MQTT messages are sent, so ebusd correctly grabs the messages and published them via MQTT, only they're missing in HA, I think because there's no rules in mqtt-hassio.cfg to map these to HA entities type/device class.

[image]

Koky05, 2024-05-31:
If you look into my config then you can see that more types was added like curve, datetime, etc.
Maybe it will help to show what you looking for.
But I understand your point of view, I also add some custom MQTT sensors to show values I exactly need and could not find in HA autodiscovery or just for retype case.