john30/ebusd-configuration #423: New message definition in 15.ctlv2.csv for zone quick veto control and information

WulfgarW, 2024-06-23:
I have a Vaillant heat pump VWL 125/5 with sensoCOMFORT module (VRC720) and sensoNET module (VR 921).
I want to be able to start (and stop) hotwater boosts and zone quick veto sessions via ebusd to use excess power of my PV system, and so I was searching the ebusd web sites and other web sites for documentation. 

For hotwater, the solution is already available on several web sites: Writing 'load' into HwcSFMode starts the hotwater boost, 
But for a zone quick veto, I did not find a documented solution to start and stop a session. 

So I did some digging ("listen -v -u" on the ebus while starting a zone quick veto via the myVaillant app) and trial and error and came to the following result:
3 new message definitions for each heating zone (z1, z2, z3) should be added to the config file 15.ctlv2.csv (here with German comments):

      r,,z1QVEndTime,Endzeit QuickVeto Zone 1,,,,1E00,,,HTI,,,end time of quick veto for zone 1
      r,,z1QVEndDate,Enddatum QuickVeto Zone 1,,,,2400,,,HDA:3,,,end date of quick veto for zone 1
      r;w,,z1QVDuration,Dauer QuickVeto Zone 1,,,,2600,,,EXP,,,duration of quick veto for zone 1

      r,,z2QVEndTime,Endzeit QuickVeto Zone 2,,,,1E00,,,HTI,,,end time of quick veto for zone 2
      r,,z2QVEndDate,Enddatum QuickVeto Zone 2,,,,2400,,,HDA:3,,,end date of quick veto for zone 2
      r;w,,z2QVDuration,Dauer QuickVeto Zone 2,,,,2600,,,EXP,,,duration of quick veto for zone 2

      r,,z3QVEndTime,Endzeit QuickVeto Zone 3,,,,1E00,,,HTI,,,end time of quick veto for zone 3
      r,,z3QVEndDate,Enddatum QuickVeto Zone 3,,,,2400,,,HDA:3,,,end date of quick veto for zone 3
      r;w,,z3QVDuration,Dauer QuickVeto Zone 3,,,,2600,,,EXP,,,duration of quick veto for zone 3

---------------------------------------------------------------------------------------------------------------------------------------------------
How to use them (explained for zone z1):
a) Starting zone quick veto sessions
   (1) The heating zone should be in automatic mode (=time controlled mode), so z1OpMode should show the value 'auto'.
   (2) The desired temperature for the quick veto is written into z1QuickVetoTemp.
   (3) The quick veto session is started by writing a duration in hours into z1QVDuration. (Allowed values are 0.5, 1.0, 1.5, 2.0, 2.5 and 3.0.)

b) Stopping or changing zone quick veto sessions
   (1) An active zone quick veto session is canceled by writing 'auto' into z1SFMode.

c) Changing zone quick veto sessions
   (1) The duration of an active zone quick veto session can be changed by writing another value into z1QVDuration. (Allowed values are 0.5, 1.0, 1.5, 2.0, 2.5 and 3.0.)

d) Getting information about active zone quick veto sessions:   
   (1) z1SFMode shows 'veto', if a zone quick veto session is active. If not, z1SFMode shows 'auto'.
   (2) z1QuickVetoTemp shows the desired temperature during the veto session.
   (3) z1QVEndDate and z1QVEndTime show the timestamp, when the veto session will end. (z1QVDuration is readable, but it always returns the value 3, which is the default duration of a veto session.) 

---------------------------------------------------------------------------------------------------------------------------------------------------

Is there someone, who can verify my results at his/her heat pump?

Who can add the 9 new message definition lines to the config file on the github repository?

chrizzzp, 2024-06-28:
@WulfgarW 
 Yes, your findings are correct.  We found the same in this PR:

https://github.com/stadid/ebusd-configuration/pull/2

There a quite a few more additions to the 15.ctvl2.csv which can be found here:
 
https://github.com/jonesPD/ebusd-configuration/blob/master/ebusd-2.1.x/en/vaillant/15.ctlv2.csv

ahemwe, 2024-11-15:
@john30 
Kindly asking to merge the values introduced in above mentioned version for `z*QuickVetoEndTime`, `z*QuickVetoEndDate` and `z*QuickVetoDuration` into `main` so that it's available as a default config.

Thanks.

john30, 2025-01-14:
reworked+done in f538bca

closed 2025-01-14 by john30

WulfgarW, 2025-01-18:
Many thanks! Tested it successfully with -c https://ebus.github.io/next/