john30/ebusd-configuration #425: How to get/write "RoomTempCorr" on VRC720?

rICTx-T1D, 2024-07-21:
Hi folks,

in file "015.e7c.csv" I found the following parameters:
r;wi,,RoomTempCorr,Room Temp Correction,,,,3300,,,temp,,,Room Temp Correction
r;wi,,OutsideTempCorr,Outside Temp Correction,,,,3600,,,temp,,,Outside Temp Correction

Now I try to read und write it from my vrc700 and and it in 15.ctlv2.csv and 08.hmu.csv. But it doesn't work. Because I can set the value in display as regular user, I think it must be possible to change it on ebusd.

Hope anyone has some idea.
Bye, RICT

JonesGW-MGD, 2024-07-22:
No idea, but I believe 15.e7c.csv is for a Protherm Thermolink P Controller and you say you have a VRC700 Controller?   

It doesn't hold true that every message for every controller would work with every other controller (if it did, then of course there would only be one controller configuration file).  

I would start with an assumption that it wouldn't work.  

If you want to explore this further, I would have thought a better approach would be to monitor traffic on the bus while you make manual changes with your controller.  By repetition you may be able to identify the message that relates to the control you are changing.  At that point you can attempt to decode the message yourself or it might be possible to match the command sequence with messages from other files (see the how-to section of the wiki: https://github.com/john30/ebusd/wiki/HowTos)

rICTx-T1D, 2024-07-23:
Hi @JonesGW-MGD , Hi @john30 

hope it is possible and allowed to write German in this issue, because my english is "not the yellow from the egg" :). 
Other people they like to adapt this on your own, please use a Google Translator. Sorry.

___

@JonesGW-MGD 
Ja ich habe ein VRC720 Controller (im warmen HWR) und dort kann man als normaler Benutzer die Raumtemperatur mit einen Wert von -10° bis +10° in 0.1K Schritte kalibrieren.  Aktuell arbeite ich mich sehr in die Steuerung und Kopplung der Vaillant-Anlage ein, wie man an den vielen Fragen sehen kann :(. 

Wie genau kann man die Daten (per ebusd und AdapterShild5) mitschneiden? Könnte ich dir ein Dump / Mitschnitt geben und du kannst das dann extrahieren????? Ich denke das wäre die Lösung für viele Anfragen, auch von anderen Personen, die den Regler im warmen Heizungsraum haben. Man könnte dann per Hausautomatisierung (bei mir OpenHab) einen Wert von anderen Thermometern (bei mir KNX) auslesen. Die VRC720-Temperatur nehmen und das Delta so setzten, dass die Heizung ein "sinnvollen" Wert hat. Auch wenn dieser nichts mehr dann mit der Luft im Raum des VRC720 zu tun hat, würde man dann die "Adaptive Heizkurve" oder Kühlfunktion sinnvoller nutzen können.

Wäre schön, wenn du etwas Zeit investieren könntest...

Viele Grüße
RICT

rICTx-T1D, 2024-07-23:
Hier ein paar ge'grab'te Werte während ich den Korrekturwert am Display eingestellt habe....
1008b5110100 / 093c0314000000000000 = 1
1008b5110101 / 09675b0080ffff0000ff = 5
1076b5040100 / 0a0311361323070224c51e = 1
1076b5110101 / 09635bb81eff6b0000ff = 5
10feb505025c00 = 1
1008b507020900 / 022e07 = 1
1038b50903298603 / 0486030080 = 1
1038b50903298703 / 0487030080 = 1
1038b50903298803 / 0488030080 = 1
1038b51303048903 / 0a8903b0042003d007b004 = 1
1076b512030f0201 / 075f03001e0314ff = 5
7108b51a03041832 / 0e1807c0cfc07b3c00b01b20000000 = 1
7108b51a03041933 / 0e1900000082000000000000000000 = 1
7108b51a03041a34 / 0e1acf2000c0022101001800061800 = 1
7108b51a03041b35 / 0e1b00000000000100b00100001000 = 1
7108b51a03041c36 / 0e1c00000000000000000000000000 = 1
1008b51009000000ffffff070000 / 0101 = 5
1038b522050002002e00 / 0100 = 1
1076b51009000000ffffff010000 / 0101 = 5
7108b5120e113c03d9029802e9021400000600 / 020000 = 4
7108b5120e113c03da029902ea021400000600 / 020000 = 3

1008b5110100 / 09390314000000000000 = 3
1008b5110101 / 09675a0080ffff0000ff = 16
1076b5040100 / 0a0314401323070224831e = 3
1076b5110101 / 09635a831eff6b0000ff = 16
10feb505025c00 = 3
10feb508020900 = 1
10feb510020600 = 1
1008b507020900 / 022207 = 3
1008b512020064 / 00 = 1
1008b5120204ff / 00 = 1
1008b513020528 / 0101 = 1
7108b509022802 / 0c020009015f0100484d553031 = 1
1038b50903298603 / 0486030080 = 3
1038b50903298703 / 0487030080 = 3
1038b50903298803 / 0488030080 = 3
1038b51303048903 / 0a8903b0042003d007b004 = 3
1076b512030f0201 / 075f0300190314ff = 16
7108b51a03042c32 / 0e2c07c0cfc07b3c00b01b20000000 = 1
7108b51a03042d33 / 0e2d00000082000000000000000000 = 1
7108b51a03042e34 / 0e2ecf2000c0022101001800061800 = 1
7108b51a03042f35 / 0e2f00000000000100b00100001000 = 1
7108b51a03043036 / 0e3000000000000000000000000000 = 1
7108b51a03043632 / 0e3607c0cfc07b3c00b01b20000000 = 1
7108b51a03043733 / 0e3700000082000000000000000000 = 1
7108b51a03043834 / 0e38cf2000c0022101001800061800 = 1
7108b51a03043935 / 0e3900000000000100b00100001000 = 1
7108b51a03043a36 / 0e3a00000000000000000000000000 = 1
7108b51a03044032 / 0e4007c0cfc07b3c00b01b20000000 = 1
7108b51a03044133 / 0e4100000082000000000000000000 = 1
7108b51a03044234 / 0e42cf2000c0022101001800061800 = 1
7108b51a03044335 / 0e4300000000000100b00100001000 = 1
7108b51a03044436 / 0e4400000000000000000000000000 = 1
1008b51009000000ffffff070000 / 0101 = 16
1038b522050002002e00 / 0100 = 3
1076b51009000000ffffff010000 / 0101 = 16
7108b5120e113a03d5029602e7021400000600 / 020000 = 6
7108b5120e113903d4029602e6021400000600 / 020000 = 3
7108b5120e113a03d4029602e7021400000600 / 020000 = 2
7108b5120e113a03d7029702e8021400000600 / 020000 = 2
7108b5120e113a03d6029602e8021400000600 / 020000 = 6

Meine VERMUTUNG ist 71->08->b51a  oder b512 da diese Häufig vorkommen.... Aber wie genau muss nun die Datei aufgebaut sein, welcher Typ ist das.... So Richtig komme ich nicht klar.

JonesGW-MGD, 2024-07-23:
Hi Rict,

I'd really love to help - I'm just not that technical I'm afraid.  I can share some thoughts and hopefully others more competent than me can jump in - I'll tell you what I think (and it's not much I'm afraid):

I think you should focus on the 10xx.... messages.  This is your VRC720 telling its 'slaves' what to do - so the 1008.... messages are directed at your heat pump, the 1038 messages are directed at your recoVAIR.

Looks like it is sending 'Status00' and 'Status01' messages (B511 01 <00>/<01>) which may have what you need to set temp.

(I think the 7108 messages is just chat between your internal hydraulic unit and the heat pump.)

My thoughts are - if you want to simply sent the desired temp, ignore the offset, write the desired temp command, but do it every 20 seconds to avoid contention with real 720?

But I am too far out of my ebusd comfort zone and your set up (I simply have a gas boiler here in the UK - nothing as sophisticated as HP and HRU!)

I'd just encourage you to keep going!  viel Glück!

Gareth

rICTx-T1D, 2024-07-23:
Hi Gareth!
> I'd really love to help - I'm just not that technical I'm afraid. I can share some thoughts and hopefully others more competent than me can jump in - I'll tell you what I think (and it's not much I'm afraid):
Thanks to try helping me! Iam a beginner on ebusd and glad for anythink idea.

> I think you should focus on the 10xx.... messages. This is your VRC720 telling its 'slaves' what to do - so the 1008.... messages are directed at your heat pump, the 1038 messages are directed at your recoVAIR.
I try to replace 71 to 10, but it doesn't work. How you defined 10 and not other number for my system as correct? How I can query / confirm the correct number of VRC720?

> Looks like it is sending 'Status00' and 'Status01' messages (B511 01 <00>/<01>) which may have what you need to set temp.
What did you mean with this status messages?

> (I think the 7108 messages is just chat between your internal hydraulic unit and the heat pump.)
Thats can be possible, maybe :).

> My thoughts are - if you want to simply sent the desired temp, ignore the offset, write the desired temp command, but do it every 20 seconds to avoid contention with real 720?
This has I doing too, but my RecoVAIR make in automode wrong heating recovery. Recov things is so hot, because the heating technical room is not a good reference for all other rooms in house.

> But I am too far out of my ebusd comfort zone and your set up (I simply have a gas boiler here in the UK - nothing as sophisticated as HP and HRU!) to offer you advice to follow, apologies.
HeatPump an HRU are so complexy, that much professional mount only on wall and think they are finished. When customer is unhappy, they can book a ultra professional for additional payment.

> I'd just encourage you to keep going! viel Glück!
Thanks!

> Gareth
Rict

rICTx-T1D, 2024-07-24:
Has anyone a idea to get the correctly adress from my grab?

JonesGW-MGD, 2024-07-25:
Hey, does this post not describe the situation you are trying to address?:

https://github.com/john30/ebusd-configuration/issues/234#issuecomment-1355625719

it discusses modifying the z1ActualRoomTempDesired setting (but depends on what OpMode you in.)

(regarding some of your other queries - the log shows your controller at addresses 15 and 10 (slave, master respectively) `address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=CTLV2;SW=0514;HW=1104"`

Similarly your heat pump is at 08, hydraulic unit at 76 (as slaves - i.e. when receiving messages)

JonesGW-MGD, 2024-07-25:
Silly question, but if your VRC720 is in the wrong room, why not move it?!!!  

It sounds like like the RecoVAIR is in a more representative room?  You should simply be able to hang the VRC720 of this just wire the bus in parallel:

[image]

rICTx-T1D, 2024-07-25:
**Sorry - I hate my professional certified installer!!!!** 
I know thats better to move to a reference room. Found same hints in manual to, but I was not ask durring building my house or before the professional installer has install anythink.
The RecoVAIR is in same room as hmu and VRC720. Because washer, dryer, solar inverter are placed into too, the room is not a good choise for a refernece room!

I has hoped I can prevent a new wireing to VRC720 in an other room by set correct value over bus. All rooms has a knx temperature dedector in switches. So I have in Openhab a very good avg of temperature from all rooms.

When we can not reach the goal by setting the value over bus, I must over-vote my wife and mounting new cables to an other room. But this is my last way to go!

JonesGW-MGD, 2024-07-26:
I feel your pain - this may not be a popular suggestion but you may be able to find a VRC720f, controller and wireless outside temp monitor  on a popular auction site - there are some in the UK for half the retail price for example; still not inexpensive but no wires or holes through walls!

rICTx-T1D, 2024-07-26:
> I feel your pain - this may not be a popular suggestion but you may be able to find a VRC720f, controller and wireless outside temp monitor on a popular auction site - there are some in the UK for half the retail price for example; still not inexpensive but no wires or holes through walls!

Thanks for your Tipp... I will be remember,  when no solution is possible by using the eebus.

How Can I find the correctly adress and type of my recorded telegrams?

axelere, 2025-04-18:
Hi Rict, 

This topic (https://github.com/john30/ebusd/discussions/1532) might interest you if you haven't found a solution.

Axel