john30/ebusd-configuration #432: vaillant/15.ctlv3.csv missing?

nam-AH, 2024-09-07:
Hello john30,
I try to connect my very new vaillant heat pump to home assistant.
In the log the following is shown:
> ebusd --foreground --mqtthost=core-mosquitto --mqttport=1883 --mqttuser=addons --mqttpass=.... --scanconfig --mqttjson --mqttint=/etc/ebusd/mqtt-hassio.cfg --mqtttopic=ebusd --device=enh:_IP-of-eBUS Adapter Shield C6_:9999 --log=all:notice
---------- skipped some lines
2024-09-07 18:05:08.207 [bus notice] scan 15: ;Vaillant;CTLV3;0709;3704
2024-09-07 18:05:08.208 [update notice] store 15 ident: done
2024-09-07 18:05:08.208 [update notice] sent scan-read scan.15  QQ=31: Vaillant;CTLV3;0709;3704
2024-09-07 18:05:08.208 [bus notice] scan 15: ;Vaillant;CTLV3;0709;3704
2024-09-07 18:05:08.361 [update notice] sent unknown MS cmd: 3115b5090124 / 09003231323333323030
2024-09-07 18:05:08.525 [update notice] sent scan-read scan.15 id QQ=31: 
2024-09-07 18:05:08.680 [update notice] sent scan-read scan.15 id QQ=31: 
2024-09-07 18:05:08.832 [update notice] sent scan-read scan.15 id QQ=31: 21;23;32;0010045478;0953;044664;N3
2024-09-07 18:05:08.832 [bus notice] scan 15: ;21;23;32;0010045478;0953;044664;N3
2024-09-07 18:05:08.986 [main error] unable to load scan config 15: no file from vaillant with prefix 15 matches ID "ctlv3", SW0709, HW3704

Is the file vaillant/15.ctlv3.csv missing at https://upd.ebusd.eu/?
Could you please create this file.
If I can help to create the file please let me know.
Regards
nam-AH

stadid, 2024-09-10:
You are using 720/3 (3rd generation). It's almost the same as 720/2 (which config is available)

1) Copy the whole config dir from Github to your device and run ebusd with  **--configpath=/path/to/config_dir**  option (replace with you local path) to use local config directory
2) in your local config dir copy 15.ctlv2.csv to 15.ctlv3.csv

That's all

I suggest to take 720/2 config file from Jones PD repository, as it's more detailed compared to officially posted one.
https://github.com/jonesPD/ebusd-configuration/blob/master/ebusd-2.1.x/en/vaillant/15.ctlv2.csv

nam-AH, 2024-09-10:
Hello @stadid 
Thank you very much for your help.
I used all the configuration files from the path in your post (../blob/master/..). Was that your recommendation?
Now the file is found. But there are several other errors. 
Some files have missing lines. Do you know, how to add these lines?
The file vaillant/f6... is missing. How to get it?
Below are the lines filtered with main error:

2024-09-10 12:04:51.476 [main error] error reading common config file vaillant/broadcast.csv: ERR: element not found, vaillant/broadcast.csv:7: ERR: element not found, field type ONOFF in field 0
2024-09-10 12:04:51.483 [main error] error reading scan config file vaillant/08.hmu.csv for ID "hmu00", SW0902, HW5103: ERR: element not found, vaillant/08.hmu.csv:25: ERR: element not found, field type SCODE in field 1
2024-09-10 12:04:54.842 [main error] error reading scan config file vaillant/15.ctlv3.csv for ID "ctlv3", SW0709, HW3704: ERR: element not found, vaillant/15.ctlv3.csv:6: ERR: element not found, field type TEMPV in field 1
2024-09-10 12:04:58.034 [main error] error reading scan config file vaillant/26.vr_71.csv for ID "vr_71", SW0201, HW0503: ERR: element not found, vaillant/26.vr_71.csv:6: ERR: element not found, field type ONOFF in field 0
2024-09-10 12:05:01.147 [main error] error reading scan config file vaillant/76.vwzio.csv for ID "vwzio", SW0902, HW5103: ERR: end of input reached, vaillant/76.vwzio.csv:1: ERR: end of input reached, missing message type/name/pbsb
2024-09-10 12:05:01.147 [main error] scan config 76: ERR: end of input reached
2024-09-10 12:05:04.088 [main error] unable to load scan config f6: no file from vaillant with prefix f6 found
2024-09-10 12:05:04.088 [main error] scan config f6: ERR: element not found
2024-09-10 12:06:46.354 [main notice] update check: version 23.3 available, broadcast.csv: different version available, memory.csv: different version available, vaillant/general.csv: different version available

nam-AH, 2024-09-10:
Thank you very much for providing the ZIP-Files.
Now I used your config.zip which is from https://github.com/jonesPD/ebusd-configuration/tree/master/ebusd-2.1.x/en
My version of ebusd (23.2.p20230716) could not follow the link in file 76.vwzio.csv to 76.vwz.csv. Therefore I copied 76.vwz.csv to 76.vwzio.csv.
Now the only error msg is "unable to load scan config f6: no file from vaillant with prefix f6 found". I will try to find this file at another location or create one by myself.
Until now, I have already a lot of entities. I still have to check which fields are valid.

nam-AH, 2024-09-10:
F6 is for the internet modul.
I found a minimal version of a f6.netx2.csv at 
https://github.com/rmalbrecht/ebusd-configuration/blob/master/ebusd-2.1.x/de/vaillant/f6.netx2.csv
Now there are only a few messages remaining regarding some fields. Maybe someone will update a 15.ctlv3.csv in the future.

closed 2024-09-10 by nam-AH