john30/ebusd-configuration #435: Vaillant VRC 720: bivalence and alternative points missing

JanCeuleers, 2024-10-01:
I have a Vaillant VRC 720 and would like to get/set the bivalence and alternative points, but it would seem that they are missing. I am interested primarily in the heating side of things, but the same gap applies to hot water as well. I would be glad to provide further information as directed.

$ ebusctl info
version: ebusd 23.3.23.3
update check: OK
device: /dev/serial/by-id/usb-ESERA_GmbH_eBus_Coupler_12001_AL8CZDNX-if00-port0, serial
signal: acquired
symbol rate: 100
max symbol rate: 394
min arbitration micros: 2
max arbitration micros: 94
min symbol latency: 4
max symbol latency: 9
scan: finished
reconnects: 0
masters: 4
messages: 547
conditional: 0
poll: 0
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0902;HW=5103", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=72000;SW=0122;HW=7703", loaded "vaillant/15.720.csv"
address 26: slave, scanned "MF=Vaillant;ID=VR_71;SW=0203;HW=5904", loaded "vaillant/26.vr_71.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0202;HW=0103"

JanCeuleers, 2024-10-07:
On 07/10/2024 17:10, chrizzz wrote:
> r;w,,HwcBivalencePoint,,,,,0100,,,tempv,,,Bivalence point hot water
> heating r;w,,HcAlternativePoint,,,,,2200,,,tempv,,,exclusively
> alternate heating below alternative point (-21=off)
> r;w,,HcBivalencePoint,,,,,2300,,,tempv,,,Bivalence point heating

Thank you very much. I confirm that this works when I merge it into
15.720.csv

chrizzzp, 2024-11-27:
@ulricdabe 

> Hi, I'm looking to the TRIVAI value I'm using my side, looks like it's not available. Nor the 3 tariffs

How about these?

from `15.ctlv2.csv`:
```
*r,,,,,,B524,02000000,,,IGN:4,,,
r,,TariffAuxHeater,,,,,1000,,,UIN,,,Secondary heater tariff value
r,,TariffPrimHeaterLow,,,,,1100,,,UIN,,,Primary heater low tariff value 
r,,TariffPrimHeaterNormal,,,,,1200,,,UIN,,,Primary heater normal tariff value
```

I'm not using tariff-based bivalence settings, so I haven't tested these. If they work, they're probably also writable.

ulricdabe, 2024-11-27:
@chrizzzp 

added to my 15.b7v.csv and it 's reading it fine, thanks !