john30/ebusd-configuration #437: VC 20CS/1-5 ebus + VRC720/2

keysim86, 2024-10-09:
I managed to start readings from the furnace via ebusd but some parameters are not read, e.g. those from energy and gas consumption. Below are the readings from the system

ebusctl info

version: ebusd 23.2.p20230716
update check: version 23.3 available
device: /dev/ttyUSB0
signal: acquired
symbol rate: 23
max symbol rate: 171
min arbitration micros: 1
max arbitration micros: 4705
min symbol latency: 5
max symbol latency: 42
scan: finished
reconnects: 0
masters: 4
messages: 660
conditional: 3
poll: 163
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0804;HW=7603", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=CTLV2;SW=0514;HW=1104", loaded "vaillant/15.ctlv2.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address f1: master #10
address f6: slave #10, scanned "MF=Vaillant;ID=NETX2;SW=4035;HW=5703"

ebusctl find | grep Energy

bai PrEnergyCountHc1 =  (ERR: invalid position for 3108b509030df600 / 00)
bai PrEnergyCountHc2 =  (ERR: invalid position for 3108b509030df800 / 00)
bai PrEnergyCountHc3 =  (ERR: invalid position for 3108b509030dfa00 / 00)
bai PrEnergyCountHwc1 =  (ERR: invalid position for 3108b509030dc600 / 00)
bai PrEnergyCountHwc2 =  (ERR: invalid position for 3108b509030dc800 / 00)
bai PrEnergyCountHwc3 =  (ERR: invalid position for 3108b509030dca00 / 00)
bai PrEnergySumHc1 =  (ERR: invalid position for 3108b509030df500 / 00)
bai PrEnergySumHc2 =  (ERR: invalid position for 3108b509030df700 / 00)
bai PrEnergySumHc3 =  (ERR: invalid position for 3108b509030df900 / 00)
bai PrEnergySumHwc1 =  (ERR: invalid position for 3108b509030dc500 / 00)
bai PrEnergySumHwc2 =  (ERR: invalid position for 3108b509030dc700 / 00)
bai PrEnergySumHwc3 =  (ERR: invalid position for 3108b509030dc900 / 00)
ctlv2 PrEnergySum =  (ERR: invalid position for 3115b52406020000005c00 / 00)
ctlv2 PrEnergySumHc = 270
ctlv2 PrEnergySumHcLastMonth = 2
ctlv2 PrEnergySumHcThisMonth = 3
ctlv2 PrEnergySumHwc = 162
ctlv2 PrEnergySumHwcLastMonth = 5
ctlv2 PrEnergySumHwcThisMonth = 1

How can this be fixed?

wizu, 2025-01-27:
@keysim86 have you resolved it?

mateuszsygula, 2025-10-05:
@keysim86 hello, i have this same issue with VC 15CS/1-5 + VRC720/2. any update for fix this issue?

keysim86, 2025-10-08:
Unfortunately not. I didn't have time to look for a solution.