john30/ebusd-configuration #440: Wolf CGB-20 "received unknown BC cmd"

MrMontesa, 2024-10-27:
Hey there,
I'm just about to setup a C6 eBUS adapter shield with my good old Wolf CGB-20 heater and a solar system for hot water.
With a bit of reading I'm now able to connect all together and have auto-created devices/entities in Homeassistant.

The ebusd addon logs still report a few unknown  BC & MM messages. What would be the correct way to decrypt them and add them to the CSV?
```
2024-10-27 10:02:04.844 [update notice] received unknown BC cmd: 71fe501710ac9d0b01f30000800080008000800080
2024-10-27 10:02:08.829 [update notice] received unknown BC cmd: 71fe50230908620900005d010000
2024-10-27 10:02:08.829 [update notice] received unknown BC cmd: 71fe50230908620900005d010000
2024-10-27 10:02:11.068 [update notice] received unknown MM cmd: 03f10800080019000e80020032
2024-10-27 10:02:19.814 [update notice] received unknown BC cmd: 71fe50180e0080008000800080008000800000
2024-10-27 10:02:34.818 [update notice] received unknown BC cmd: 71fe501710ac9d0b01f30000800080008000800080
2024-10-27 10:02:37.109 [update notice] received unknown MM cmd: 30715023094c660000005d010000
2024-10-27 10:02:37.435 [update notice] received unknown MM cmd: 30715023092cc40900005d010000
2024-10-27 10:02:37.764 [update notice] received unknown MM cmd: 3071502309b01300f4015d010000
2024-10-27 10:02:38.091 [update notice] received unknown MM cmd: 3071502309b8720000005d010000
2024-10-27 10:02:08.829 [update notice] received unknown BC cmd: 71fe50230908620900005d010000
2024-10-27 10:02:38.930 [update notice] received unknown BC cmd: 71fe50230908620900005d010000
2024-10-27 10:02:40.081 [update notice] received unknown BC cmd: 30fe50230958140100005d010000
2024-10-27 10:02:41.906 [update notice] received unknown MM cmd: 7130502309b0f40200045d010000
2024-10-27 10:02:42.077 [update notice] received unknown MM cmd: 03f10800080019000e80020032
2024-10-27 10:02:49.918 [update notice] received unknown BC cmd: 71fe50180e0080008000800080008000800000
2024-10-27 10:03:04.922 [update notice] received unknown BC cmd: 71fe501710ac9d0b01f30000800080008000800080
```
Let me klnow if I'm missing any details or can generate any reports using ebusctl comand, etc.

Also I had to copy the `ebusd-2.x.x` files in order to have them detected by most recent ebusd. If this repo cloning process is still the most recent one, I would be willing to create a PR to have this updated.

Thanks for your help and this awesome app.
 Best Sebastian

Kolbi, 2024-10-31:
Until eBus 24.1 this has been the latest version: https://github.com/john30/ebusd-configuration/tree/master/ebusd-2.1.x and with eBus 24.1 it's using the new "technique" (https://upd.ebusd.eu/). Maybe first upgrade your ebusd and your ebusd version to the latest.

MrMontesa, 2024-10-31:
Hey @Kolbi,
thanks for getting back. Appreciate your support.
Just saw that the new 24.1 is just 4 days old. That sounds like an awesome enhancement. Will certainly watch the Homeassistant addon for new releases.

As per my understanding Wolf details need to be added to ebusd-configuration/src though in order for 24.1 to pick up any new configuration via ebus.github.io. And as per today there isnt even a Wolf/kromschroeder folder jet.
Happy to help here with data from my system. I'm familiar with PRs, etc. Just need to know which data I can provide. For testing I can also spin up a docker using the new version.
Thanks much