john30/ebusd-configuration #441: VRC 720/3 not recognized, no .csv loaded

raspstar, 2024-10-27:
Hi,

i'm stuck in getting a new VRC 720/3 to be recognized in ebusd. Athough cloned this repo
   
   https://github.com/jonesPD/ebusd-configuration

ebusd does not load the correspoding 15.ctlv3 file / 15.ctlv2 link:

2ad9b828-ebusd:/# ebusctl info
version: ebusd 23.2.p20230716
update check: version 24.1 available
device: 192.168.0.120:9999, enhanced, firmware 1.1[4a1b].1[4a1b]
access: *
signal: acquired
symbol rate: 158
max symbol rate: 167
min arbitration micros: 5
max arbitration micros: 99
min symbol latency: 3
max symbol latency: 68
scan: finished
reconnects: 0
masters: 4
messages: 314
conditional: 3
poll: 221
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=1104;HW=7603", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 10: master #2
**address 15: slave #2, scanned "MF=Vaillant;ID=CTLV3;SW=0709;HW=3704"**
address 26: slave, scanned "MF=Vaillant;ID=VR_71;SW=0201;HW=0503", loaded "vaillant/26.vr_71.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd, scanning
address ec: slave, scanned "MF=Vaillant;ID=SOL00;SW=0709;HW=3704", loaded "vaillant/ec.sol.sc.csv"
address f1: master #10
address f6: slave #10, scanned "MF=Vaillant;ID=NETX3;SW=0121;HW=0404"

Any idea's?

Thanks in advance & regards...

closed 2024-10-27 by raspstar

cjoe-51, 2024-11-20:
Hi, 
I guess I had a similar problem when making the configuration local. It was working even worse then with the online config. 
The reason for this was that I copied in the vaillant folder only those ?4 files which I expected will be used. Then for BAI only one file was loaded instead of two and I believe for the CTLV3 was not loaded anything. 
But when I copied everything from online vaillant folder to the local vaillant folder then for BAI both files were loaded and for CTLV3 the linked 15.ctlv2 was finally loaded.
However in the end it has been useless as for my CTLV3 device no command from the 15.ctlv2 definition was recognized ("no data stored" for all commands in the  ebusctl find report)  :(

john30, 2024-12-28:
ctlv3 done with dec8be9e12b8a85d04cd2db9b41e5e0633d84969

cjoe-51, 2024-12-30:
@burmistrzak - is the 15.basv3 definition more useful than the 15.ctlv2? 
I don't want to experiment much because my last status when using the 15.ctlv2 enhanced by some pieces from other resources made my heating almost not working - among other the external temperature sensor was read in rotation as -40C and the real temperature in some few minutes interval

cjoe-51, 2025-01-12:
@burmistrzak  thank you, I was only now able to make tests. However no real success.
I tried latest John30 version of 15.ctlv2 from ebusd-2.1.x, so as JonesPD, so as RmAlbrecht. I also tested using the 15.700 configuration.
I can get very few data like time or room temperature but almost all other data is not reachable. On  Ebusctl Find I get almost for every item "no data stored" and when I read one after another manually then most of the items don't get any value but "read timeout", "invalid position" or some other negative answer :(

Here example of results from the 15.700 configuration : 
```
pi@pi4IobPnl:~/ebus_config_data/ebusd-2.1.x/en/vaillant $ sudo ebusctl read z1RoomTemp
11.9125

pi@pi4IobPnl:~/ebus_config_data/ebusd-2.1.x/en/vaillant $ sudo ebusctl read z1DayTemp
ERR: invalid position in decode

pi@pi4IobPnl:~/ebus_config_data/ebusd-2.1.x/en/vaillant $ sudo ebusctl read z1NightTemp
ERR: read timeout
```
z1HeatingRoomTempDesiredManualControlled which you mentioned above and which is available i guess at John30 latest 15.ctlv2 is also not recognized by my VRC 720/3 :(

Shall i open a new issue for the VRC720/3 with all details of my configuration?

chrizzzp, 2025-01-13:
@cjoe-51 

> I'm really missing that I'm unable to read and write the day temperature. Presently I don't need any detailed and complex info about any valves, pump etc.

The register 'z1DayTemp' register does not work for the VRC720/x regulators. Instead I think what you want is:

```
r;w,,z1HeatingRoomTempDesiredManualControlled,Desired Heating Room Temperature Zone 1 manual controlled,,,,2200,,,tempv,,,manual controlled desired heating room temperature for zone 1
```

I also recommend the repo of jonesPD (https://github.com/jonesPD/ebusd-configuration/blob/master/ebusd-2.1.x/en/vaillant/15.ctlv2.csv) until merged in the main repo.

cjoe-51, 2025-01-13:
> I also recommend the repo of jonesPD (https://github.com/jonesPD/ebusd-configuration/blob/master/ebusd-2.1.x/en/vaillant/15.ctlv2.csv) until merged in the main repo.

So I tried even the JonesPD configuration but the z1HeatingRoomTempDesiredManualControlled gives read timeout.
Well I may take it that my HW configuration is somehow wrong and causes those timeouts like @burmistrzak wrote. :(
Im not able to change it now.

cjoe-51, 2025-01-13:
@burmistrzak On your recommendation I updated the ebusd to v. 24.1.  After restarting the ebusd I however didnt get proper csv loading for the CTLV3 device:
`version: ebusd 24.1.24.1
update check: OK, broadcast.csv: different version available, vaillant/broadcast.csv: different version available, vaillant/hcmode.inc: different version available
device: esp-ebus.local:3335, TCP, enhanced
signal: acquired
symbol rate: 23
max symbol rate: 111
min arbitration micros: 5
max arbitration micros: 15
min symbol latency: 7
max symbol latency: 31
scan: finished
reconnects: 0
masters: 3
messages: 234
conditional: 3
poll: 0
update: 15
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=1103;HW=7603", loaded "vaillant/bai.308523.inc", "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=CTLV3;SW=0709;HW=3704"
address 31: master #8, ebusd
address 36: slave #8, ebusd
`
Why is this ? Before the update (today) the symlinked 15.ctlcv3 was loaded properly. I havent changed anything in the meantime, the locally stored config files are taken from JonesPD as discussed before.
(I hate updating!)

cjoe-51, 2025-01-15:
@burmistrzak 
> @cjoe-51 I’d recommend the following to help with timeouts:
> 
>     * Update to the [latest](https://github.com/john30/ebusd/releases/tag/24.1) ebusd version.
> 
- I updated ebusd to 24.1 - no improvement
>     * Get an [official](https://adapter.ebusd.eu/v5-c6/index.en.html) adapter, because eBus requires precise timing, preferably done in hardware.
> 
- This will potentially be the last step; I dont see any confirmation about bad parameters of the adapter I'm using in the respective forums.
>     * Don‘t use any kind of networking. Connect the adapter directly to your `ebusd` host via USB or UART.
- This suggestion I don't understand.  
Firstly even the [latest official ebus adapter](https://adapter.ebusd.eu/v5-c6/) still offers WiFi and Ethernet connections as standard connections. 
Then when I imagine how it works I expect that "read timeout" message is about the "reading" directly from the ebus bus and I expect that reading from the bus is done purely by the micro-controller of the adapter and the ebusd SW is not involved in it in any way; only after getting the data from the bus the data is given over to ebusd for further interpretation. Am I wrong?

cjoe-51, 2025-01-18:
@burmistrzak @chrizzzp 
**Problem was in powering the adapter directly from the bus.**  When I changed the powersource for external (USB) the adapter is no able to read also the other parameters.
So my result for now is that the unofficial adapter (from Daniel Kucera) is not a problem, the WiFi is also no problem. But powering from external source is a must (at least for my heating and its ebus).
Thank you for your help.

john30, 2025-01-20:
> [@cjoe-51](https://github.com/cjoe-51) I’d recommend the following to help with timeouts:
> 
>     * Update to the [latest](https://github.com/john30/ebusd/releases/tag/24.1) ebusd version.
> 
>     * Get an [official](https://adapter.ebusd.eu/v5-c6/index.en.html) adapter, because eBus requires precise timing, preferably done in hardware.
> 
>     * Don‘t use any kind of networking. Connect the adapter directly to your `ebusd` host via USB or UART.

networking is no issue at all when using Ethernet though, as the transfer is very reliable and very low latent (compared to WIFI) and the speed is uncomparable to UART (10MBps versus 115kBps)

mar3k, 2025-04-09:
@cjoe-51 I am so glad I found this issue! I've been setting up ebusd with WIFI adapter for a few weeks and getting frustrated with different kind of problems. Initially after tuning it for a few days I got it working (many read timeouts but did the minimum I wanted). But then strange issues started appearing (erased energy data for day/yesterday/this/lastmonth on both boiler and vrc720 regulator; started reporting -40 outside temperature; etc etc). Tried restarting the adapter, tried tons of EBUSD config options, modified csv config files ... no luck.

This morning switched to a closer router (before -80-90 dBm rssi; now -50-60 dBm rssi), worked a bit more stable from the beginning) but again changed to report -40 degrees temperature.

I am going to try USB adapter and hope my life will be better ;)