john30/ebusd-configuration #443: 08.hmu.csv will not be loaded after updating ebus to 24.1

Dette70, 2024-11-08:
Today I updated my ebus daemon to 24.1 I did not change anything else.
After that my **08.hmu.csv** conf file will not be loaded. Other files will be loaded successfully (see below).

If I go back to 23.1 the file is loading.

My conf files are located local (because of some specalities), but I see the same behaivor if I use the config file from the server.

....:~# ebusctl i
version: ebusd 24.1.24.1
device: 192.168.25.116:9999, TCP, enhanced, firmware 1.1[4a1b].1[4a1b]
signal: acquired
symbol rate: 81
max symbol rate: 127
min arbitration micros: 8
max arbitration micros: 9
min symbol latency: 5
max symbol latency: 30
scan: running
reconnects: 0
masters: 5
messages: 517
conditional: 0
poll: 0
update: 8
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0902;HW=5103"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=CTLV2;SW=0514;HW=1104", loaded "vaillant/15.ctlv2.csv"
address 26: slave, scanned "MF=Vaillant;ID=VR_71;SW=0203;HW=5904", loaded "vaillant/26.vr_71.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0902;HW=5103"
address f1: master #10

jan666, 2024-11-20:
I guess you have 4PortValveStatus, 4PortValveHours and 4PortValveSwitchCount in your csv. Change it to eg FourPortValveStatus and it should work again. Starting with a number is no longer supported in 24.1

Dette70, 2024-11-20:
> I guess you have 4PortValveStatus, 4PortValveHours and 4PortValveSwitchCount in your csv. Change it to eg FourPortValveStatus and it should work again. Starting with a number is no longer supported in 24.1

Thanks Jan, I will check for my local config.

But why does this behaivor occurs withe the "official" config files from the repo as well?

jan666, 2024-11-20:
It's just a guess because I had the same problem. The 08.hmu.csv from this repo does not have the 4PortValveStatus so it should not happen I guess.

Dette70, 2024-11-20:
> I guess you have 4PortValveStatus, 4PortValveHours and 4PortValveSwitchCount in your csv. Change it to eg FourPortValveStatus and it should work again. Starting with a number is no longer supported in 24.1

Thanks, that's the solution

closed 2024-11-20 by Dette70