john30/ebusd-configuration #446: missing HC1 Returntemp after updating ebus to 24.1

Jpaul1001, 2024-11-19:
thank you very much for the excellent work!

unfortunately the value for the return temperature of heating circuit 1 is no longer available!?

before: V23.3
ebusd_hmu00_returntemp_temps2
ebusd_hmu00_status01_temp1_2

switschel, 2025-01-10:
Same here returnTemp is missing in the current & next ebusd configuration. Any reasons it was dropped or not added?

roe-lz, 2025-02-14:
I'm also looking for an accurate source of returnTemp for vaillant heatpumps to be able to calculate COP. But I'm wondering if there actually is a good source for this?

I think it was available in older csv versions as ebusd_hmu00_returntemp_temps2 or sometimes just returnTemp, but afaik always pointing to b5140529.

b5140529 is still present in the latest and next versions, but is now called TestReturnTemp, which makes sense as it is under the installer test menu of the heat pump controller. 

However, I cannot actively read this value. It is only broadcasted on the ebus when I enter the installer test menu.  

So if someone is able to read accurate returnTemp from the hmu I am very interested :)

roe-lz, 2025-02-27:
After a lot of testing and trying, I found two addresses that seem to work for me, and are accurate enough for what I want to do (calculate COP).

I added these in 08.hmu.csv:

`r,,,RunDataFlowTemp,,,,b509,540200fc08,value,,IGN:4,,,,value,,EXP,,°C,temperature,
r,,,RunDataReturnTemp,,,,b509,5402000609,value,,IGN:4,,,,value,,EXP,,°C,temperature`

If others can test this on their installation and let me know if it works for them too then I will try to create a PR for these.

Jpaul1001, 2025-02-28:
Problem solved,
I made the following entry at the end of the 08.hmu.csv:
r1,,ReturnTemp,,,,,2903FFFF,,,temps2,10,°C,Return temperature (°C),,,,,,,,,,,,,,,,,,
!include,hcmode.inc,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,

this reads out the return temperature.

closed 2025-02-28 by Jpaul1001

switschel, 2025-03-03:
@roe-lz Would it be possible to make a pull request out of your finding so everybody can benefit from it?

Jpaul1001, 2025-03-04:
sorry, I can't explain the exact function of the entries...
I found them in another forum (NL)...

among other things, there are also other values ​​to read from the Vaillant Arotherm
WaterThroughput, Fan1,Fan2 - Speed, AirInletTemp, CompressorOutletTemp, CompressorInletTemp, etc

if you want, I could put the corresponding entries from the 08.hmu here...

julezman, 2025-03-26:
@roe-lz - I've also found that the status01 field provides these values and seems to update more frequently.  Any idea what, if any, the difference between these values and the RunData ones would be?

r,,,Status01,Vorlauftemperatur/Rücklauftemperatur/Aussentemperatur/WW Temperatur/Speichertemperatur/Pumpenstatus,,,b511,01,temp,,D1C,,°C,temperature,temp_1,,D1C,,°C,temperature,temp_2,,D2B,,°C,temperature,temp_3,,D1C,,°C,temperature,temp_4,,D1C,,°C,temperature,pumpstate,,UCH,0=off;1=on;2=overrun;4=hwc,,pump state

roe-lz, 2025-03-26:
Yes the Status01 has these values, but only accurate to half a degree. If that is sufficient for you it is probably the better choice. I use flow and return temps to calculate generated heat where half a degree makes a lot of difference, especially if the delta T is small.

julezman, 2025-03-26:
Fair enough.  I use the values in a similar way (calculating COP) but although the the Status01 values are less precise they are updated more frequently and therefore may provide better accuracy over time.  I'll probably end up experimenting with both.  Thanks again.