john30/ebusd-configuration #449: I miss information from my heat pump

HSoft99, 2024-11-26:
I use ebusd with my Vaillant heat pump and get some information to mqtt.

[image]

I miss some information like consumption, ...

**The result of  ebusctl info:**
version: ebusd 24.1.24.1
update check: OK, device firmware v5-c6 1[4a1b] up to date
device: IP-Adress:9999, TCP, enhanced, firmware 1.1[4a1b].1[4a1b]
signal: acquired
symbol rate: 43
max symbol rate: 210
scan: finished
reconnects: 0
masters: 5
messages: 84
conditional: 0
poll: 0
update: 10
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0902;HW=5103", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=CTLV3;SW=0709;HW=3704"
address 26: slave, scanned "MF=Vaillant;ID=VR_71;SW=0201;HW=0503", loaded "vaillant/26.vr_71.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0902;HW=5103"
address f1: master #10
address f6: slave #10, scanned "MF=Vaillant;ID=NETX3;SW=0121;HW=0404"

**ebusctl scan result**
08;Vaillant;HMU00;0902;5103;21;23;32;0010021115;1610;007452;N0
15;Vaillant;CTLV3;0709;3704;21;23;37;0010045478;0953;050226;N9
26;Vaillant;VR_71;0201;0503;21;24;20;0020184846;0082;009459;N3
76;Vaillant;VWZIO;0902;5103;21;23;25;0010023523;3110;009382;N9
f6;Vaillant;NETX3;0121;0404;21;24;11;8000015501;0933;033133;N5

ebusctl find shows a lot of element with "no data stored". For example ConsumptionTotal, YieldTotal,  Broadcast Datetime and more elements

The vaillant app shows me the information. What is the reason, that i don't get the information for my heat pump over ebusd

john30, 2024-12-28:
done wtih dec8be9e12b8a85d04cd2db9b41e5e0633d84969

closed 2024-12-28 by john30

john30, 2024-12-29:
please check the [next](https://ebus.github.io/next/) generated csvs with ebusd 24.1 (or current source) as described there or by using a local clone of https://github.com/eBUS/ebus.github.io and using the "next" folder in there