john30/ebusd-configuration #450: Config issues after upgrade to 24.x Vaillant

mike-bike, 2024-11-28:
After upgrade to 24.1 from 23.x I found issues with the configuration. Eg WPPWMPower was no longer defined (I also do see many other not founds in the log file).

`ebusctl i` shows different files as a result of a scan for bai:

`address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0703;HW=7401", loaded "vaillant/bai.308523.inc" ([id_hw=7401]), "vaillant/08.bai.csv"`

When I load 24.1 with 2.1 text config I do get the "old" file name and WPPWMPower delivers results:

`address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0703;HW=7401", loaded "vaillant/bai.0010006101.inc" ([PROD='']), "vaillant/08.bai.csv"`

I do not know how to load the online configuration to compare, but something seems to be different.

Any ideas?

Regards,
Michael

topa-LE, 2024-12-29:
Unfortunately, I can't get to grips with the config files at the moment. I've only had my heat pump for 4 weeks and am trying to load the files locally.

From version 24.1 it is recommended to use the tsp files instead of the old .csv files.

https://github.com/john30/ebusd-configuration/tree/master/src

My Scan is:
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0902;HW=5103"
address 15: slave #2, scanned "MF=Vaillant;ID=CTLV3;SW=0808;HW=8004"
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0902;HW=5103"
address ec: slave, scanned "MF=Vaillant;ID=SOL00;SW=0808;HW=8004"
address f6: slave #10, scanned "MF=Vaillant;ID=NETX3;SW=0124;HW=0404"

Here, for example, **08.HMU00.tsp** should be used. It should also be clarified what exactly the local path should look like.

--configpath=/etc/ebusd/src/vaillant/

The **CTLV3** file is also missing here.

Thank You.

mike-bike, 2024-12-29:
Hmm, looking at my original post at the top the 24.1 version was loading .csv files online. I have not seen any comment re .tsp, but honestly did also not much research. As I had seen so many errors with the new online version of the config files, I had reverted to local copy of .csv and everything is running fine since then.
I am happy to test a bit but would appreciate some guidance.

ebusd reports many updates of the config files:

```
version: ebusd 24.1.24.1
update check: OK, broadcast.csv: newer version available, memory.csv: different version available, vaillant/15.ui.csv: newer version available, vaillant/25.solsy.hwc.csv: different version available, vaillant/26.solsy.hc.csv: different version available, vaillant/50.solsy.mc.csv: different version available, vaillant/bai.0010006101.inc: different version available, vaillant/broadcast.csv: different version available, vaillant/ec.solsy.sc.csv: different version available, vaillant/errors.inc: different version available, vaillant/general.csv: different version available, vaillant/hcmode.inc: different version available, vaillant/hwcmode.inc: different version available, vaillant/iotest620.inc: different version available, vaillant/mcmode.inc: different version available, vaillant/quick.inc: different version available, vaillant/roomtempoffset.inc: dif
...
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0703;HW=7401", loaded "vaillant/bai.0010006101.inc" ([PROD='']), "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=UI   ;SW=0501;HW=6201", loaded "vaillant/15.ui.csv"
address 23: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0500;HW=6301", loaded "vaillant/23.solsy.cc.csv"
address 25: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0500;HW=6301", loaded "vaillant/25.solsy.hwc.csv"
address 26: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0500;HW=6301", loaded "vaillant/26.solsy.hc.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 3f: master #23
address 50: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0500;HW=6301", loaded "vaillant/50.solsy.mc.csv"
address 7f: master #24
address ec: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0500;HW=6301", loaded "vaillant/ec.solsy.sc.csv"
```

Happy New Year

topa-LE, 2024-12-29:
Yes, I would also like to have the necessary config files (csv or tps) locally instead of always having to access them online.

The only one that provides me with data is 08hmu.csv . The tps are not loaded correctly.

But I also urgently need the **15.ctlv3.csv**, because only the hmu and broadcast provide me with too little relevant data. Especially the power consumption in kWh of the heat pump is extremely important to me.

If you can provide me with your data, I can have a look.

Happy New Year.

topa-LE, 2024-12-29:
Ok, I completely misunderstood. The new CSV files are converted from the tsp to SRC and made available under the new CDN.

I will now :
https://github.com/eBUS/ebus.github.io/tree/main/next/de
use this data locally from the Github clone. Let's see if I can solve this ...

mike-bike, 2024-12-29:
I think I use the old 2.1 config files from [here](https://github.com/john30/ebusd-configuration/releases). I had them extracted locally and use a local configuration 

```
EBUSD_OPTS="--configpath=/home/pi/ebusd-conf/de --scanconfig -d /dev/EBUS0 -p 8888 --httpport=8080 --mqttjson --mqtttopic=ebusd --mqttport=1883 --mqtthost=mqtt -l /var/log/ebusd/ebusd.log --log=all:error --pidfile=/var/run/ebusd/ebusd.pid"
```

I had not spent much time on analysis with the newly provided online configs back in Nov'2024. Could be, that the files had some conversion errors which have been fixed meanwhile. 
Maybe a give the new version another try shortly. However, 2.1 is running fine for my 12 yrs old Vaillant equipment. Not sure if I miss anything important other than better / more complete documentation of all the Vaillant parameters.

topa-LE, 2024-12-29:
https://github.com/john30/ebusd/issues/1420

so mutch, it worked ;-) You Thanks.

git-developer, 2025-01-12:
I'm observing issues after upgrade to 24.1, too.

#### Missing `WPPWMPower`
The value `WPPWMPower` (Vaillant BAI) is missing.

I'm not sure if I understood everything correctly:
* @mike-bike solved the issue by reverting from online config to local config, old CSV format (not typespec-generated), from 2024-11
* @topa-LE solved the issue by cloning eBUS/ebus.github.io, then reverting from online config to local config using the (typespec-generated) CSVs from `next/de`

I'd like to use the typespec-generated config. Unfortunately, `WPPWMPower` is missing even after switching to `next/de` from eBUS/ebus.github.io.

Log excerpt that might be relevant:
```
[bus notice] scan 08: ;Vaillant;BAI00;0414;7401
[bus notice] scan 08: ;Vaillant;BAI00;0414;7401
[main notice] read scan config file vaillant/08.bai.csv for ID "bai00", SW0414, HW7401
[bus notice] scan 08: ;21;09;12;0010003862;3100;005747;N1
[bus notice] scan 15: ;Vaillant;43000;0127;2002
[bus notice] scan 15: ;Vaillant;43000;0127;2002
[bus notice] scan 15: ;21;07;16;0020028515;0008;013271;N2
[main notice] read scan config file vaillant/15.430.csv for ID "43000", SW0127, HW2002
```

The CSV files (2.1 vs. ebus.github.io) differ, but I don't understand if this is problematic:
```
< de/vaillant/bai.0010003857.inc:r,,WPPWMPower,d.15 Pumpendrehzahl Istwert,,,,"7300",,,percent0,,,Output of pwm pump
---
> de/vaillant/bai.0010003857.inc:r,,,WPPWMPower,d.15 Pumpendrehzahl Istwert: Output of pwm pump,,,b509,0d7300,value,,UCH,,%,
```

Can someone help out?

#### Periodic timeouts for `AdpPreH*` values
I see periodic timeouts in the log for
- `AdpPreHCurrentRoomTemp`
- `AdpPreHOutdoorTemp`
- `AdpPreHOutdoorTempStart`
- `AdpPreHRoomTempDesired`
- `AdpPreHRoomTempStart`

Is this related or a separate issue?