john30/ebusd-configuration #451: Sensors in HA vr71

pascalwinters, 2024-12-02:
How do I get those sensors in HA with theHA mqqt config file?

https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/en/vaillant/26.vr_71.csv

Koky05, 2024-12-03:
You need to change filter name to include Flow:
`filter-name = Flow|...`
and also directions to show writable values:
`filter-direction = r|u|^w`

pascalwinters, 2024-12-04:
> You need to change filter name to include Flow: `filter-name = Flow|...`

This doesn't help and the filter-direction was already configured on that way.
So the entity is not added to HA 

And also `temp` was already added to the filer-name, so it should be already added. Also because `sensor.ebusd_basv_hc1actualflowtempdesired_value` is already added.

chrizzzp, 2024-12-04:
@pascalwinters 

Changing the message type from 'w' to 'uw' (update-write) works for me:

```
*r,,,,,,B523,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
*uw,,,,,,B523,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
uw,,Mc1FlowTempDesired,,,,,0200,FTStatus,,onoff,,,,FTDesired,,temp1,,,,MixerStatus,s,onoff,,,,MixerMovement,s,percents,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
uw,,Mc2FlowTempDesired,,,,,0201,FTStatus,,onoff,,,,FTDesired,,temp1,,,,MixerStatus,s,onoff,,,,MixerMovement,s,percents,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
uw,,Mc3FlowTempDesired,,,,,0202,FTStatus,,onoff,,,,FTDesired,,temp1,,,,MixerStatus,s,onoff,,,,MixerMovement,s,percents,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
```

pascalwinters, 2024-12-04:
> @pascalwinters
> 
> Changing the message type from 'w' to 'uw' (update-write) works for me (all filters disabled in mqtt-hassio.cfg):
> 
> ```
> *r,,,,,,B523,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
> *uw,,,,,,B523,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
> uw,,Mc1FlowTempDesired,,,,,0200,FTStatus,,onoff,,,,FTDesired,,temp1,,,,MixerStatus,s,onoff,,,,MixerMovement,s,percents,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
> uw,,Mc2FlowTempDesired,,,,,0201,FTStatus,,onoff,,,,FTDesired,,temp1,,,,MixerStatus,s,onoff,,,,MixerMovement,s,percents,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
> uw,,Mc3FlowTempDesired,,,,,0202,FTStatus,,onoff,,,,FTDesired,,temp1,,,,MixerStatus,s,onoff,,,,MixerMovement,s,percents,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
> ```

What do you mean with `uw`?
At the moment I have configured it like this: `filter-direction = r|u|^w`

Koky05, 2024-12-04:
@chrizzzp points to modify [26.vr_71.csv](https://github.com/john30/ebusd-configuration/blob/master/ebusd-2.1.x/en/vaillant/26.vr_71.csv), because it show that sensors are only for write not for read.

chrizzzp, 2024-12-04:
> What do you mean with `uw`?
> At the moment I have configured it like this: `filter-direction = r|u|^w`

Message type is defined in the CSV. I use local CSVs and edited `26.vr_71.csv`.

pascalwinters, 2024-12-04:
Should we use those files: `https://github.com/john30/ebusd-configuration/tree/master/src` for the latest version?

chrizzzp, 2024-12-04:
As far as I understood they apply to ebusd version >24.x where the CSV files are generated form these source files. I haven't really looked into 24.x. So I still use ebusd version 23.x with local CSVs. But I'm quite sure this is possible also in version 24.x

pascalwinters, 2024-12-04:
> chrizzzp

When I use the csv locally, I see this error:

[image]

Can you share the content of the file you have?

pascalwinters, 2024-12-04:
I think I need those generated files: https://ebus.github.io/mapping.html
For example: https://ebus.github.io/en/vaillant/26.vr_71.csv
@john30  But where to download those for using them locally?

chrizzzp, 2024-12-04:
> When I use the csv locally, I see this error:

Are you sure you also changed the default message type to 'uw'?
`
*uw,,,,,,B523,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,`

This is the local `26.vr_71.csv` I use:

```
*r,,,,,,B523,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
*uw,,,,,,B523,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
uw,,SetActorState,,,,,05,R1,,UCH,0=off;20=on,,,R2,,UCH,0=off;20=on,,,R3,,UCH,0=off;20=on,,,R4,,UCH,0=off;20=on,,,R5,,UCH,0=off;20=on,,,R6,,UCH,0=off;20=on,,,R7,,UCH,0=off;20=on,,,R8,,UCH,0=off;20=on,,,R9,,UCH,0=off;20=on,,,R10,,UCH,0=off;20=on,,,R11,,UCH,0=off;20=on,,,R12,,UCH,0=off;20=on,,,Rx,,HEX:2,,,
uw,,Mc1FlowTempDesired,,,,,0200,FTStatus,,onoff,,,,FTDesired,,temp1,,,,MixerStatus,s,onoff,,,,MixerMovement,s,percents,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
uw,,Mc2FlowTempDesired,,,,,0201,FTStatus,,onoff,,,,FTDesired,,temp1,,,,MixerStatus,s,onoff,,,,MixerMovement,s,percents,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
uw,,Mc3FlowTempDesired,,,,,0202,FTStatus,,onoff,,,,FTDesired,,temp1,,,,MixerStatus,s,onoff,,,,MixerMovement,s,percents,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
r,,SensorData1,,,,,06,S1,,temp,,,,S2,,temp,,,,S3,,temp,,,,S4,,temp,,,,S5,,temp,,,,S6,,temp,,,,S7,,temp,,,,ignore,,HEX:2,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
r,,SensorData2,,,,,07,S8,,temp,,,,S9,,temp,,,,S10,,temp,,,,S11,,temp,,,,ignore,,HEX:2,,,,ignore,,HEX:2,,,,ignore,,HEX:3,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
!include,errors.inc,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
```

chrizzzp, 2024-12-04:
> But where to download those for using them locally?

WIth local CSVs you have to run ebusd with the option

`-c, --configpath=PATH`

If you use the HA ebusd addon you can define the local path in the configuration tab of the addon. Local files have to be in a location ebusd can access (important if you run it in a container like the HA ebusd addon does).

john30, 2025-01-14:
I don't see any action to be taken as the mentioned vr71 is already part of the repo.
someone likes to explain what is supposed to be done?

john30, 2025-01-20:
> A change from message type 'w' to 'uw' is requested for the following definitions of the `26.vr_71.csv`:
> 
> Explanation: when the type of the write messages is changed from `w` (write) to `uw` (update write), the respective entitities are autodiscovered in HA and can be monitored (e.g. every write command sent by the regulator to the vr71).

so will there never be a need to active send those messages from ebusd?
if yes, I'd say its better to change to uw. if no, solution of https://github.com/john30/ebusd/issues/1113#issuecomment-2442520744 would be better

chrizzzp, 2025-01-21:
> so will there never be a need to active send those messages from ebusd? if yes, I'd say its better to change to uw. if no, solution of [john30/ebusd#1113 (comment)](https://github.com/john30/ebusd/issues/1113#issuecomment-2442520744) would be better

To answer a question asked with a negation in English is always difficult for me to answer unambigously... ;-)

I'll try: Yes (meaning correct), I think there is never a need to the active send those messages from ebusd. These writes transmit the desired mixer position and desired flow temperature for the heating circuit. IMHO I would not want to mess with these values and leave this to the system regulator. So I suggest changing to 'uw'.

Considering the 'No' answer (i.e. there is need to active send those messages from ebsud) I couldn't find the solution here: https://github.com/john30/ebusd/issues/1113#issuecomment-2442520744. If I understand the message thread correctly, they also ended up suggesting changing it to 'uw' (as 'u' alone did not work). Or did you mean something else?

BTW: when logging ebus traffic I don't see any read messages of these messages, only very frequent writes (like every 15 seconds for a mixed heat circuit).