john30/ebusd-configuration #452: How can I ignore the first two bytes of the master data area?

SwissFanboy, 2024-12-03:
Hi
I'm in the process of expanding my 08.hmu.csv. The "hydraulic station: installer level -> test menu -> sensor/actor test" fields are already included.
Now I would like to integrate the "hydraulic station: installer level _> test menu -> statistics" fields. I've already done the analysis. I was able to identify and assign 12 of the 16 fields:
[image]
Several messages appear on the ebus for each field. Unique identification are the third and fourth fields of the message data part (gray background). The correct value is in the fourth and fifth bytes (UIN) of the slave response (green background).
I know how I can supress the first three bytes of the slave answer. But how kann I ignore the first two bytes of the master command?
*r,,,,,,B51a,,,,IGN:3,,,,,,,,,,,,,,,,,,,,,
r,,Kompressor_Std,,,,,xxxx3400,,,UIN,,h,Kompressor Std,,,,,,,,,,,,,,,,,,,
Many thanks for your help
Regards,
Daniel

chrizzzp, 2024-12-04:
@SwissFanboy 

> I know how I can supress the first three bytes of the slave answer. But how kann I ignore the first two bytes of the master command?
> *r,,,,,,B51a,,,,IGN:3,,,,,,,,,,,,,,,,,,,,,
r,,Kompressor_Std,,,,,xxxx3400,,,UIN,,h,Kompressor Std,,,,,,,,,,,,,,,,,,,

Use 'F' for each byte to be "ignored".

This works (I included the first byte '05' in the default definition because it's the same in all messaged):

```
*r,,,,,,B51A,05,,,IGN:3,,,,,,,,,,,,,,,,,,,,,
r,,Kompressor_Std,,,,,FF3400,,,UIN,,h,Kompressor Std,,,,,,,,,,,,,,,,,,,
```

Sorry for the spoiler, but a lot of these statistics/sensor values found in the installer menu of the hmu have been decoded and all went into this repo:

https://github.com/jonesPD/ebusd-configuration/blob/master/ebusd-2.1.x/en/vaillant/08.hmu.csv

SwissFanboy, 2024-12-04:
Cooool. Many thanks!

SwissFanboy, 2024-12-06:
I despair!
If I enter an absolute value in the ID field in the csv, ebusd uses this definition. 
As soon as I work with the wildcard, the csv entry remains unused.
What am I doing wrong?

[image]

Can somebody help me?

chrizzzp, 2024-12-06:
Hm... obviously FF does not act as a wildcard but is a regular part of the ID. Sorry for the confusion. This needs clarification.

For my system (Arotherm Split) your second message definition works. I assume the IDs vary among different systems.

According to your first (working) example, the master message ID does not have to be specified completely (050103ffff) in the message, a partial match seems sufficient (050103). So how about trying the following:

```
*r,,,,,,B51A,05,,,IGN:3,,,,,,,,,,,,,,,,,,,,,
r,,Energieertrag_Tag_Heizen,,,,,3200,,,UIN,10,kWh,,,,,,,,,,,,,,,,,,,
r,,Energieertrag_Tag_Kuehlen,,,,,3201,,,UIN,10,kWh,,,,,,,,,,,,,,,,,,,
```

If this does not work then I would probably go for the complete ID:

```
*r,,,,,,B51A,05,,,IGN:3,,,,,,,,,,,,,,,,,,,,,
r,,Energieertrag_Tag_Heizen,,,,,bb3200,,,UIN,10,kWh,,,,,,,,,,,,,,,,,,,
r,,Energieertrag_Tag_Kuehlen,,,,,493201,,,UIN,10,kWh,,,,,,,,,,,,,,,,,,,
```

Question remains: is 
`r,,Energieertrag_Tag_Heizen2,,,,,eb3200,,,UIN,10,kWh,,,,,,,,,,,,,,,,,,,`
a different message although the slave answer is identical (corresponding to 4279,9 kWh).

SwissFanboy, 2024-12-07:
Many thanks

For me (hydraulic module VWZ MEH 97/6 to aroTHERM plus VWL 125/6 A S2) it seems to be slightly different.
- The IDs for the HMU Sensor/Actor Test fields are unique.
- For HMU Yield Indicator, Live Monitor and Statistics fields, the 2nd ID byte appears randomly. It isn't hex:ff.

[image]

With the third and fourth ID bytes (ID_2) I could clearly assign the message.
To do this I would have (at least) to be able to ignore the second ID byte. (The first byte is always Hex:05.)

How can I do that? Or is there another approach? @john30 

Regards,
Daniel

SwissFanboy, 2024-12-07:
I notice that the second ID byte is the same as the first response byte. What does that mean?

chrizzzp, 2024-12-07:
Interesting. For my system `7108b51a0405ff3202` would be a poll by the vwz for 
 
`r,,YieldHwcDay,,,,,FF3202,,,UIN,10,kWh,,,,,,,,,,,,,,,,,,,`

In your example it seems to be different. There are two consecutive polls for `05a3 3202`  and `05a4 3202` which result in different answers from the same ID_2. So byte 2 of ID_1 could be some sort of offset which is added to the ID to read different values (e.g. historical data) - just guessing... 
BTW ebusd can do indexed reading (ebustcl option `-i VALUE     read additional message parameters from VALUE`).

The slave answer is also different. On my system the poll for 'YieldHcDay' is encoded like this:
`3108b51a0405ff3200 / 0aff082f67000000000000`
The result data bytes starte with the fourth byte (2f67) while in your example the data seem to be encoded in the last bytes (6b00).

SwissFanboy, 2024-12-08:
Problem solved. The crucial tip or suggestion came from you. Thank you.

I never said that my data/analyses come from requests directly on the HMU user interface.
And it took me some time to realize that the second ID byte has no relevance and what that mean. Namely that I can enter whatever I want there. Also Hex:FF. I just have to make the query using ebusd (ebusctl). Then ebusd will insert FF and the slave will also give its answer with FF. And that's exactly how it is. So it's a kind of wildcard. But it doesn't work with the user interface - which is not the finally use case  :-)

[image]

Thanks again for your support.

closed 2024-12-08 by SwissFanboy

chrizzzp, 2024-12-08:
> Problem solved. The crucial tip or suggestion came from you. Thank you.

Thanks for sharing the info. This explains the behaviour. To get automatic updates you could also publish regular MQTT polls for this mesage.

BTW if you can decode any new ebus messages from the Arotherm Plus it would be great if you could share them, too.