john30/ebusd-configuration #457: How and where to define a certain value template for a certain sensor

Connor0308, 2024-12-22:
Dear,

after having tweaked my configuration back and forth, it seems that I now have quite a good starting point for my project.
However, one thing that still worries me is a simple value template correction.

Using this configuration line

`r;w,,OffsetOutsideTemp,,,,,2D00,,,D2C,,K,Offset outside temperature (used to determined DisplayedOutsideTemp) `

I wanted to make the offset for the temperature sensor writable.
This is how the the sensor is appearing in the MQTT explorer.

```
{
  "unique_id":"ebusd_ctlv2_OffsetOutsideTemp_0",
  "name":"OffsetOutsideTemp ",
  "device":{
   "identifiers":"ebusd_ctlv2",
   "manufacturer":"ebusd.eu",
   "name":"ebusd ctlv2",
   "via_device":"ebusd",
   "sw_version":"23.2",
   "suggested_area":"Heizung"
  },
  "value_template":"{{value_json[\"0\"].value}}",
  "state_topic":"ebusd/ctlv2/OffsetOutsideTemp",
  "command_topic":"ebusd/ctlv2/OffsetOutsideTemp/set",
  "min":-2047.94,
  "max":2047.94,
  "step":0.06,
  "unit_of_measurement":"K",
  "device_class":"temperature"
 }
```

While I can generally live with the strange min and max values, the a step of 0.06 is not favorable.
How can I fix this?

Thank you!
Regards

Connor

Connor0308, 2024-12-23:
For anyone getting here (including a future me) - a possible workaround is to send the configuration 
to the topic followed by /config and to retain the message.

The above mentioned configuration would look like

```
{
  "unique_id":"ebusd_ctlv2_OffsetOutsideTemp_0",
  "name":"OffsetOutsideTemp ",
  "device":{
   "identifiers":"ebusd_ctlv2",
   "manufacturer":"ebusd.eu",
   "name":"ebusd ctlv2",
   "via_device":"ebusd",
   "sw_version":"23.2",
   "suggested_area":"Heizung"
  },
  "value_template":"{{value_json[\"0\"].value}}",
  "state_topic":"ebusd/ctlv2/OffsetOutsideTemp",
  "command_topic":"ebusd/ctlv2/OffsetOutsideTemp/set",
  "min":-20,
  "max":20,
  "step":0.5,
  "unit_of_measurement":"K",
  "device_class":"temperature"
 }

```

cyberthom42, 2025-01-15:
I have the same issue. I would like to set the min max and step somewhere. I even did not understood where the actual values for min, max and step are taken from. It would not make sense to take it form the datatype. It would be great if it could be set in the csv for each value. 
Is there any way to set this in HA?

Koky05, 2025-01-16:
@cyberthom42 You cann't set any definition in HA for Autodiscovery entities. Only way is like @Connor0308 did or change `mqtt-hassio.cfg` based on data type what you already know it is not the bes method.
On the other hand you still can create your own MQTT sensor in HA:
```
mqtt:
  number:
    - name: 'OffsetOutsideTemp'
      unique_id: ebusd_ctlv2_OffsetOutsideTemp_0
      device:
        identifiers: 'ebusd_ctlv2'
        manufacturer: 'ebusd.eu'
        name: '"ebusd ctlv2'
        via_device: 'ebusd'
        sw_version: '23.2'
        suggested_area: 'Heizung'
      availability:
        payload_available: true
        payload_not_available: false
        topic: "ebusd/global/signal"
      icon: mdi:temperature
      value_template: >
        {% if value_json['0']['value'] is defined %}
          {{ (value_json['0']['value']|float) | float(0) }}
        {% else %}
          0
        {% endif %}
      state_topic: 'ebusd/ctlv2/OffsetOutsideTemp'
      command_topic: 'ebusd/ctlv2/OffsetOutsideTemp/set'
      min: -20
      max: 20
      step: 0.5
      mode: slider
      unit_of_measurement: "K"
      device_class: temperature
```

Connor0308, 2025-01-18:
@Koky05 : Great!
Thank you, this is what I was looking for!

To understand you correct - by defining the corresponding sensors via the config/ mqtt.yaml like this, I can leave the retained messages approach aside?

Regards

Connor

Koky05, 2025-01-18:
@Connor0308 you still need retained message not to loose last received information from eBUSd. This approuche just define preciously type of value and sensor.

Connor0308, 2025-01-18:
This looks good to me. At the moment, I am not retaining the last values, but the configuration topic.

Thank you!

closed 2025-01-18 by Connor0308