john30/ebusd-configuration #463: Wolf / Brink devices .csv library support

gruijter, 2025-01-02:
Hi @john30,

I found this library that has all Brink (=Wolf) device configs as .csv. I have no clue how to implement this in your ebusd source, but was hoping you have a way to easily add it to the available defaults :)

https://github.com/pvyleta/ebusd-brink-hru

The info was collected and documented by @pvyleta

PS: I have a Brink Renovent Excellent300 myself, so at least support for that would be awesome!

pvyleta, 2025-01-05:
@gruijter Thanks for tagging me and I am happy you find my work useful! 

Don't worry, I already opened a PR and I am trying my best to have these improvements merged: https://github.com/john30/ebusd-configuration/pull/406

john30, 2025-01-14:
closed as duplicate of #406

closed 2025-01-14 by john30