john30/ebusd-configuration #470: Does ebusd support more than one heat pump on the same ebus?

smcarley, 2025-01-14:
I have two Vaillant aroTHERM plus 12 kW working together in my installation. I can only get data from one of them (via HMU).  Does ebusd support multiple heat pumps?

These are the devices on the bus and the csv files loaded.

```
root@ebusd-deployment-b949559db-hvrds:/# ebusctl info
version: ebusd `24.1.24.1
update check: OK, device firmware v5-c6 1[4c15] available, broadcast.csv: different version available, memory.csv: different version ```
available, vaillant/08.hmu.csv: different version available, vaillant/broadcast.csv: different version available, vaillant/errors.inc: different version available, vaillant/general.csv: different version available, vaillant/hcmode.inc: different version available
device: 192.168.107.79, TCP, enhanced, firmware 1.1[4819].1[4819]
signal: acquired
symbol rate: 142
max symbol rate: 188
min arbitration micros: 3
max arbitration micros: 115
min symbol latency: 4
max symbol latency: 34
scan: finished
reconnects: 0
masters: 6
messages: 239
conditional: 0
poll: 73
update: 23
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0902;HW=5103", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 13: master #12
address 15: slave #2, scanned "MF=Vaillant;ID=BASV3;SW=0760;HW=7304"
address 18: slave #12, scanned "MF=Vaillant;ID=V32;SW=0121;HW=9802"
address 26: slave, scanned "MF=Vaillant;ID=VR_71;SW=0201;HW=0503", loaded "vaillant/26.vr_71.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0202;HW=0103"
address f1: master #10
address f6: slave #10, scanned "MF=Vaillant;ID=NETX3;SW=0124;HW=0404"
```

Koky05, 2025-01-15:
If you can see it in recognized devices on different addresses you can reach data from them. I think you need to located files 76.vwzio.csv and f6.next3.csv to obtain data from your heatpumps. If file it that prefix doesn't exist but they exist with different refix you can rename them or create simlinks.

smcarley, 2025-01-15:
You're bang on. The second heat pump appears to be behind the v32 coupler. Setting latency to 100000 and copying 08.hmu to 18.v32 appears to find something but I get no updates other than the below.  Is anything else required to receive messages from the heat pump behind the v32 coupler?

`2025-01-15 09:43:45.281 [update notice] received read v32 currenterror QQ=f1: -;-;-;-;-`

For reference these are related enquiries from a while back
https://github.com/john30/ebusd/issues/109
https://github.com/john30/ebusd/issues/198

smcarley, 2025-01-15:
Assuming I'm doing correctly, no.

```
ebusctl read -f -c v32 State
ERR: element not found
```

JonesGW-MGD, 2025-01-17:
@smcarley - what do you get with `ebusctl find -V State` out of interest?

smcarley, 2025-01-17:
The reason why reading v32 was not working was because I was using the JonesPD config and when I copied 08.hmu to create 18v32 there was a config file clash (ERR: duplicate entry).  I only spotted when I scrutinised the ebusd log.  I fixed this by updating the field definitions to ensure the addresses didn't clash. So I now see data from the second heat pump hiding behind the V32 coupler!

Next challenge is to work out the daily COP of each heat pump which I think I will have to use TotalEnergyUsage and YieldHc and their respective values at start of day.

smcarley, 2025-01-17:
> [@smcarley](https://github.com/smcarley) - what do you get with `ebusctl find -V State` out of interest?

```
ebusctl find -V State
hmu DCFSignalState = no data stored [ZZ=08, active read]
hmu State00 = S00_SupplyTemp=54.75 °C [temperature];field2=14;S00_CompressorModulation=45.5 %;CompressorState=24;HeatingState=Heating;field6=0;Defrost=inactive;S00_CompressorPower=25.0 % [ZZ=08, lastup=2025-01-17 15:59:34, active read]
hmu State01 = temp1=54.5 °C [temperature];temp1=50.5 °C [temperature];temp2=- °C [temperature];temp1=- °C [temperature];temp1=- °C [temperature];pumpstate=warm water circuit [pump state] [ZZ=08, lastup=2025-01-17 15:59:40, active read]
hmu State02 = no data stored [ZZ=08, active read]
hmu State03 = no data stored [ZZ=08, active read]
hmu State07 = power=25 % [Compressor power];energy=82.4 kWh [daily environmental energy yield];disp1ay_b0_heaterenabled=on;display_b1=off;display_b2_backupheater=off;display_b3=off;display_b4=off;display_b5_noisereduction=off;display_b6_dhwecomode=on;display_b7=on;DisplayHeaterMain=dhw;DisplaySystemPressure=1.40 bar;DisplayHeaterBackup=standby [ZZ=08, lastup=2025-01-17 15:59:47, active read]
v32 DCFSignalState = no data stored [ZZ=18, active read]
v32 State00 = S00_SupplyTemp=53.94 °C [temperature];field2=15;S00_CompressorModulation=45.5 %;CompressorState=24;HeatingState=Heating;field6=0;Defrost=inactive;S00_CompressorPower=25.0 % [ZZ=18, lastup=2025-01-17 15:59:35, active read]
v32 State01 = temp1=53.5 °C [temperature];temp1=50.0 °C [temperature];temp2=- °C [temperature];temp1=- °C [temperature];temp1=- °C [temperature];pumpstate=warm water circuit [pump state] [ZZ=18, lastup=2025-01-17 15:59:30, active read]
v32 State02 = no data stored [ZZ=18, active read]
v32 State03 = no data stored [ZZ=18, active read]
v32 State07 = no data stored [ZZ=18, active read]
```