john30/ebusd-configuration #475: Missing parameters on Arotherm VWZ AI (compr noise reduction, conf.DHW.build.pump)

zarch1972, 2025-01-17:
Was wondering if anyone has managed to get the following parameters from the VWZ AI heat pump controller available?

- Conf.heat.build.pump
- Conf.cool.build.pump
- Conf.DHW.build.pump
- Compr.noise reduct

It would be great to be able to read (and set these values) remotely rather than having to go to the controller all the time.

Thank you.

zarch1972, 2025-01-17:
I'm using JonesPD CSV files

https://github.com/jonesPD/ebusd-configuration

```
mick@ebusd:~ $ ebusctl info
version: ebusd 23.3.23.3
update check: version 24.1 available, device firmware 1[4c15] available, broadcast.csv: newer version available, vaillant/08.hmu.csv: different version available, vaillant/broadcast.csv: different version available, vaillant/hcmode.inc: different version available, vaillant/yield3f40.inc: newer version available
device: /dev/ttyAMA0, serial high speed, enhanced, firmware 1.1[3501].1[3501]
signal: acquired
symbol rate: 84
max symbol rate: 410
min arbitration micros: 6
max arbitration micros: 3183
min symbol latency: 0
max symbol latency: 13
scan: finished
reconnects: 0
masters: 6
messages: 870
conditional: 0
poll: 0
update: 18
address 00: master #1
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0902;HW=5103", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=BASV0;SW=0217;HW=8803", loaded "vaillant/15.basv.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd, scanning
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0202;HW=0103", loaded "vaillant/76.vwzio.csv"
address ec: slave, scanned "MF=Vaillant;ID=SOL00;SW=0217;HW=8803", loaded "vaillant/ec.sol.sc.csv"
address f1: master #10
address f6: slave #10, scanned "MF=Vaillant;ID=NETX2;SW=4035;HW=5703"
```

chrizzzp, 2025-01-19:
@zarch1972 
 
> * Compr.noise reduct

If this refers to the noise reduction level (40-60%), on my (split) system I can read/write this value directly to the `hmu`:

```
*r,,,,,,B51A,05,,,IGN:3,,,,,,,,,,,,,,,,,,,,,
*w,,,,,,B51A,06,,,,,,,,,,,,,,,,,,,,,,,,
r;w,,CompressorSilentOperationLevel,,,,,FF3428,,,percent0,,,Compressor Silent Operation level,,,,,,,,,,,,,,,,,,
```

read: `3108b51a0405ff3428 / 0aff021b280028003c0001`

stgm, 2026-05-09:
@burmistrzak Oops that was a bit daft. When adding the above codes to my config they work fine, providing the same answer as to the vwzio. I was assuming that I could test any code by using the `read -h` command in ebusd, but apparently not. What's the usual way to test a code ad-hoc?