john30/ebusd-configuration #477: Support Wolf CHA

J11522, 2025-01-22:
Over at https://github.com/puni2k/ebusd-configuration-wolf/pull/3 there is an existing config which I've extended with values from the ism7mqtt project

I was wondering how to best coordinate this effort to integrate the Wolf CHA config into the central ebusd-configuration.
Or if you are not planning to include external config.

domasist, 2025-02-19:
@J11522  I am also interested. I just ordered batch of [ebus adapters](https://www.elecrow.com/ebus-adapter-shield-c6.html), not delivered yet, so I am still waiting. I would like to test Wolf CHA-10 units with Home Assistant.  How does CHA works over ebusd? Did you test CHA in linux or also over MQTT to Home Assistant?

J11522, 2025-02-20:
From what I can tell, it works rather well. I have ebusd running in PVE and get the data into homeassistant via MQTT and vice versa.

For some settings, I'm not sure if they can be controlled via EBUS, e.g. heat curve.
For others, there seems to be multiple ways of controlling. E.g. the temperature adjustment. This can be controlled via the param `(WP009) Temperaturüberhöhung Sammler` but this doesn't reflect on the display.

domasist, 2025-02-20:
> From what I can tell, it works rather well. I have ebusd running in PVE and get the data into homeassistant via MQTT and vice versa.
Thank you. It's great to know that. When I got those adapters I will update here what I get from Wolf CHA.  Till then, could you plaese upload screenshots of Wolf CHA details/entities inside Home Assistant? 

> For some settings, I'm not sure if they can be controlled via EBUS, e.g. heat curve. For others, there seems to be multiple ways of controlling. E.g. the temperature adjustment. This can be controlled via the param `(WP009) Temperaturüberhöhung Sammler` but this doesn't reflect on the display.
So it's not correct on the display of CHA or BM-2 module? Even so, will it update the desired temparature correctly?

domasist, 2025-06-02:
> From what I can tell, it works rather well. I have ebusd running in PVE and get the data into homeassistant via MQTT and vice versa.

@J11522 Thank you for the info. It's great to know that. When I got those adapters I will update here what I get from Wolf CHA.  Till then, could you plaese upload screenshots of Wolf CHA details/entities inside Home Assistant? 

> For some settings, I'm not sure if they can be controlled via EBUS, e.g. heat curve. For others, there seems to be multiple ways of controlling. E.g. the temperature adjustment. This can be controlled via the param `(WP009) Temperaturüberhöhung Sammler` but this doesn't reflect on the display.

So it's not correct on the display of CHA or BM-2 module? Even so, will it update the desired temparature correctly?