john30/ebusd-configuration #484: Partload on  "MF=Vaillant;ID=BAI00;SW=0703;HW=7401", loaded "vaillant/bai.0010006101.inc" ([PROD='']), "vaillant/08.bai.csv

m8haben, 2025-02-08:
I can no longer set my "PartloadHcKW" setting to an appropriate value using the "w -c bai PartloadHcKW 10.0" command. The command is not implemented, although I then read 10.0 with the command "r -c PartloadHcKW. 
The value on the gas boiler remains unchanged. 
I believe the wrong csv is being loaded to my gas boiler. The assignment doesn't fit. With the previous version of ebusd the "08.bai.HW7401.csv" was loaded and with the new version the "bai.0010006101.inc" which does not fit. Which csv is the right one?
How can I fix this error?

version: ebusd 23.3.23.3
update check: version 24.1 available
device: /dev/ttyUSB0, serial
signal: acquired
symbol rate: 66
max symbol rate: 179
min arbitration micros: 3
max arbitration micros: 236
min symbol latency: 0
max symbol latency: 19
scan: finished
reconnects: 0
masters: 7
messages: 871
conditional: 24
poll: 1
update: 12
address 01: master #6
address 03: master #11
address 06: slave #6, scanned "MF=Vaillant;ID=PMS00;SW=0107;HW=4302", loaded "vaillant/06.pms.csv"
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0703;HW=7401", loaded "vaillant/bai.0010006101.inc" ([PROD='']), "vaillant/08.bai.csv"
address 0a: slave, scanned "MF=Vaillant;ID=PMW00;SW=0117;HW=4402", loaded "vaillant/0a.pmw.hwc.csv"
address 10: master #2
address 12: slave, scanned "MF=Vaillant;ID=PMW00;SW=0117;HW=4402"
address 15: slave #2, scanned "MF=Vaillant;ID=UI   ;SW=0501;HW=6201", loaded "vaillant/15.ui.csv"
address 23: slave
address 25: slave
address 26: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0500;HW=6301", loaded "vaillant/26.solsy.hc.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 3f: master #23
address 50: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0500;HW=6301", loaded "vaillant/50.solsy.mc.csv"
address 7f: master #24
address ec: slave, scanned "MF=Vaillant;ID=SOLSY;SW=0500;HW=6301", loaded "vaillant/ec.solsy.sc.csv"
address ed: slave, scanned "MF=Vaillant;ID=PMS00;SW=0107;HW=4302", loaded "vaillant/ed.pms.sc.csv"
address f7: master #20
address fc: slave #20, scanned "MF=Vaillant;ID=PMW00;SW=0117;HW=4402"

Best regards Roland

JonesGW-MGD, 2025-02-08:
Hi Roland,

HW=7401 should load bai.308523.inc (I'm not sure there ever was a 08.bai.HW7401.csv unless that's one you crafted yourself?)

Looks like it might be loading bai0010006101.inc because you have no product ID.

Have you had the PCB changed?

JonesGW-MGD, 2025-02-08:
Hi Roland,

HW=7401 should load bai.308523.inc (I'm not sure there ever was a 08.bai.HW7401.csv unless that's one you crafted yourself?)

Looks like it might be loading bai0010006101.inc because you have no product ID.

If you do `ebusctl find id`. what does the line starting `scan.08 id` show?

Have you had the PCB changed by any chance?

m8haben, 2025-02-08:
Hi, there is nothing.
scan.08 id = ;;;;;;
Yes, it was changed once, but I'm not sure whether it still worked afterwards. I think it worked after that.

scan id = no data stored
scan.06 id = 21;12;29;0020071488;3110;005985;N5
scan.08 id = ;;;;;;
scan.0a id = 21;12;30;0010007267;3110;006857;N3
scan.12 id = 21;12;30;0010007267;3110;006857;N3
scan.15 id = 21;10;49;0020080465;0907;005064;N0
scan.26 id = 21;10;49;0020080463;0907;005377;N3
scan.50 id = 21;10;49;0020080463;0907;005377;N3
scan.ec id = 21;10;49;0020080463;0907;005377;N3
scan.ed id = 21;12;29;0020071488;3110;005985;N5
scan.fc id = 21;12;30;0010007267;3110;006857;N3

JonesGW-MGD, 2025-02-08:
In any event, loading  bai.0010006101.inc should allow you to write PartloadHcKW.

When you look (installer level) at the diagnostic code 000 on the boiler does it show 'Auto' or a numerical value?

m8haben, 2025-02-08:
At the moment it says 6 because I entered that in the diagnostic menu. I usually only need 6KW because I also have a wood stove with a water heat exchanger. But when the outside temperatures go well below -4° Celsius, the power is not enough and I increase the value to 8 or 10KW. So that I don't have to change this manually on the boiler, I use ebusctl. This is very convenient for me.

ebusctl w -c bai PartloadHcKW 10.0
done

ebus@ebuspi:~ $ ebusctl r -c bai PartloadHcKW
10

But nothing change on the Boiler. 
When I read the boiler in the diagnostic menu, it still says 6