john30/ebusd-configuration #485: Recovair 360 behind VR32/3

maxp1256, 2025-02-11:
Hello!

What's the current state about the communication with this setup and ebusd? Should I install 2 ebus-adapters or is there any configuration-option to get it working with one adapter in the "big" bus with multimatic (after VR32/3)?

JonesGW-MGD, 2025-02-11:
Hi, generally it's possible with a single ebus-adpater.  you need to rename the recovair 08. file to <add>.vr32.csv (s oof course this means locally hosting your configuration files) and adjusting the latency to 100000.  there are quite a few posts on the discussion forum of ebusd (Ethan than this configuration files site), you may wish to start here:

https://github.com/john30/ebusd/discussions/1223#discussioncomment-8749250

https://github.com/john30/ebusd/discussions/1430

JonesGW-MGD, 2025-02-13:
Great!