john30/ebusd-configuration #487: Help - How to Offset Value of EXP 2 bytes

muraus, 2025-02-20:
Hi,

I am trying to read following HEX Code from ebus via configuration file:

`ebusctl hex 08b5160114`

The read-operation works perfect:

[image]

When I decode the message the match is in the EXP decoding with an offset of 2 bytes:

[image]

How do I configure this correctly in the "old" .csv file?
I tried following:

[image]

But it doesn't work.

Question: Anybody knows how to offset in the config file (2 - bytes)?

chrizzzp, 2025-02-21:
You only need to ignore (offset) 1 byte, as the first byte contains only the number of bytes following in the message.

Try this:

`r,hmu,PowerHMU,,,,b516,14,,IGN:1,EXP,,W,,,,,,,,,,,,,,,,,,,`

Alternative: use a message template to define the offset:

08.hmu.csv:
```
*r,,,,,,B516,,,,IGN:1,,,,,,,,,,,,,,,,,,,,,
r,,PowerHMU,,,,,14,,,EXP,,W,,,,,,,,,,,,,,,,,,,
```

muraus, 2025-02-21:
Perfect - thank you sooo much. Didn't know the `IGN:` command to ignore bytes.

closed 2025-02-21 by muraus