john30/ebusd-configuration #488: ebusd-configuration "different version available"

Thde23, 2025-02-23:
After the change from ebusd 23.2->24.1, the [CDN repository](https://github.com/eBUS/ebus.github.io) was also cloned to /home/admin/ebus.github.io/. The old cloned configurations of (https://github.com/john30/ebusd-configuration.git) were removed.
How to activate the mentioned online configuration or where to set it up, I could not find out so far.

Initially, I had the same problem as here: (https://github.com/john30/ebusd-configuration/issues/450). Obviously, the wrong configuration was loaded here: “vaillant/bai.308523.inc” instead of “vaillant/bai.0010006101.inc”
After a reboot, however, the problem had disappeared and the system worked so far. 

Last night, however, communication between the boiler and the controller broke down. The cause was the connection between the eBUS adapter shield and the heating. I had to use an older raspiBackup because I couldn't get the system to run again. 

Now I suspected that the messages listed below “different version available” had led to the problems! 
How can I fix this?

/etc/default/ebusd
EBUSD_OPTS=" --device=ens:/dev/ttyAMA0
--latency=50

**--configpath=/home/admin/ebus.github.io/de**
--pollinterval=5
--scanconfig=full

..."

ebusctl info
version: ebusd 24.1.24.1
update check: OK, device firmware v5 1[5210] up to date, broadcast.csv: different version available, memory.csv: different version available, vaillant/08.bai.csv: different version available, vaillant/15.430.csv: different version available, vaillant/bai.0010006101.inc: different version available, vaillant/broadcast.csv: different version available, vaillant/general.csv: different version available, vaillant/scan.csv: different version available
device: /dev/ttyAMA0, serial high speed, enhanced, firmware 1.1[5210].1[5210]
access: *
signal: acquired
symbol rate: 41
max symbol rate: 155
min arbitration micros: 0
max arbitration micros: 109
min symbol latency: 4
max symbol latency: 12
scan: finished
reconnects: 0
masters: 3
messages: 472
conditional: 16
poll: 1
update: 16
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0706;HW=7401", loaded "vaillant/bai.0010006101.inc" ([Scan_id_product='']), "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=43000;SW=0215;HW=2002", loaded "vaillant/15.430.csv"
address 26: slave, scanned "MF=Vaillant;ID=43000;SW=0215;HW=2002"
address 31: master #8, ebusd
address 36: slave #8, ebusd, scanning

Pi 5 mit Debian GNU/Linux 12 (bookworm)
ebusd: V24.1
System: Linux Pi5 6.6.74+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux
eBUS Adapter Shield v5: [ESP32-C3, rev 4], V.2025-02-16, reported as 1[5210]

Tux92, 2025-03-14:
I'm in the same situation all works well with 23.2 and a custom config forked from original but now I want to update to 24.1, but unfortunately I have only correct message for 15.ctlv2 and my 08.hmu is dead.
on my docker compose I use: `EBUSD_CONFIGPATH: "https://ebus.github.io/next/"` as I read in other issue

```
version: ebusd 24.1.24.1
update check: OK, device firmware v5 1[5210] up to date, broadcast.csv: different version available, vaillant/08.hmu.csv: different version available
device: 10.0.0.33:9999, TCP, enhanced, firmware 1.1[5210].1[5210]
signal: acquired
symbol rate: 23
max symbol rate: 156
min arbitration micros: 1
max arbitration micros: 23
min symbol latency: 4
max symbol latency: 43
scan: finished
reconnects: 0
masters: 4
messages: 680
conditional: 0
poll: 392
update: 29
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0607;HW=5103", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=CTLV2;SW=0514;HW=1104", loaded "vaillant/15.ctlv2.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZIO;SW=0202;HW=0103", loaded "vaillant/76.vwzio.csv"
```

pulla2908, 2025-08-04:
I have a brand new setup / installation and I get this error / info / warning from update check too. This is my output:

````
version: ebusd 25.1.p20250414
update check: revision 25.1 available, device firmware v5-c6 1[560f] up to date, broadcast.csv: different version available, memory.csv: different version available, vaillant/08.hmu.csv: different version available, vaillant/15.b7v.csv: different version available, vaillant/broadcast.csv: different version available, vaillant/general.csv: different version available, vaillant/scan.csv: different version available
device: 192.168.8.154:9999, TCP, enhanced, firmware 1.1[560f].1[560f]
access: *
signal: acquired
symbol rate: 43
max symbol rate: 200
min arbitration micros: 1
max arbitration micros: 5
min symbol latency: 5
max symbol latency: 25
scan: finished
reconnects: 0
masters: 5
messages: 706
conditional: 0
poll: 253
update: 29
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=HMU00;SW=0522;HW=5103", loaded "vaillant/08.hmu.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=B7V00;SW=0422;HW=5503", loaded "vaillant/15.b7v.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd
address 71: master #9
address 76: slave #9, scanned "MF=Vaillant;ID=VWZ00;SW=0522;HW=5103", loaded "vaillant/76.vwz.csv"
address f1: master #10
address f6: slave #10, scanned "MF=Vaillant;ID=NETX3;SW=0126;HW=0404"

````

But all csv files are working (08.hmu.csv, 15.b7v.csv, 76.vwz.csv) and I get all the entities based on the filter (filter-name) in mqtt-hassio.cfg

I cloned [this](https://github.com/eBUS/ebus.github.io) and configured it like this: /config/repos/ebus.github.io/next/de. I use [this](https://github.com/LukasGrebe/ha-addons) AddOn to run eBUSd.

Another thing which confuses me is this:

[image]

I have installed 25.1. The latest version is 25.1. So why do I get the information about an available update?