john30/ebusd-configuration #490: Adding support for reading gas/electricity consumption for Vaillant bai HW:7603 - advice on implementing the solution properly

filippz, 2025-02-27:
Long story short - I've found a way to read gas/electricity consumption from my _VUI 25/32 CS/1-5_ (`MF=Vaillant;ID=BAI00;SW=0807;HW=7603`) and I need a little advice on what would be the best way to implement the support for it.

I was having an issue reading reading gas/electricity as I would get `ERR: invalid position in decode` when trying to read `PrEnergyCountHc1`/`PrEnergyCountSum1` and all the rest of `PrEnergyCount`/`PrEnergySum` values. This seem to be common occurrence (for example: https://github.com/john30/ebusd-configuration/issues/437, https://github.com/john30/ebusd/issues/620, https://github.com/john30/ebusd/discussions/1126,...) with one common "denominator": HW version of all those reports seem to be **7603**

By looking at the communication between `bai` and `ctlv2` I saw some unsupported commands/messages that pointed me to the right track with the following conclusions:

- `ctlv2` sends `bai` 30 commands each one for a single Wh value for:
heating/hot water (2) x
gas/electricity/total (3) x
previous month/current month/previous year/current year/total (5)
- commands are `b516` with 100**X**ffff0**Y**0**ZW**13**Q**` as a payload where:
**X**: 0=All, 2=Month, 3=Year (1 probably means Day)
**Y**: 3=Electricity, 4=Gas
**Z**: 0=All, 3=Heating, 4=Hot water
**W**: 2=Previous, 4=Current (only valid for Month, ie X=2, otherwise it has a value of 2)
**Q**: 2=Previous, 4=Current (only valid for Year, ie X=3, otherwise it has a value of 2)
- replies are 11 bytes long where the last 4 bytes are Wh value in EXP format

To support this, I've added:

- an `energye` entry to _templates.csv:
`energye,IGN:7;EXP,,Wh,`
- 30 entries to bai.308523.inc like:
```
r,,ElectricitySumHcPreviousYear,,,,b516,1003ffff03032130,,,energye,,,
r,,GasSumHcPreviousYear,,,,b516,1003ffff04032130,,,energye,,,
...
```

Before I prepare a proper PR, I would like to hear some opinions if those entries make sense or there is a more logical approach? Should HW:7603 be addressed separately and `PrEnergyCount`/`PrEnergySum` moved outside bai.308523.inc or it can be in the same file (and filtered out with [HW=7603] or similar)?

chrizzzp, 2025-02-28:
@filippz 
@burmistrzak

Wow, excellent reverse engineering (I guess)! I was investigating into decoding the b516 energy statistics, too, for some time, but got stuck (https://github.com/john30/ebusd-configuration/pull/472#issuecomment-2609437358). Your work finally solved it! **Thanks a lot!**

Good news: they also work for my heat pump (hmu, `MF=Vaillant;ID=HMU00;SW=0902;HW=5103`) using the same regulator (ctlv2, but will most probably work also for the related basv, basv3, ctlv3,... regulators).

Current hypothesis: for heat pumps these registers also deliver the enviromental energy statistics (energy gained from the environment) - see below! 

>     * commands are `b516` with 100**X**ffff0**Y**0**ZW**13**Q** as a payload where:
>       **X**: 0=All, 2=Month, 3=Year (1 probably means Day)

So far I had no success in getting any non-zero values with X=1 (days). See below.

>       **Y**: 3=Electricity, 4=Gas

Y: 2=environmental yield

>       **Z**: 0=All, 3=Heating, 4=Hot water
>       **W**: 2=Previous, 4=Current (only valid for Month, ie X=2, otherwise it has a value of 2)
>       **Q**: 2=Previous, 4=Current (only valid for Year, ie X=3, otherwise it has a value of 2)

I would expect there should also be a W/Q-like specifier valid for Day (X=1). I will monitor ebus traffic to investigate into daily energy statistics.

>     * 30 entries to bai.308523.inc like:
> 
> 
> ```
> r,,EnergySumHcPreviousYear,,,,b516,1003ffff03032130,,,IGN:7;energye,,,
> r,,FuelSumHcPreviousYear,,,,b516,1003ffff04032130,,,IGN:7;energye,,,
> ...
> ```

Yes, probably the way to go for the moment. I'm not sure if the new TypeScript-based definitions are more flexibible in this context. Maybe @burmistrzak can comment?

> Before I prepare a proper PR, I would like to hear some opinions if those entries make sense or there is a more logical approach? Should HW:7603 be addressed separately and [...]

I think it can be done using HW-specific 'conditions' in the csv.

filippz, 2025-02-28:
> [@filippz](https://github.com/filippz) [@burmistrzak](https://github.com/burmistrzak)
> 
> Wow, excellent reverse engineering (I guess)! I was investigating into decoding the b516 energy statistics, too, for some time, but got stuck ([#472 (comment)](https://github.com/john30/ebusd-configuration/pull/472#issuecomment-2609437358)). Your work finally solved it! **Thanks a lot!**
> 
> Good news: they also work for my heat pump (hmu, `MF=Vaillant;ID=HMU00;SW=0902;HW=5103`) using the same regulator (ctlv2, but will most probably work also for the related basv, basv3, ctlv3,... regulators).
> 

Took me a while figure it out - it kinda helped that my kid was learning about IEEE 754 format a couple of weeks ago which I wasn't thinking about for some years. Since `ctlv2` refreshes those values only at midnight and startup I was pulling `ctlv2` from the wall and back a few times, but I'm glad it's working out for you also :) Also, I'd tell that regulators are not important here - since we're not talking with them, but we'll have to figure out for which devices/versions should this be used.

> So far I had no success in getting any non-zero values with X=1 (days). See below.

It was just a guess, since my `ctlv2` doesn't display daily values but I have those numbers on the boiler itself and I haven't tested it further because these are not important for my user case.

> Y: 2=environmental yield

I have 0 experience with HP so this is good for those users but we need to se if those can be used universally (ie even on devices other than heat pumps) or need to be added separately. Do you get an reply for 4=Gas which simply returns 0 or you get an error?

> I would expect there should also be a W/Q-like specifier valid for Day (X=1). I will monitor ebus traffic to investigate into daily energy statistics.

Please do - from my perspective W/Q don't make sense since I would simply expect one boolean value which would indicate current/previous and be ignored for total. So those W/Q values might have some additional function/values which in my case can simply be used for current/previous. If you're up to it - you can try pulling your `ctlv2` from the wall and back to force refresh and catch those messages.

> Yes, probably the way to go for the moment. I'm not sure if the new TypeScript-based definitions are more flexible in this context. Maybe [@burmistrzak](https://github.com/burmistrzak) can comment?

That's why we're chatting here :) I don't really know that much about all of this so experienced users might do it differently.

> I think it can be done using HW-specific 'conditions' in the csv.

I was assuming that we're speaking about HW=7603, but since it obviously includes more devices and not only `bai` maybe some "includes" that would be pulled from a couple of places would be in order.

Thanks for the feedback!

chrizzzp, 2025-02-28:
@filippz 

> > Y: 2=environmental yield
> 
> I have 0 experience with HP so this is good for those users but we need to se if those can be used universally (ie even on devices other than heat pumps) or need to be added separately. Do you get an reply for 4=Gas which simply returns 0 or you get an error?

For 4=Gas my HP returns 0.
Indeed it would be interesting to hear from people with e.g. geothermal HP, whether they also can read the corresponding values using the Y=2 (environmental yield) or a different parameter (source yield).

> Please do - from my perspective W/Q don't make sense since I would simply expect one boolean value which would indicate current/previous and be ignored for total. So those W/Q values might have some [...]

You don't actually need to restart your `ctlv2` everytime to catch these messages (i.e. pulling the regulator from the wall), all EBUS messages (known and unknown) can actually be found in the ebusd logfile (loglevel 'notice') and will appear every night shortly after midnight, as you've noticed, too. So you just have to wait for them...

filippz, 2025-02-28:
> For 4=Gas my HP returns 0.

Cool - I'd say that this means that there is no need for splitting the definitions across devices as they all support all commands and simply return 0 for stuff they don't use for their operation

> So you just have to wait for them

Being from Balkans waiting it out simple wasn't on the table :) Logic was to force those messages at will and compare them with whatever `ctlv2` was displaying at that time and see which replies contained different payloads and what changed on the display. I even tried changing time to 23:59 and wait it out a full minute but `ctlv2` would change back time to real one just before 00:00 (this might be some ebusd magic). At that time the Neanderthal part of my genome prevailed and off the wall it went...

chrizzzp, 2025-02-28:
@filippz 

I think there is more information available than only from current and previous month/year. At least on my HP it seems more complex than:
 
> **X**: 0=All, 2=Month, 3=Year (1 probably means Day)
>       **Y**: 3=Electricity, 4=Gas
>       **Z**: 0=All, 3=Heating, 4=Hot water
>       **W**: 2=Previous, 4=Current (only valid for Month, ie X=2, [...]

To illustrate: when requesting the **annual** data (**X=3**, Y=2/3, Z=0/3/4,W=2) for my setup:

Q=0/1 result in same (previous years?) value
`ebusctl hex 08b516081003ffff03002130 / 0b0301ff03002130fc0d0a4a`
`ebusctl hex 08b516081003ffff03002131 / 0b0301ff03002130fc0d0a4a`
Q=2/3 result in (current years?) value
`ebusctl hex 08b516081003ffff03002132 / 0b0300ff0300213231895d49`
`ebusctl hex 08b516081003ffff03002133 / 0b0300ff0300213231895d49`
Q=4/5... result in 0 (see below!)

I have no idea why the same values are stored in consecutive registers (Q=0/1, 2/3).
The slight differences between them (for the current year values) reflect the current standby power consumption of the HP, as I am requesting with Z=0).

BTW: I get more data when iterating W and Q (X=3, Y=2/3, Z=0/3/4):

W=0/1 and Q=0/1: deliver same (previous years?) value as W=2/3 and Q=0/1
```
08b516081003ffff03000130 / 0b0301ff03002130fc0d0a4a
08b516081003ffff03000131 / 0b0301ff03002130fc0d0a4a
08b516081003ffff03001130 / 0b0301ff03002130fc0d0a4a
08b516081003ffff03001131 / 0b0301ff03002130fc0d0a4a
```
W=0/1 and Q=2/3: deliver (almost) same (this years?) values 
```
08b516081003ffff03000132 / 0b0300ff03002132b3a55d49
08b516081003ffff03000133 / 0b0300ff03002132b3a55d49
08b516081003ffff03001132 / 0b0300ff03002132b5a55d49
08b516081003ffff03001133 / 0b0300ff03002132b6a55d49
```
W=2/3 and Q=0/1: deliver same (previous years?) value 
```
08b516081003ffff03002130 / 0b0301ff03002130fc0d0a4a
08b516081003ffff03002131 / 0b0301ff03002130fc0d0a4a
08b516081003ffff03003130 / 0b0301ff03002130fc0d0a4a
08b516081003ffff03003131 / 0b0301ff03002130fc0d0a4a
```
W=2/3 and Q=2/3: deliver (almost) same (this years?) value
```
08b516081003ffff03002132 / 0b0300ff0300213229a75d49
08b516081003ffff03002133 / 0b0300ff030021322aa75d49
08b516081003ffff03003132 / 0b0300ff030021322ba75d49
08b516081003ffff03003133 / 0b0300ff030021322ba75d49
```

W=0/1 and Q=4/5... result in 0
W=2/3 and Q=4/5... result in 0
W=4/5 and Q=4/5... result in 0
etc.

Interestingly, when interating W from 0 to F (hex) with Q=2 I get the same (non-zero) values:

```
08b516081003ffff03000132 / 0b0300ff030021325aa75d49
08b516081003ffff03001132 / 0b0300ff030021325aa75d49
08b516081003ffff03002132 / 0b0300ff030021325aa75d49
08b516081003ffff03003132 / 0b0300ff030021325aa75d49
08b516081003ffff03004132 / 0b0300ff030021325aa75d49
08b516081003ffff03005132 / 0b0300ff030021325ba75d49
08b516081003ffff03006132 / 0b0300ff030021325ba75d49
08b516081003ffff03007132 / 0b0300ff030021325da75d49
08b516081003ffff03008132 / 0b0300ff030021325da75d49
[... 94 lines cut ...]
W=A/B May
W=C/D June
W=E/F July
Q=1
W=0/1 August
W=2/3 September
W=4/5 October
W=6/7 November
W=8/9 December

Similar for this years monthly values:
Q=2
W=2/3 January
W=4/5 February
etc.
Q=3
W=0/1 August
W=2/3 September
etc.

What do you think?

Could you try to confirm this on your `bai` unit to figure out if this is specific to the HP units?

Concerning Q: maybe more then two years are recorded, then Q could become >3 (like Q=4 with your bai unit. How many years do you have your system running?)

filippz, 2025-02-28:
> for **monthly** data (**X=2**, Y=2/3, Z=0/3/4, **Q=0**) iterating W

I think you're right - I tested the theory by adding all the 12 monthly values for energy last year and this matches the value returned by 1003ffff03002130. That would then make 2 x 12 entries naming the month explicitly...

> Concerning Q: maybe more then two years are recorded, then Q could become >3 (like Q=4 with your bai unit. How many years do you have your system running?)

My system also reports the same for Q0/1 and Q2/3 - Since `cltv2` sends Q0/2 I only considered those values.
Q4/5 returns 0 for me despite my unit being from 2023. I think system might just keep current/previous and total since 1000ffff03002132 (total electric energy) gives me more than 1003ffff03002130 + 1003ffff03002132 (previous + current year).

chrizzzp, 2025-02-28:
Thanks for checking!

> I think you're right - I tested the theory by adding all the 12 monthly values for energy last year and this matches the value returned by 1003ffff03002130. That would then make 2 x 12 entries naming the month explicitly...

Yes, and even more if you want the **complete** monthly data:
x3 (for Z: 0=All, 3=Heating, 4=Hot water) = 72
x2 (for Y: 3=Electricity, 4=Gas or 2=Environment) = 144

I'm not sure I want this in my CSV...

However, there is an -i (index) option in the 'ebusctl read' command (https://github.com/john30/ebusd/wiki/3.1.-TCP-client-commands#read) which could be used  to reduce the number of required message definitions in the CSV. It's used for instance to read the 'errorhistory' or the timers in the regulators.

> My system also reports the same for Q0/1 and Q2/3 - Since `cltv2` sends Q0/2 I only considered those values. Q4/5 returns 0 for me despite my unit being from 2023. I think system might just keep [...]

So how about the Q=4 readings you mentioned above? Do they work?

>Q: 2=Previous, 4=Current (only valid for Year, ie X=3, otherwise it has a value of 2)

filippz, 2025-02-28:
> So how about the Q=4 readings you mentioned above? Do they work?

Sorry - my original post had a copy/paste error (fixed it now) - `ctlv2` uses Q=0 for this year and Q=2 for previous one. Q=4 returns 0 for me.
For now I'm leaving W in original post as is - March starts tomorrow and we'll confirm if new queries from `ctlv2` will be with W=6 as expected based on your findings.

> I'm not sure I want this in my CSV...

I'm only using two total gas values (heating + hot water) for HomeAssistant as they're enough to monitor gas usage and I'm measuring electricity with Shelly Pro EM and not monitoring individual device usages. I guess that for the typical user this might be just enough.

filippz, 2025-02-28:
We would also need to change the tittle of this issue as it turns out this is relevant to much more devices that I anticipated...

filippz, 2025-03-01:
> For now I'm leaving W in original post as is - March starts tomorrow and we'll confirm if new queries from `ctlv2` will be with W=6 as expected based on your findings.

Confirmed by looking at the messages from `ctlv2` this morning - I've updated first post accordingly.

chrizzzp, 2025-03-01:
> **W**+**Q**: - Month/Year:
 
>     * for Year (X=3):
>       
>       * W: has a fixed value of 2
>       * Q: 0=Previous, 2=Current

As written in my post above, for annual values (X=3) W does not need to be a fixed value (W=2), any value of W will give the same result. I assume the same is true for Q and W with the total values (X=). Haven't checked this, though.

BTW: I just noticed the `b516` registers can also be read from the hydraulic station (inside unit) of my HP (`MF=Vaillant;ID=VWZ00;SW=0522;HW=5103`) which houses the backup (immersion) heater. The energy statistics are then for the backup heater. So another device supporting this - nice!

And: Since HP can also cool, there should also be separate cooling energy statistics. I suspect there is another Z value for this (0=All, 3=Heating, 4=Hot water, ?=Cooling). Will look for this at a warmer time of the year (I normally don't use cooling).

filippz, 2025-03-01:
> As written in my post above, for annual values (X=3) W does not need to be a fixed value (W=2), any value of W will give the same result. I assume the same is true for Q and W with the total values (X=). Haven't checked this, though.

I'd prefer to use values that `cltv2` is using as "safe" ones. As you noted, it might be totally irrelevant, but it might be relevant to certain device only while others ignore it, or these might be "reserved for future use" and may change over time. If we encounter any regulation device that sends different values we could investigate further if those devices do get back different values.

> BTW: I just noticed the `b516` registers can also be read from the hydraulic station (inside unit) of my HP

I was pondering if ebus has anything that might help to figure out if the device supports this kind of commands - ie by sending a single `b516` and see if the device replies with a valid reply then load appropriate inc/csv files...

chrizzzp, 2025-03-02:
> I'd prefer to use values that `cltv2` is using as "safe" ones. 

OK, I see this point.

> I was pondering if ebus has anything that might help to figure out if the device supports this kind of commands - ie by sending a single `b516` and see if the device replies with a valid reply then [...]

I don't think it is possible within ebus right now. Only way I think is to probe product/HW/SW IDs (from scanning result) and then use conditions to include certain CSV files, like used here: https://github.com/eBUS/ebus.github.io/blob/main/next/en/vaillant/08.bai.csv

BTW could you try the following `b516` register on your `bai` unit which yields the current electrical power in W (on my HP system)? I'm just wondering how 'universal' the `b516` registers are.

```
*r,,,,,,B516,,,,IGN:1,,,,,,,,,,,,,,,,,,,,,
r,,CurrentConsumedPowerWatt,,,,,14,,,EXP,,,current consumed electrical power (Watt),,,,,,,,,,,,,,,,,,
```

filippz, 2025-03-02:
> could you try the following `b516` register on your `bai`

I get an error (maybe I'm doing something wrong?):

```
ebusctl read -f -V CurrentConsumedPowerWatt
ERR: invalid position in decode: bai CurrentConsumedPowerWatt
```
And the log is:
```
2025-03-02 11:30:33.806 [update error] unable to parse read bai CurrentConsumedPowerWatt from 3108b5160114 / 00: ERR: invalid position
2025-03-02 11:30:33.806 [bus notice] >3108b516011451<000000>00
2025-03-02 11:30:33.806 [main error] read bai CurrentConsumedPowerWatt: decode ERR: invalid position
```

chrizzzp, 2025-03-02:
> I get an error (maybe I'm doing something wrong?):

Ok thanks. No, the message sent by the master looks OK (`08b5160114`). Obiously, it doesn't work for your unit. We previously found for the HP units that this register was introduced with later firmwares (SW>9). 

There is a second register which yields the current electrical power from the `b509` region (a block which contains a lot of diagnostic data) working also with older firmwares of our HP units. Not sure, whether the `b509` data are really device-specific, though.

```
*r,,,,,,B509,540200,,,IGN:4,,,,,,,,,,,,,,,,,,,,,
r,,RunDataCurrentConsumedPower,,,,,5B0D,,,powerv,,,current consumed electrical power (Watt),,,,,,,,,,,,,,,,,,
```

chrizzzp, 2025-03-03:
> * **Z**: 0=All, 3=Heating, 4=Hot water

I could now confirm for the Arotherm heat pump:

Z=5 - Cooling energy statistics

Even the mixer function module VR71 (FM5, `MF=Vaillant;ID=VR_71;SW=0104;HW=0503`) has its own `b516` energy statistics, counting maybe the power consumptions of the connected pumps/mixer unit? Another proof that this register block is indeed present in many devices. Maybe you should rename the PR with something like "general `b516` energy statistics"?

BTW: everytime just before the energy statistics are polled at midnight from the different devices, the regulator sends a "generic" `b516 11` message to the slaves. The response seems to be always the same, maybe some sort of configuration...? Do you find this message, too?

```
2025-03-02 00:00:01.262 [update notice] received unknown MS cmd: 1008b5160111 / 09020309100934090a09
2025-03-02 00:00:01.530 [update notice] received unknown MS cmd: 1026b5160111 / 090103010e0130010301
...
2025-03-02 23:59:59.025 [update notice] received unknown MS cmd: 1008b5160111 / 09020309100934090a09
2025-03-02 23:59:59.294 [update notice] received unknown MS cmd: 1026b5160111 / 090103010e0130010301
```

filippz, 2025-03-04:
@chrizzzp
> Z=5 - Cooling energy statistics

Added to first post - thanks.

> Maybe you should rename the PR with something like "general `b516` energy statistics"?

We're still not at PR stage, but we're getting close. For this discussion maybe "Reveres engineering b516 energy statistics" would be OK?

> BTW: everytime just before the energy statistics are polled at midnight from the different devices, the regulator sends a "generic" `b516 11` message to the slaves. The response seems to be always the same, maybe some sort of configuration...? Do you find this message, too?

Yes - mine is always `1008b5160111 / 090105040e0430040a04`. I ignored these earlier, but now I'm thinking that in this way the `cltv2` is actually _asking_ `bai` what statistic can be read from it and based on the reply it actually goes into reading them out. I guess at some point (if the assumption is correct) we might first query the device with `b516 11` and only then load the appropriate ebusd definitions. Out of curiosity I might change the display language to see if the reply changes...

@burmistrzak
> What is returned when you asked your boiler for some unsupported value? E.g.: `Z=5`
```
ebusctl decode exp $(ebusctl hex 08b516081000ffff03052132  | tail -c 10)
0.0
```
I'd say that all `b516` _devices_ simply return 0 for unsupported statistics (as noted earlier by @chrizzzp: _For 4=Gas my HP returns 0._) 

> Would be great to have a single TypeSpec definition for as many units as possible.

Yes it would. If you're up to it you're more than welcome to prepare the actual commits/PR since I don't have too much experience with all of this and my boiler probably has only a small subset of statistics that your HP is able to provide.

chrizzzp, 2025-03-04:
@filippz 

> > Z=5 - Cooling energy statistics
> 
> Added to first post - thanks

Could you also add:

Y=2 environmental energy

This is the energy heat pumps extract from the air (or ground source).

Thanks.

filippz, 2025-03-04:
> Y=2 environmental energy

Added - I also changed the tile while at it...

Thanks.

chrizzzp, 2025-03-04:
@burmistrzak 

While I generally agree we shouldn't mess with the heat generators too much by generating a lot of ebus traffic, I think in the special case of the `b516` energy statistics we are bound to poll them from the circuits directly, if we want the full data. During the nightly poll my regulator (ctlv2) seems to read from the heat pump only:

1. monthly electrical energy values of the current and previous month (all modes/heating/dhw/cooling)
2. annual electrical energy values of the current and previous year (all modes/heating/dhw/cooling)
3. system cumulative electrical energy values (all modes/heating/dhw/cooling) 
4. monthly environmental energy values of the current and previous month (all modes only)
5. annual environmental energy values of the current and previous year (all modes only)
6. system cumulative environmental energy value (all modes only)

Statistics 1. to 3. are also polled every night from `vwz00` and `vr_71`, but these data don't appear directly but - I assume - are used to calculate total electrical energy values for the complete system (as shown in the regulator).

This "reduced" set of energy data seems to be polled by the "current" generation of regulators (ctlvx, basvx, ...), while other ("previous") regulators indeed polled more values (e.g. monthly data for the complete two years, as shown in the picture below for the VRC700 regulator). I wonder why Vaillant didn't implement these detailed statistics also in the newer regulators...

[image]

However, if we want to limit the polls to the "newer" regulators, we'll have to find the 'copy' registers in the regulators, which contain the energy statistics, because the `b516` block of e.g. the `ctlv2` contains only the particular energy statistics of the regulator circuit (and solar circuit in my case), not the statistics from the heat generator circuits etc.

Anyway, a few of these 'copy' registers for the ctlvx/basvx are known, some are missing (e.g. for cooling and all modes):

`15.ctlv2.csv`:
```
*r,,,,,,B524,02000000,,,IGN:4,,,
*w,,,,,,B524,02010000,,,,,,
r,,SolarYieldTotal,total solar yield,,,,3D00,,,energy4,,,solar yield total
r,,YieldTotal,total environmental yield,,,,3E00,,,energy4,,,envrionment yield total
r;w,,PrEnergySumHcThisMonth,this months energy consumption heating,,,,4F00,,,energy4,,,current month electricity consumption heating
r;w,,PrEnergySumHwcThisMonth,this months energy consumption hot water,,,,5000,,,energy4,,,current month electricity consumption hot water
r;w,,PrEnergySumHcLastMonth,last months energy consumption heating,,,,5300,,,energy4,,,last month electricity consumption heating
r;w,,PrEnergySumHwcLastMonth,last months energy consumption hot water,,,,5400,,,energy4,,,last month electricity consumption hot water
r;w,,PrEnergySumHc,total energy consumption heating,,,,5700,,,energy4,,,total year electricity consumption heating
r;w,,PrEnergySumHwc,total energy consumption hot water,,,,5800,,,energy4,,,total year electricity consumption hot water
```

In my oppinion we should leave polling the complete `b516` registers optional, e.g. by putting all these registers in a separate e.g. `b516_statistics.inc` CSV, to be invoked "per request" by the different circuits. The good thing is that the `b516` registers really seem to be "universal" for many circuits and will result in a zero-value response (no error message) if a certain energy statistic is not present on a particular device. Consequently, one `b516_statistics.inc` could be used for all devices.

filippz, 2025-03-04:
@chrizzzp 
> There's a typo in the title, should be 'reverse' I guess?

Fixed - thanks.

> 2\. **new:** _solar energy_ from solar water circuit (**Y=1**)

Added to first post.

> The last block using **Y=9** is interesting again. I'm not sure what kind of 'total' energies it shows. Maybe you can have a look on your units?

`bai` returns 0 - I guess my combo with only gas boiler + regulator is the most simple one of the lot so a lot of readings doesn't make too much sense

@burmistrzak 

> We probably shouldn’t poll that many (~30) registers regularly… 😅
> Might be better to simply define the registers and tag them as `u` (i.e. update/passive) only.

Agreed  - I only really use 2 (total electrical energy and gas).
I guess we could try  using TypeScript to to define a couple of arrays (X,Y,Z,W+Q) and use for loops to loop trough those arrays and dynamically generate all the combinations/definitions. But as you noted - the real questions is whether there is any reason to have all the definitions, or just the "most useful ones".

djramipro, 2025-03-10:
@filippz 
I am newbie in using John30's ebusd but very interested in being able to collect energy data from my gas boiler.

My model is VMW 32CS/1-5 (N-ES) ("MF=Vaillant;ID=BAI00;SW=0806;HW=7603") and the regulator is VRT 380f but haven't show energy statistics.

I've played with X=1 but I only get a nonzero value for 08b516081001ffff0403**6**132. I get 41,750 Kwh. It is a value that can be for a day but I couldn't check for whick correspond. From the boiler's panel I only have for the Heating circuit a yesterday gas consuption of 27 Kwh.

chrizzzp, 2025-03-10:
@djramipro 

Interesting find. I also get a few non-zero values from my heat pump with X=1. Of course, not for gas (Y=4), but for electricity (Y=3), environmental energy (Y=2) and the "environmental-related", yet unknown energy (Y=9). And so far only for 'heating' (Z=3) and 'all' (Z=0, same as for heating).

Haven't iterated W/Q yet...

```
08b516081001ffff03036132 / 0b0109000303613200081f46
08b516081001ffff02036132 / 0b010901020361320096fe46
08b516081001ffff09036132 / 0b0109020903613200011547
```

Could these be yesterdays energies?

filippz, 2025-03-10:
> I've played with X=1 but I only get a nonzero value for 08b516081001ffff04036132. I get 41 Kwh.

I've tried with `ebusctl decode exp $(ebusctl hex 08b516081001ffff04036132  | tail -c 10)` and I get 33733 Wh back (34kWh) which doesn't correspond to any value for today/yesterday displayed on the boiler.

> Haven't iterated W/Q yet...

For me, only W/Q combo that doesn't return 0.0 for X=1 is 6/2 - I wonder if W**13**Q should be something other than 13....

djramipro, 2025-03-11:
@filippz, @filippz 

I thought the value of “13” was fixed.

Iterating with the first digit “1” and for gas consumption of the heating circuit and current month:

08b516081001ffff04036**1**32  | 41750  | 
08b516081001ffff04036**2**32  | 49036  | 
08b516081001ffff04036**3**32  | 49840  | 
08b516081001ffff04036**4**32  | 33700  | 
08b516081001ffff04036**5**32  | 28729  | 
08b516081001ffff04036**6**32  | 30838  | 
08b516081001ffff04036**7**32  | 34571  | 
08b516081001ffff04036**8**32  | 42183  | 
08b516081001ffff04036**9**32  | 27697  | 
08b516081001ffff04036**a**32  | 38103  | matches my gas consumption yesterday
08b516081001ffff04036**b**32  | 0.0       | 
08b516081001ffff04036**c**32  | 0.0       | 
08b516081001ffff04036**e**32  | 0.0       | 
08b516081001ffff04036**f**32   | 0.0       | 

later with more time I will continue testing.

chrizzzp, 2025-03-11:
@filippz @djramipro 

Obviously we work in parallel... ;-)

> which doesn't correspond to any value for today/yesterday displayed on the boiler.

Correct, when reading them today these values do not change.

>For me, only W/Q combo that doesn't return 0.0 for X=1 is 6/2 - I wonder if W13Q should be something other than 13....

Correct, I now iterated the first digit after W and got non-zero values (see below). I think we need to extend the variable names: I suggest **V** for the digit following W (as XYZ is already taken...) and U for the second digit.

100**X**ffff0**Y**0**ZWVUQ**

I get 11 non-zero values with X=1, Y=2/3/9, Z=0/3/4, W=6, U=3, Q=2 when iterating V from 1 to B (hex), e.g.

```
08b516081001ffff03036132 / 0b010a000303613200081f46
08b516081001ffff03036B32 / 0b01000003036b32b6c1c445
```

djramipro, 2025-03-11:
@chrizzzp 
So We have to test that "UV" are only used for X=1

chrizzzp, 2025-03-11:
@djramipro 

Yes, I think this is the case and confirms @filippz initial hypothesis, that X=1 refers to daily statistics.
For X=0/2/3 I get identical energy values for every "day".
So, U and V are used to index the days.

chrizzzp, 2025-03-11:
Found some more non-zero daily statistcis with **W=5**, and V ranging from 8 to C.

```
ebusctl decode exp $(ebusctl hex 08b516081001ffff03005732  | tail -c 10) / 0.0
ebusctl decode exp $(ebusctl hex 08b516081001ffff03005832  | tail -c 10) / 16971
ebusctl decode exp $(ebusctl hex 08b516081001ffff03005932  | tail -c 10) / 9415
ebusctl decode exp $(ebusctl hex 08b516081001ffff03005A32  | tail -c 10) / 8145
ebusctl decode exp $(ebusctl hex 08b516081001ffff03005B32  | tail -c 10) / 13692
ebusctl decode exp $(ebusctl hex 08b516081001ffff03005C32  | tail -c 10) / 12052
ebusctl decode exp $(ebusctl hex 08b516081001ffff03005D32  | tail -c 10) / 0.0
```

These seem to be the daily energies from the last 5 days of the previous month (february). Interestingly, there don't seem to be more values from last month. 

Question: are only the values from the last 16 days (11+5) stored? We should see tomorrow if any of the W=5 values changes or is reset to zero...

chrizzzp, 2025-03-12:
> Question: are only the values from the last 16 days (11+5) stored? We should see tomorrow if any of the W=5 values changes or is reset to zero...

Just checked it: 

a) yesterdays first value of last month 'daily energy' history (16 days ago) is zero today.

```
ebusctl decode exp $(ebusctl hex 08b516081001ffff03005832  | tail -c 10) / 0.0
ebusctl decode exp $(ebusctl hex 08b516081001ffff03005932  | tail -c 10) / 9415
```

b) todays (current)  value in this month 'daily energy' history (V= 'C') is a non-zero value now

` ebusctl decode exp $(ebusctl hex 08b516081001ffff03006C32  | tail -c 10) / 2059.79`

So, now we have 4 values for last month (W=5) and 12 values for this month (W=6) in the daily history, which means the values of the last 15 days + todays value are stored.

Open questions:
1) is W used for the daily statistics to index the month as we found for the monthly statistics (I suspect so)?
2) is Q used to index first (Jan-Jul) and second (Aug-Dec) half of the year as found for the monthly statistics?
e.g.
Q=2
W=2/3 January
W=4/5 February
W=6/7 March
W=8/9 April
W=A/B May
W=C/D June
W=E/F July
Q=3?
W=0/1 August
W=2/3 September
W=4/5 October
W=6/7 November
W=8/9 December

3) if V reaches value 'F'? (after day 15 of each month): is U iterated or rather the "second" value of W is used for the days 16-31? Is suspect the latter which would also explain why two values of W encode for each month (in the monthly statisctics), because two 16-byte blocks are needed to fit the daily energy values for 31 days.

chrizzzp, 2025-03-12:
@burmistrzak 

> Just let me know when you think we’re ready for a TypeSpec implementation. 😊
> It should be feasible to have a single energy stats definition file that can be included where needed.

Now, with the daily energy statistics available, this will be a challenge, I think. Although only 16 days are stored in the daily energy history (as it seems now) the address range for these 16 days is "floating", as every day seems to have its own register. I wonder whether this can be accounted for in TypeSpec...

Maybe reading out the `b516` energy statistics needs a completely different ("dynamic") approach?

chrizzzp, 2025-03-12:
@djramipro 

Thanks, your analysis confirms what I observed.

> In my case, I believe that yesterday's value will be enough for register it in HA.

The problem: the Ebus register for "yesterday" (daily energy) is different for every day and therefore cannot simply be used as a "general" HA entity definition in the CSV.

djramipro, 2025-03-13:
@chrizzzp 
Today's testing continues to confirm the assumptions for the values for **W="6"**, days of the current month.
Regarding the values for **W="5"** I don't quite understand it. I explain:
- Before yesterday, **V="a"** had the value 19184 and yesterday it became 0.
- Before yesterday and yesterday, **V="b"** had the value 19926 and today it became 0.
- It is not clear what these values correspond to because I haven't had a way to check them.

@filippz
I would like to test with local configuration files but I don't known what is the logic to add the entries to _templates.csv and bai.308523.inc. Also the changes in mqtt-hassio.cfg

You tested by using:
r,,EnergySumHcPreviousYear,,,,b516,1003ffff03032130,,,IGN:7;energye,,,

but @chrizzzp mention that there is an -i (index) option:

*r,,,,,,B516,,,,IGN:1,,,,,,,,,,,,,,,,,,,,,
r,,CurrentConsumedPowerWatt,,,,,14,,,EXP,,,current consumed electrical power (Watt),,,,,,,,,,,,,,,,,,

Could you tell me some reference to understand how to build these files (including the template) to be able to test locally?

@filippz
I interested in known What will be your proposal for this PR? 
Do you have enough information or do you think more testing and analysis is needed? 
If you need any proof from my side, let me know.

filippz, 2025-03-13:
Nice progress with daily values!
While I do think that documenting all the findings (like the role of **VU**) is important, we have already had discussions whether there is any _real world_ usage case in which defining all those messages makes sense. I for example use only two messages (total gas consumption for hot water and heating) and use HA which stores those values in it's database with date/time after which they can be used in a much more friendly manner.

Earlier, @burmistrzak suggested defining all the messages as passive so everyone can use them only when actually needed, but if we want to use days we would have 366 definitions (multiplied with different combinations) while as @chrizzzp  and @djramipro note only the values of last 15 days are available.

Earlier, I've misread TypeSpec for TypeScript  and concluded that I could use loops to iterate over "index values" of X, Y, Z, W+Q,... we know/need but I don't see loops in TypeSpec. I could reuse my LibreOffice Calc file to generate CSV entries with some modifications for time being.

@djramipro 
> I would like to test with local configuration files

I'm still using my first version described in first post (now fixed), so basically:

- append ebusd-2.1.x/en/vaillant/_templates.csv with `energye,EXP,,Wh,`
- append ebusd-2.1.x/en/vaillant/bai.308523.inc with
```
r1,,FuelSumHc,,,,b516,1000ffff04032132,,,IGN:7;energye,,,
r1,,FuelSumHwc,,,,b516,1000ffff04042132,,,IGN:7;energye,,,
```
- change mqtt-hassio.cfg add `|fuel` to the `filter-name` value and `sensor,gas,total_increasing = fuel|,Wh$` line to `type_switch-number`

In HA you should now see `ebusd bai FuelSumHc` and `ebusd bai FuelSumHwc`

@chrizzzp has a more elegant solution as you noted but I'm still using the old one as I was thinking of going the TypeScript loops route until I figured out that I was googling for TypeScript instead of TypeSpec

chrizzzp, 2025-03-13:
@djramipro 

> Regarding the values for **W="5"** I don't quite understand it. I explain:
>
>     * Before yesterday, **V="a"** had the value 19184 and yesterday it became 0.
>
>     * Before yesterday and yesterday, [...]

Explanation is that the unit seems to store only the last 15 days (+ today), while it resets values older than 5 days (i.e. values become zero, as you observed). The last three values of W=5 (V=d/e/f) remained zero, because February had only 28 days.
So we have sort of "rolling" data for the last 15 days, all other are reset to zero (I don't know why they don't store **all** days of the year...).

According to the current understanding, W=5 corresponds to the 'second block' (15 days) of February (W=4/5) and within the next days we will see these energy values become zero, while more from W=6 ('first block' of March) will appear. As soon as we arrive in the 'second block' of March, W should be 7 (I guess after  the 16th of March).

chrizzzp, 2025-03-13:
@filippz 

> While I do think that documenting all the findings (like the role of **VU**) is important, we have already had discussions whether there is any _real world_ usage case in which defining all those messages makes sense.

I totally agree. I think we should make all decoding information available in some way. Besides providing definitions for e.g. the total values as examples we could document all possible options as remarks, so everyone can create the definitions they're interested in. 
(On could also write a script which creates these definitions based on parameters given...). Just an idea.

djramipro, 2025-03-14:
@filippz @chrizzzp 
thanks for the explanations!

@chrizzzp 
> and the last three values of W=5 (V=d/e/f) remained zero, because February had only 28 days.

Those zeros was what I didn't understand, now it is clear!

filippz, 2025-03-14:
I've made a python script available at https://gist.github.com/filippz/c0d04e4d607e27fc6b89d127db0ee941 that generates names/payloads.

It has no support for days (X=1) and I've I've skipped Cooling on Fuel/Gas as an illogical combination (Y=4 and Z=5) which still leaves 405 combinations.

It should be fairly easy to generate actual CSV lines from it, but the generated values need testings and before all reduction to a reasonable number of useful combinations. For me "reasonable number" is just 2 as I only monitor gas consumption for heating and hot water - for HP this would be obviously more, but probably not hundreds...

chrizzzp, 2025-03-14:
@filippz 

Great, thanks for sharing this!

What I noticed so far:

1) Incrementing W (and, V if incorporated later) should be in hexadecimal (not decimal). For example:

```
'SolarCurrentYearMonth4': '1002ffff01008132'
'SolarCurrentYearMonth5': '1002ffff010010132'
```

should be:

```
'SolarCurrentYearMonth4': '1002ffff01008132'
'SolarCurrentYearMonth5': '1002ffff0100A132'
```

2) There's a typo:

`'Enivormental...'`
should be
`'Environmental'`

3) I would use the suffix 'System' (or 'Total' or something like this) for all total values with X=0
4) I would use the suffix 'Sum' (or something like this) for values with Z=0 (but so far these values are not included in the script)?

filippz, 2025-03-15:
@chrizzzp Fixed.

Z=0 were included in the script for example as `ElectricalCurrentYear` and after the changes it's now named `ElectricalSumCurrentYear` - in both cases with `1003ffff03000132` as payload.

chrizzzp, 2025-03-16:
> According to the current understanding, W=5 corresponds to the 'second block' of February (W=4/5) and the last three values of W=5 (V=d/e/f) remained zero, because February had only 28 days. Within the next days we will see all of these W=5 energy values become zero day by day, while more from W=6 ('first block' of March) will appear. As soon as we arrive in the 'second block' of March, W should be 7 (I guess after the 15th of March).

Just confirmed today, 16th of march, after V has reached value 'f' yesterday (with W=6). Today the 'second block' of march has started and W is incremented to 7 (V=0). 

`ebusctl decode exp $(ebusctl hex 08b516081001ffff03007032  | tail -c 10) / 11505.2`

@filippz  Maybe we should update the first post to summarize our findings.

djramipro, 2025-03-16:
And confirmed from my side:

```
/ # ebusctl decode exp $(ebusctl hex 08b516081001ffff04036132 | tail -c 10) 0.0 (1st of March)
/ # ebusctl decode exp $(ebusctl hex 08b516081001ffff04036232 | tail -c 10) 0.0 (2nd of March)
/ # ebusctl decode exp $(ebusctl hex 08b516081001ffff04036332 | tail -c 10) 49840
...
/ # ebusctl decode exp $(ebusctl hex 08b516081001ffff04036f32 | tail -c 10) 40991  (15th of March)
/ # ebusctl decode exp $(ebusctl hex 08b516081001ffff04037032 | tail -c 10) 30287  (16th of March)

```

chrizzzp, 2025-03-25:
> > And totally **off topic**: do any of you actually use HP for cooling? I'm in the process of planing heating/cooling systems for a family home and there is a lot of conflicting opinions especially in the area of using HP for cooling so experience of more sophisticated users like yourselves would help a lot.
> 
> Our aroTHERM plus is configured for cooling, but our first summer with this thing is still ahead of us. We‘ll be trying out underfloor cooling, mainly in the living room. Theoretically, excess PV energy and cooling are a match made in heaven, but I‘m not sure how effective it actually will be… 🤞

My HP is also prepared to cool (it needs a special coding resistor), but I have only partially tested it as my house is not fully prepared yet for cooling yet. One issue I see with this type of cooling is the relative humidity of the air and the corresponding dew point which (if not monitored) could easily lead to condensation problems. My underfloor heating is underneath a wooden floor so I definitely don't want any condensate water there. Alternatives are HP-driven convectors that cool the air (and not the floor), e.g. the Vaillant aroVAIR units.
I was told from a Vaillant engineer one should not expect the same cooling capacities from a HP-driven cooling systems as those a regular AC unit can provide. The temperature gradient is really smaller here and AFAIK no active dehumidification takes place. 

So I assume while one can indeed lower the room temperature with HPs in cooling mode, it will not "feel" as cold as the relative humidity increases with cooling (unless dehumidifaction is involved). But as I said, no real practical experience from my side...

chrizzzp, 2025-03-25:
@filippz 

> You're referring to "V" and "U" for daily values? Can you suggest what to actually change in the first post - I haven't really worked with daily values so I guess you're be able to provide more meaningful explanation

Yes, I will prepare a description for the top post.

chrizzzp, 2025-03-25:
@filippz 
Just edited my last post with my suggestion for the summary including the daily energy statistics in order to update the top post. Let me know if there are any errors.

filippz, 2025-03-26:
> Just edited my [previous post](https://github.com/john30/ebusd-configuration/issues/490#issuecomment-2750602450) with a suggestion

> Would you mind updating your script to also include daily stats?

@chrizzzp @burmistrzak 
I've updated the first post with your suggestions but rearranged it a bit. I'm off to update the script with daily values and see if there is any mismatched with the contents of the first post.

chrizzzp, 2025-03-26:
@filippz 

> I've updated the first post with your suggestions but rearranged it a bit. I'm off to update the script with daily values and see if there is any mismatched with the contents of the first post.

OK, maybe you could remove the striked-out entries (or keep them separate)? They might confuse people and the formatting gets lost during copy/paste.

filippz, 2025-03-26:
> OK, maybe you could remove the striked-out entries (or keep them separate)? They might confuse people and the formatting gets lost during copy/paste.
> 
> Edit: And mabye you could add the information that for daily stats only the last 16 days are stored.

Done - thanks.

filippz, 2025-03-26:
> I'm off to update the script with daily values

@burmistrzak 
Script has been updated - it now generates 11565 combinations 🤦

chrizzzp, 2025-03-27:
@filippz @burmistrzak 

> Script has been updated - it now generates 11565 combinations

Impressive... thanks for the update! I think we can halmost half this number just by discarding almost all of the daily statistics for the previous year (Q=0/1). As only 16 days are stored we only need the last 16 days of the previous year. All other daily stats values from the previous year will always be zero. 

Still too much to be included in CSVs...

Generally, as @filippz mentioned earlier, the real world benefit of most of the `b516` energy statistics is questionable. Although it is satisfying to have them decoded and I think it's good to share it, I think all that is needed (and most important) are the **total** ('System') energy values, from which all other values can be deduced outside of ebusd (as long a the total values are tracked on a daily basis). So this would be the following 15 payloads (taken from the script output):

```
1000ffff01000030 SolarSumSystem
1000ffff01030030 SolarHeatingSystem
1000ffff01040030 SolarHotWaterSystem
1000ffff01050030 SolarCoolingSystem
1000ffff02000030 EnvironmentalSumSystem
1000ffff02030030 EnvironmentalHeatingSystem
1000ffff02040030 EnvironmentalHotWaterSystem
1000ffff02050030 EnvironmentalCoolingSystem
1000ffff03000030 ElectricalSumSystem
1000ffff03030030 ElectricalHeatingSystem
1000ffff03040030 ElectricalHotWaterSystem
1000ffff03050030 ElectricalCoolingSystem
1000ffff04000030 FuelSumSystem
1000ffff04030030 FuelHotWaterSystem
1000ffff04040030 FuelHotWaterSystem
```
BTW: for definition naming in the CSV I suggest to change the order of the name-parts and add a prefix that distinguishes these definitions as energy statistics, e.g. `EStat`, 'Energy' or similar.

Something like this (@filippz I know, this might need some rearrangements in your script...):
```
SolarSumSystem -> EnergySystemSolarSum
SolarHeatingSystem -> EnergySystemSolarHeating
SolarHotWaterSystem -> EnergySystemSolarHotWater
SolarCoolingSystem -> EnergySystemSolarCooling
```

All other values could be easily extracted from the script output if someone is interested in them.

What do you think?

filippz, 2025-03-27:
> As only 16 days are stored we only need the last 16 days of the previous year.

I did think of that, but it might be that some devices (either existing or possibly future ones) do/will keep track of more than 16 days. On the other hand, the same logic could be applied to Q values as some devices might keep track of more than 1 year in the past...

> I believe all that is needed (and most important) are the **total** ('System') energy values

Agreed. IMHO those are all that is needed - for example, since I'm monitoring electricity with Shelly I'm only interested in gas consumption on my boiler, and I'm just using **r1** for reading gas consumption for heating and hot water polling which renders quite nicely in HA (one random day displayed):

[image]

If we decide to use only those `System` values then previously discussed day definitions might be irrelevant.

> add a prefix that distinguishes these definitions as energy statistics, e.g. `EStat`, `Energy` or similar.

I've saw entries prefixed with _PrEnergy_ in bai csv, but I didn't figure out what what _Pr_ would stand for. On the other hand, I deliberately avoided using `Energy` since I'm not entirely sure how it would affect **mqtt-hassio.cfg** so I changes it like this (form the first post):

>Since I use those in Home Assistan, I've had to add |fuel to the filter-name value and sensor,gas,total_increasing = fuel|,Wh$ line to type_switch-number in mqtt-hassio.cfg

I'm fairly certain that using `EStat` (or something like that) wouldn't break anything

chrizzzp, 2025-03-28:
@filippz @burmistrzak 

> I did think of that, but it might be that some devices (either existing or possibly future ones) do/will keep track of more than 16 days. On the other hand, the same logic could be applied to Q [...]

OK, then it's good to have them in the script if required. I also favor not putting daily statistics in the CSVs...

> I've saw entries prefixed with _PrEnergy_ in bai csv, but I didn't figure out what what _Pr_ would stand for. On the other hand, I deliberately avoided using `Energy` since I'm not entirely sure how it would affect **mqtt-hassio.cfg** so I changes it like this (form the first post):

'Pr' could stand for 'PredictiveMaintenanceData'. 

We could also adopt this prefix, but would have to makes sure they are not mixed up with other Pr* data which are sometimes from different `b5` blocks. Then maybe 'PrEnStat' or something like this...

If we follow existing conventions then we should probably also substitute the following?

```
heating -> hc
hot water -> dhw
```

filippz, 2025-03-28:
> I _assume_ all of them are updated regularly, without polling?

In my case they are being queried by `ctlv2` on either start of `ctlv2` or midnight when `ctlv2` requests fresh data. I didn't see `bai` sending them periodically by itself.

> sing something like e.g. `EnergyStats*` as a prefix shouldn’t be a problem, AFAICT.
> Theoretically, no changes to the MQTT config required. Ideally, the names we settle on are globally (i.e. project-wide) unique.

IIRC I needed to adopt `mqtt-hassio.cfg` in order to get proper units (Wh) and to able to add gas consumption to the appropriate area of HA Energy sensors which HP don't have and therefore might not need any changes in `mqtt-hassio.cfg` - to be safe I intentionally avoided using Energy in that context:

[image]

> Let’s keep `heating` b/c of `cooling`, and change hot water to `hwc`.

In my view we should either use:

- all "short notation", like Hc, Hwc and Cc (as in Cooling circuit, same as Hc might bee Heating circuit)
- all "long" notation like Heating, HotWater and Cooling (like script is doing right now)

In any case - I suppose we'll get some more feedback once we create the PR, so it might not be that important to "get it" perfect for the initial commit/PR.

mar3k, 2025-03-30:
Finally found other guys trying to solve the same problem :) I have the same gas heater "MF=Vaillant;ID=BAI00;SW=1104;HW=7603" and energy reading doesn't work:

`bai PrEnergyCountHc1 =  (ERR: invalid position for 3108b509030df600 / 00)
bai PrEnergyCountHc2 =  (ERR: invalid position for 3108b509030df800 / 00)
bai PrEnergyCountHc3 =  (ERR: invalid position for 3108b509030dfa00 / 00)
bai PrEnergyCountHwc1 =  (ERR: invalid position for 3108b509030dc600 / 00)
bai PrEnergyCountHwc2 =  (ERR: invalid position for 3108b509030dc800 / 00)
bai PrEnergyCountHwc3 =  (ERR: invalid position for 3108b509030dca00 / 00)
bai PrEnergySumHc1 =  (ERR: invalid position for 3108b509030df500 / 00)
bai PrEnergySumHc2 =  (ERR: invalid position for 3108b509030df700 / 00)
bai PrEnergySumHc3 =  (ERR: invalid position for 3108b509030df900 / 00)
bai PrEnergySumHwc1 =  (ERR: invalid position for 3108b509030dc500 / 00)
`
In addition I have "MF=Vaillant;ID=CTLV3;SW=0709;HW=3704" VRC720 SensoComfort unit but in terms of energy consumption it only returns one parameter:

`ctlv3 PrEnergySum =  (ERR: invalid position for 3115b52406020000005c00 / 00)
ctlv3 PrEnergySumHc = 111
ctlv3 PrEnergySumHcLastMonth = no data stored
ctlv3 PrEnergySumHcThisMonth = no data stored
ctlv3 PrEnergySumHwc = no data stored
ctlv3 PrEnergySumHwcLastMonth = no data stored
ctlv3 PrEnergySumHwcThisMonth = no data stored
ctlv3 PrFuelSum = no data stored
ctlv3 PrFuelSumHc = no data stored
ctlv3 PrFuelSumHcLastMonth = no data stored
ctlv3 PrFuelSumHcThisMonth = no data stored
ctlv3 PrFuelSumHwc = no data stored
ctlv3 PrFuelSumHwcLastMonth = no data stored
ctlv3 PrFuelSumHwcThisMonth = no data stored
`
So I can't practically read any energy consumption data into my smart home system. 

Can't wait for your PR :)

chrizzzp, 2025-03-31:
> Can't wait for your PR :)

@mar3k 
If you use local CSVs you can start right away, e.g. sticking to the instructions @filippz gave in his [post](https://github.com/john30/ebusd-configuration/issues/490#issuecomment-2721947204) above.

chrizzzp, 2025-03-31:
@mar3k 

> and energy reading doesn't work:
> 
> bai PrEnergyCountHc1 = (ERR: invalid position for 3108b509030df600 / 00)
> bai PrEnergyCountHc2 = (ERR: invalid position for 3108b509030df800 / 00)

These definitions might be valid only for certain (older?) types of heat generators. They also don't work for the recent heat pump units (`hmu`). That's why this issue was started (now solved). The energy statistics are available from the `b516` block (not `b509`).

>In addition I have "MF=Vaillant;ID=CTLV3;SW=0709;HW=3704" VRC720 SensoComfort unit but in terms of energy consumption it >only returns one parameter:
>
> ctlv3 PrEnergySumHc = 111
> ctlv3 PrEnergySumHcLastMonth = no data stored
> ctlv3 PrEnergySumHcThisMonth = no data stored
> ...

Did you try to actively poll these? E.g. with

`ebusctl r -f -c ctlv3 PrEnergySumHcLastMonth`

mar3k, 2025-04-02:
@chrizzzp now I managed to switch from Docker-run ebusd to locally installed and made recommended modifications. 

Question: very recently (beginning of a month?) I noticed my heater display and VRC720 device show majority of gas/electricity consumption parameters as empty (zero)!?!? Only total consumption is kept correctly. Is it possible ebusd wiped/overwrote those values in my system by error? (this happened before enabling custom configuration btw). 

Regarding the new custom configuration I tried:
ebusctl r -f -c bai FuelSumHc
1.00761e+07

Is this value returned ok? It looks way too small ... 

Btw @filippz could you share your CSV for energy consumption reads using b516 block? I am new to this and not quite sure how to modify it.

chrizzzp, 2025-04-03:
@mar3k 

> Question: very recently (beginning of a month?) I noticed my heater display and VRC720 device show majority of gas/electricity consumption parameters as empty (zero)!?!? Only total consumption is [...]

I'm not sure what you mean by 'custom configuration' (local CSVs?) but this should not happen. Although it is theoreticall possible to issue unintended write commands using the `hex` mode of `ebusctl` (did you use it?) if there are typos, I'm not aware of a command that erases most of the statistics. The statistics show in the regulator are updated every night after midnight. Did you check whether they are back?

You could also check directly in the heater statistics if these values are really zero.

e.g.
`ebusctl decode exp $(ebusctl hex 08b516081003ffff04032132  | tail -c 10)`

should give you this years total gas consumption used for heating.

> Regarding the new custom configuration I tried: ebusctl r -f -c bai FuelSumHc 1.00761e+07
> 
> Is this value returned ok? It looks way too small ...

Well, could you post the definition line of `FuelSumHc` from the CSV?

What I assume: in contrast to the `b516` statistics (in Wh), the above values are given in kWh (template `energy` or `energy4`?). Your reading would then correspond to 10'076'100 kWh. Does this make more sense? I'm not used to such high numbers, but maybe your house is a bit bigger than mine...

mar3k, 2025-04-03:
> I'm not sure what you mean by 'custom configuration' (local CSVs?) but this should not happen. Although it is theoreticall possible to issue unintended write commands using the `hex` mode of [...]

Yes, local CSVs as per instructions. No I haven't used hex mode as I am not familiar with it yet. Statistics today are somewhat improved on the regulator (still off), but very bad on the heater itself (gas consumption for today is showing ok, but yesterday/lastmonth/lastyear is 0, total is correct).

> 
> You could also check directly in the heater statistics if these values are really zero.
> 
> e.g. `ebusctl decode exp $(ebusctl hex 08b516081003ffff04032132 | tail -c 10)`

This particular one returns "ERR: invalid numeric argument". Where do I get these codes/hexa-numbers from for all my readings?

> 
> should give you this years total gas consumption used for heating.
> 
> Well, could you post the definition line of `FuelSumHc` from the CSV?

Sure, this is the one:
`r1,,FuelSumHc,,,,b516,1000ffff04032132,,,IGN:7;energye,,,`

> 
> What I assume: in contrast to the `b516` statistics (in Wh), the above values are given in kWh (template `energy` or `energy4`?). Your reading would then correspond to 10'076'100 kWh. Does this make more sense? I'm not used to such high numbers, but maybe your house is a bit bigger than mine...

Hmm actually my total consumption is ~10.000 kWh so looks like it's 10'076'100 Wh.

chrizzzp, 2025-04-03:
@mar3k 

> ure, this is the one: `r1,,FuelSumHc,,,,b516,1000ffff04032132,,,IGN:7;energye,,,`

Ah, I see. You're already using the "new" `b516` energy stats. Then the number is of course in Wh.

> Hmm actually my total consumption is ~10.000 kWh so looks like it's 10'076'100 Wh.

correct

[34 more replies]