john30/ebusd-configuration #491: Mqtt hassio errors

pascalwinters, 2025-03-08:
- Error 'The unit of measurement `%` is not valid together with device class `temperature`' when processing MQTT discovery message topic: 'homeassistant/sensor/ebusd_vr_71_Mc3FlowTempDesired_MixerMovement/config', message: '{'unique_id': 'ebusd_vr_71_Mc3FlowTempDesired_MixerMovement', 'device': {'identifiers': 'ebusd_vr_71', 'manufacturer': 'ebusd.eu', 'via_device': 'ebusd', 'sw_version': '24.1', 'suggested_area': 'Heating', 'name': 'ebusd vr_71'}, 'value_template': '{{value_json["MixerMovement"].value}}', 'state_topic': 'ebusd/vr_71/Mc3FlowTempDesired', 'unit_of_measurement': '%', 'state_class': 'measurement', 'device_class': 'temperature', 'name': 'Mc3FlowTempDesired MixerMovement'}'

- Error 'The unit of measurement `bar` is not valid together with device class `temperature`' when processing MQTT discovery message topic: 'homeassistant/sensor/ebusd_vwzio_RunDataSystemPressure_pressv/config', message: '{'unique_id': 'ebusd_vwzio_RunDataSystemPressure_pressv', 'device': {'identifiers': 'ebusd_vwzio', 'manufacturer': 'ebusd.eu', 'via_device': 'ebusd', 'sw_version': '24.1', 'suggested_area': 'Heating', 'name': 'ebusd vwzio'}, 'value_template': '{{value_json["pressv"].value}}', 'state_topic': 'ebusd/vwzio/RunDataSystemPressure', 'unit_of_measurement': 'bar', 'state_class': 'measurement', 'device_class': 'temperature', 'name': 'RunDataSystemPressure '}'

chrizzzp, 2025-03-10:
Same here. Could this be related to the update of the HA core (to 2025.3.1)? Seems the values don't get updated since then anymore.

- Error 'The unit of measurement `%` is not valid together with device class `temperature`' when processing MQTT discovery message topic: 'homeassistant/sensor/ebusd_vr_71_Mc1FlowTempDesired_MixerMovement/config', message: '{'unique_id': 'ebusd_vr_71_Mc1FlowTempDesired_MixerMovement', 'device': {'identifiers': 'ebusd_vr_71', 'manufacturer': 'ebusd.eu', 'via_device': 'ebusd', 'sw_version': '24.1', 'suggested_area': 'Heating', 'name': 'ebusd vr_71'}, 'value_template': '{{value_json["MixerMovement"].value}}', 'state_topic': 'ebusd/vr_71/Mc1FlowTempDesired', 'unit_of_measurement': '%', 'state_class': 'measurement', 'device_class': 'temperature', 'name': 'Mc1FlowTempDesired MixerMovement'}'

chrizzzp, 2025-05-18:
See here: https://github.com/john30/ebusd/issues/1512

Changing `Mc1FlowTempDesired` to `Mc1FlowStatus` solved the issue for me (using ebusd v 25.1 with original `mqtthassio.cfg`). Don't really understand why, though...

```
*r,,,,,,B523,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
*uw,,,,,,B523,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
uw,,Mc1FlowStatus,,,,,0200,Status,,onoff,,,,FlowTempDesired,,temp1,,,,MixerStatus,s,onoff,,,,MixerMovement,s,percents,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,,
```

However, I still have two definitions in my CSV which still throw a similar error, wrongly assigning the device class 'temperature' to a pressure reading:

```
Entity sensor.ebusd_hmu_status07_displaysystempressure (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'bar' which is not a valid unit for the device class ('temperature') it is using; expected one of ['°F', 'K', '°C']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
Entity sensor.ebusd_vwz00_rundatasystempressure_pressv (<class 'homeassistant.components.mqtt.sensor.MqttSensor'>) is using native unit of measurement 'bar' which is not a valid unit for the device class ('temperature') it is using; expected one of ['°F', 'K', '°C']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+mqtt%22
```

Any problems with the message definition maybe (I know `Status07` is a complex one...) ?
```
*r,,,,,,B511,,,,,,,,,,,,,,,,,,,,,,,,,
r,,Status07,,,,,07,power,,UCH,,%,Compressor power,dailyenvyield,,energy,10,kWh,daily environmental energy yield,disp1ay_b0_heaterenabled,,BI0,0=off;1=on,,,display_b1,,BI1,0=off;1=on,,,display_b2_backupheater,,BI2,0=off;1=on,,,display_b3,,BI3,0=off;1=on,,,display_b4,,BI4,0=off;1=on,,,display_b5_noisereduction,,BI5,0=off;1=on,,,display_b6_dhwecomode,,BI6,0=off;1=on,,,display_b7,,BI7,0=off;1=on,,,DisplayHeaterMain,,state07heatermode,,,,DisplaySystemPressure,,press30,,,,DisplayHeaterBackup,,state07heatermode,,,,
```
 using the following template defintion:

`state07heatermode,UCH,0x00=standby;0x01=standby;0x0a=error;0x0b=error;0x08=hc;0x09=hc;0x10=cool;0x11=cool;0x80=dhw;0x81=dhw,,
`
and

```
*r,,,,,,B509,,,,,,,,,,,,,,,,,,,,,,,,,
r,,RunDataSystemPressure,,,,,02007409,,,EXP,,bar,,,,,,,,,,,,,,,,,,,

```