john30/ebusd-configuration #494: 'ebusd/bai/HcPumpMode', got '3', allowed: post_run, permanent, winter

Kolbi, 2025-03-13:
Logs:

> 2025-03-13 09:10:35.295 WARNING (MainThread) [homeassistant.components.mqtt.sensor] Ignoring invalid option received on topic 'ebusd/bai/HcPumpMode', got '3', allowed: post_run, permanent, winter

Using: https://ebus.github.io/next/

`ebusctl scan result
08;Vaillant;BAI00;0104;7803;21;19;15;0010021924;0001;006480;N6
15;Vaillant;70000;0510;6403;21;19;08;0020242192;0082;014704;N2
26;Vaillant;VR_71;0104;0503;21;19;10;0020184846;0082;005617;N9
`

`ebusctl info
version: ebusd 24.1.24.1-30-ga8f876d
update check: revision 24.1 available, device firmware v5 1[5210] up to date, broadcast.csv: different version available, vaillant/08.bai.csv: different version available
device: 192.168.21.239, TCP, enhanced, firmware 1.1[5210].1[5210]
signal: acquired
symbol rate: 23
max symbol rate: 116
min arbitration micros: 2
max arbitration micros: 17
min symbol latency: 5
max symbol latency: 35
scan: finished
reconnects: 0
masters: 3
messages: 714
conditional: 3
poll: 249
update: 16
address 03: master #11
address 08: slave #11, scanned "MF=Vaillant;ID=BAI00;SW=0104;HW=7803", loaded "vaillant/bai.0010021961.inc" ([id_hw=7803]), "vaillant/08.bai.csv"
address 10: master #2
address 15: slave #2, scanned "MF=Vaillant;ID=70000;SW=0510;HW=6403", loaded "vaillant/15.700.csv"
address 26: slave, scanned "MF=Vaillant;ID=VR_71;SW=0104;HW=0503", loaded "vaillant/26.vr_71.csv"
address 31: master #8, ebusd
address 36: slave #8, ebusd`

Kolbi, 2025-03-13:
Seems to be related to this PR but needs to be implemented in more files: https://github.com/john30/ebusd-configuration/pull/480

john30, 2025-04-12:
merged+reworked in [1b28da7](https://github.com/john30/ebusd-configuration/commit/1b28da7402f8d292fb58122ad08eefb8e52175e6)

closed 2025-04-12 by john30